espressif/esp-idf #410: ESP32 - iBeacon scanner - how to do that ?

Netoperz, 2017-03-08:
Hello,
My project was based on esp8266, the device was scanning iBeacons, to get the rssi (and calculate the distance to nearest becaon) i was using my own firmware on hc08 bt module and was communicating with that module over uart.

Now I'm porting all the code i have from esp8266 to esp 32, the device will be commercial device and i have one question, how to scan for ibeacons using esp32 BT function to get the list of iBeacons with their rssi  to calculate the nearest one ?

Is there any sample on hint how to use bt function od esp32 as iBeacon canner ?

That would be very useful for me.

Sorry for posting thisn as an issue, but nowhere alse noone knows how to do that.

...

heyinling, 2017-03-08:
Hi,

  I think you're looking for BLE scan examples. `esp-idf/examples/bluetooth/gatt_client` contains the usage of scan APIs.

Thanks!

Netoperz, 2017-03-08:
Thank you,
is there a way to do that on arduino esp ?

Because all code was build on arduino IDE, so if scanning is possible using that way, it would be much faster to port, than refactor  and rewrite everything for rtos :/

copercini, 2017-03-09:
You can use IDF code in arduino IDE and arduino code in IDF....

Here is a example of GATT client for arduino (I know, it's not a simple code, but works)
https://gist.github.com/me-no-dev/5f8851e1e1112b7c3d9eefe6e1e1cf9d

neilyoung, 2017-03-13:
@Netoperz : Would you mind to give my little snippet a try as published here? https://github.com/espressif/esp-idf/issues/421

FayeY, 2017-05-02:
Hi Netoperz, is this problem still unsolved?

Netoperz, 2017-05-04:
Hello, the case is open, because we are focused on another project because we have signed contract that needs to be finished in next two weeks, and we had to froze tasks for this project. We should start working with this feature in the second half of May.

I'll post results wen i get some.

BrunDog, 2017-05-19:
Hi Guys,

I could use some help with the scanner... trying to read iBeacons with ESP32 via Arduino IDE. I appreciate the swapability betwwen the IDF and the Arduino IDE but I am not having success understanding porting the code over. Any advice appreciated. Anyone posting a code example REALLY appreciated!

duncanhoggan, 2017-09-14:
In the ESP_GAP_BLE_SCAN_RESULT_EVT callback the scan_result object is formed and the rssi is accessible there...
`scan_result->scan_rst.rssi`

FayeY, 2017-11-22:
See [iBeacon example](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/ble_ibeacon)

closed 2017-11-22 by FayeY