espressif/esp-idf #411: Improving FreeRTOS configurability

ammaree, 2017-03-08:
In order to port some applications from the CC3200 to the ESP32 we need to have a number of FreeRTOS configuration macros exposed through (I assume) "make menuconfig". Currently the required variables include:
	configGENERATE_RUN_TIME_STATS	Required for task runtime stats, also need definitions for
									portGET_RUN_TIME_COUNTER_VALUE and
									portCONFIGURE_TIMER_FOR_RUN_TIME_STATS
	configUSE_TRACE_FACILITY		Debugging/trace
	configMAX_TASK_NAME_LEN		Not critical, current minimum limit is still TMR_SRV
	configQUEUE_REGISTRY_SIZE		Used in debugging/trace tool
	configTIMER_TASK_PRIORITY

Thanks

forthlightning, 2017-03-09:
I would add configTIMER_QUEUE_LENGTH to this list for convenience

nkolban, 2017-06-01:
I'd like +1 for the `configUSE_TRACE_FACILITY`

Dazza0, 2017-11-21:
Please see commits d1853dbbc06fcc3f163f7e9ece1af3007305c184 and 3c338076895f28b6b8029eb5d6548447209a913b .

closed 2017-11-22 by FayeY