espressif/esp-idf #412: WIFI Doesnt Connect after pulling latest commit

llewellynnu, 2017-03-08:
I havent changed anything in the code . All i did was pulled the latest commit for the esp-idf and the wifi stopped connecting . It just freezes and doesnt connect

wilberforce, 2017-03-09:
Erase flash and resend the firmware. There might have been changes in the wifi config.

igrr, 2017-03-09:
Also did you happen to do `git submodule update` after `git pull`?

llewellynnu, 2017-03-09:
Yeah i did a submodule update since it prompted me to do so. 

I also tried to erase the flash but i get the following error

A fatal error occurred: ESP32 ROM does not support function erase_flash.

llewellynnu, 2017-03-09:
I managed to erase the flash by running make erase_flash and then reflashed the ESP32 but the same problem persists . It fails to connect to Wifi AP

igrr, 2017-04-11:
@llewellynnu does the issue still happen if you try v2.0 release?
If it does, could you please post the model of your WiFi router?

llewellynnu, 2017-04-11:
I havent tried the V2.0 yet as it is way behind and misses alot of things that i need for my project . 
I will still have a go at Wifi using V2.0 and let you kno how it goes 

Regards

igrr, 2017-04-11:
Just to clarify, are you able to reproduce this using one of the examples bundled with the ESP-IDF, when using latest master?

FayeY, 2017-05-02:
Hi llewellynnu, is this problem still unsolved?

llewellynnu, 2017-05-02:
Hi ,

Yeah it is resolved you can close this issue

Regards

-- 
*Llewellyn Fernandes*
Developer
Numadic <http://numadic.com/>
Undisrupting logistics.™

5th Floor
Central Towers
Panjim, Goa
Sales & Support: 777-600-5544
Mobile : +91 8806460566
_____________________________

This transmission is intended solely for the person or organization to whom
it is addressed and it may contain privileged and confidential information.
If you are not the intended recipient you must not copy, distribute or take
any action in reliance on it. If you believe you received this transmission
in error please notify the sender.

closed 2017-05-02 by FayeY