espressif/esp-idf #415: LWIP loopback feature is unavailable

young-mu, 2017-03-10:
I write these codes to test LWIP loopback feature, it can not work now. So please validate it and fix it. Thanks.

```c
    tcpip_adapter_init();
    struct sockaddr_in addr;
    struct sockaddr_in raddr;
    int addr_len = sizeof(struct sockaddr_in);
    memset(&addr, 0, sizeof(addr));

    addr.sin_family = AF_INET;
    addr.sin_addr.s_addr = htonl(INADDR_LOOPBACK);
    addr.sin_port = htons(8888);

    int sfd = 0;
    int cfd1 = 0;
    int cfd2 = 0;

    sfd = socket(AF_INET, SOCK_STREAM, 0);
    if (sfd == -1) {
        printf("socket sfd failed\n");
        goto clean;
    }

    if (bind(sfd, (struct sockaddr*)&addr, addr_len) == -1) {
        printf("bind failed\n");
        goto clean;
    }

    if (listen(sfd, 5) == -1) {
        printf("listen failed\n");
        goto clean;
    }

    cfd1 = socket(AF_INET, SOCK_STREAM, 0);
    if (cfd1 == -1) {
        printf("socket cfd1 failed\n");
    }

    if (connect(cfd1, (const struct sockaddr*)&addr, addr_len) == -1) {
        printf("connect failed with errno %d\n", errno);
        goto clean;
    }

    cfd2 = accept(sfd, (struct sockaddr*)&raddr, (socklen_t *)&addr_len);
    if (cfd2 == -1) {
        printf("accept failed\n");
        goto clean;
    }

    printf("server IP: %s:%d\n", inet_ntoa(addr.sin_addr), ntohs(addr.sin_port));
    printf("client IP: %s:%d\n", inet_ntoa(raddr.sin_addr), ntohs(raddr.sin_port));

    printf("server socket = %d\n", cfd2);
    printf("client socket = %d\n", cfd1);

clean:
    if (sfd != -1) {
        close(sfd);
    }

    if (cfd1 != -1) {
        close(cfd1);
    }

    if (cfd2 != -1) {
        close(cfd2);
    }
```

The log shows 'connect failed with errno 113' if connected to AP and 'connect failed with errno 118' if NOT connected.

nkolban, 2017-03-11:
This might be a duplicate of or be related to ... https://github.com/espressif/esp-idf/issues/114

young-mu, 2017-03-13:
This is no summary/conclusion for that similar issue. So I open this.

FayeY, 2017-11-22:
Duplicate issue. Please see #114 for more details.

closed 2017-11-22 by FayeY