espressif/esp-idf #418: BLE characteristic "long write"

h-vishal, 2017-03-11:
In the gatt_server example the CPU crashes if you try and write more than 20 or so bytes without modifying MTU size - this was tested using the Android 6 GATT client. The expected behaviour would be to be able to write values up to ~512 bytes without modifying the MTU size.

danicampora, 2017-03-20:
@igrr this is the same issue that I was reporting a few days ago. Any comments from your side?

xcguang, 2017-04-06:
Hi @h-vishal @danicampora ,
The result of crash is that the demo does not handle the prepare write event.Maybe you can handle it like my process.
```
static void gatts_profile_a_event_handler(esp_gatts_cb_event_t event, esp_gatt_if_t gatts_if, esp_ble_gatts_cb_param_t *param) {
    static esp_gatt_rsp_t rsp;
    static esp_gatt_rsp_t temp_rsp;
    static esp_gatt_rsp_t backup_rsp;
    switch (event) {
    case ESP_GATTS_REG_EVT:
        ESP_LOGI(GATTS_TAG, "REGISTER_APP_EVT, status %d, app_id %d\n", param->reg.status, param->reg.app_id);
        gl_profile_tab[PROFILE_A_APP_ID].service_id.is_primary = true;
        gl_profile_tab[PROFILE_A_APP_ID].service_id.id.inst_id = 0x00;
        gl_profile_tab[PROFILE_A_APP_ID].service_id.id.uuid.len = ESP_UUID_LEN_16;
        gl_profile_tab[PROFILE_A_APP_ID].service_id.id.uuid.uuid.uuid16 = GATTS_SERVICE_UUID_TEST_A;

        esp_ble_gap_set_device_name(TEST_DEVICE_NAME);
#ifdef CONFIG_SET_RAW_ADV_DATA
        esp_ble_gap_config_adv_data_raw(raw_adv_data, sizeof(raw_adv_data));
        esp_ble_gap_config_scan_rsp_data_raw(raw_scan_rsp_data, sizeof(raw_scan_rsp_data));
#else
        esp_ble_gap_config_adv_data(&test_adv_data);
#endif
        esp_ble_gatts_create_service(gatts_if, &gl_profile_tab[PROFILE_A_APP_ID].service_id, GATTS_NUM_HANDLE_TEST_A);
        rsp.attr_value.len = 4;
        rsp.attr_value.value[0] = 0xde;
        rsp.attr_value.value[1] = 0xed;
        rsp.attr_value.value[2] = 0xbe;
        rsp.attr_value.value[3] = 0xef;
        break;
    case ESP_GATTS_READ_EVT: {
        ESP_LOGI(GATTS_TAG, "GATT_READ_EVT, conn_id %d, trans_id %d, handle %d\n", param->read.conn_id, param->read.trans_id, param->read.handle);
        ESP_LOGI(GATTS_TAG, "GATT_READ_EVT, value len %d\n", rsp.attr_value.len);
        rsp.attr_value.handle = param->read.handle;

        esp_ble_gatts_send_response(gatts_if, param->read.conn_id, param->read.trans_id,
                                    ESP_GATT_OK, &rsp);
        break;
    }
    case ESP_GATTS_WRITE_EVT: {
        ESP_LOGI(GATTS_TAG, "GATT_WRITE_EVT, conn_id %d, trans_id %d, handle %d\n", param->write.conn_id, param->write.trans_id, param->write.handle);
        ESP_LOGI(GATTS_TAG, "GATT_WRITE_EVT, value len %d, value %08x\n", param->write.len, *(uint32_t *)param->write.value);

        if (param->write.is_prep) {
            memset(&temp_rsp, 0, sizeof(esp_gatt_rsp_t));
            temp_rsp.attr_value.handle = param->write.handle;
            memcpy(temp_rsp.attr_value.value,param->write.value,param->write.len);
            temp_rsp.attr_value.len += param->write.len;
            temp_rsp.attr_value.offset += param->write.len;
            if ((backup_rsp.attr_value.len + param->write.len) <= ESP_GATT_MAX_ATTR_LEN) {
                memcpy(backup_rsp.attr_value.value + backup_rsp.attr_value.offset,param->write.value,param->write.len);
                backup_rsp.attr_value.len += param->write.len;
                backup_rsp.attr_value.offset += param->write.len;
                esp_ble_gatts_send_response(gatts_if, param->write.conn_id, param->write.trans_id, ESP_GATT_OK, &temp_rsp);
            } else {
                esp_ble_gatts_send_response(gatts_if, param->write.conn_id, param->write.trans_id, ESP_GATT_PREPARE_Q_FULL, &temp_rsp);
            }
        } else {
            rsp.attr_value.len = param->write.len;
            memcpy(rsp.attr_value.value,param->write.value,param->write.len);
            esp_ble_gatts_send_response(gatts_if, param->write.conn_id, param->write.trans_id, ESP_GATT_OK, NULL);
        }
        break;
    }
    case ESP_GATTS_EXEC_WRITE_EVT:
        ESP_LOGI(GATTS_TAG, "EXEC_WRITE_EVT, value len %d\n", backup_rsp.attr_value.len);
        if (param->exec_write.exec_write_flag == ESP_GATT_PREP_WRITE_EXEC) {
            ESP_LOGI(GATTS_TAG, "EXEC_WRITE_EVT, write\n");
            rsp.attr_value.len = backup_rsp.attr_value.len;
            memcpy(rsp.attr_value.value,backup_rsp.attr_value.value,backup_rsp.attr_value.len);
        } else {
[... 51 lines cut ...]
                 param->start.status, param->start.service_handle);
        break;
    case ESP_GATTS_STOP_EVT:
        break;
    case ESP_GATTS_CONNECT_EVT:
        ESP_LOGI(GATTS_TAG, "SERVICE_START_EVT, conn_id %d, remote %02x:%02x:%02x:%02x:%02x:%02x:, is_conn %d\n",
                 param->connect.conn_id,
                 param->connect.remote_bda[0], param->connect.remote_bda[1], param->connect.remote_bda[2],
                 param->connect.remote_bda[3], param->connect.remote_bda[4], param->connect.remote_bda[5],
                 param->connect.is_connected);
        gl_profile_tab[PROFILE_A_APP_ID].conn_id = param->connect.conn_id;
        break;
    case ESP_GATTS_DISCONNECT_EVT:
        esp_ble_gap_start_advertising(&test_adv_params);
        break;
    case ESP_GATTS_OPEN_EVT:
    case ESP_GATTS_CANCEL_OPEN_EVT:
    case ESP_GATTS_CLOSE_EVT:
    case ESP_GATTS_LISTEN_EVT:
    case ESP_GATTS_CONGEST_EVT:
    default:
        break;
    }
}
```

closed 2017-06-19 by FayeY