espressif/esp-idf #419: Race conditions or other timing issue in gethostbyname()

nkolban, 2017-03-12:
I have a large project which is using "gethostbyname".  In my environment I am disabling DHPC client and using tcpip_adapter_set_ip_info() to set the IP, GW and netmask.  This all works great.  I am then calling dns_setserver() to set the DNS server to use (I am using googles 8.8.8.8).  When I call gethostbyname() for a well known server name, I found myself getting the return code 210 stating host not found.  I dug all over the place and found myself stymied.  I removed as much code as I could and still it wasn't working but yet "should".  When I connected as a station and used DHCP client, all worked.

In an effort to diagnose the problem, I switched on LWIP debugging by setting `LWIP_DEBUG = LWIP_DBG_ON` and setting `DNS_DEBUG = LWIP_DBG_ON`.   Merely by switching on this debugging, the `gethostbyname()` returned with no errors or issues.  I switch off the debugging, the exact same code returns an error that hostname was not found.

My logic for invoking `gethostbyname()` is in a task that is started from the WiFi Event Handler callback upon receipt of the GOT_IP event.

My coding is in C++.  And that's about it ... I can demonstrate on my system but when I try and capture debug, all works.  My "gut" feel is that the tracing output is causing pause in the internals that is causing something to work ... i.e. some kind of race condition.

I fully realize that this report isn't a great report ... I don't have a project for you where you can run a recreate.  If needed, we can email/chat directly and I can zip-up the whole project sub-directories plus instructions and mail them to you.  Alternatively (or in addition), I can make myself available for a screen share.  My ideal would be that your Subject Matter Experts in this area might know of race condition issues relating to `gethostbyname()` when DHCP client was not used to obtain the DNS server information.

negativekelvin, 2017-03-12:
This is where your error seems to come from https://github.com/espressif/esp-idf/blob/master/components/lwip/core/dns.c#L1489

nkolban, 2017-03-12:
Do you think there is a logic error in my code or a problem in the SDK or somewhere in between?  I am confident that I made a previous call to dns_setserver() with index 0 passing in the IP address of 8.8.8.8.

negativekelvin, 2017-03-12:
Not sure, you may want to print the DNS IP before you call gethostbyname and add some more debug statements to the lwip code.  Not sure why dns_send is called 3 times and there is no error or timeout for the first two.

nkolban, 2017-03-12:
I logged the DNS IP using `dns_getserver()` before making the `gethostbyname()` call and it was good.  

I also checked endianness ... all still good.

negativekelvin, 2017-03-12:
Is SYSTEM_EVENT_STA_GOT_IP valid when dhcp is disabled?

nkolban, 2017-03-12:
I haven't seen a state diagram of the ESP32 networking subsystems but experience seems to show me that if we switch off the DHCP client functions and then make a WiFi connect to an access point, the SYSTEM_EVENT_STA_GOT_IP event is fired and any callback handler is invoked.  The IP address information supplied contains the same IP address that we had requested before making the station connect API call.

projectgus, 2017-03-13:
Do you have an example of code which does this?

There may be a race condition between SYSTEM_EVENT_STA_GOT_IP and the LWIP interface being configured for routing, if the DHCP server is disabled.

nkolban, 2017-03-13:
My code base is large ... what I will try and do is create a small subset that illustrates the issue.  This also appears to be a strangely non-deterministic problem ... but not non-deterministic in time ... but non-deterministic in "surrounding code".  Bizarre as it sounds, if I create a C++ class that has NOTHING to do with networking and create an instance of that class, the name resolution works ... while if I don't instantiate the class, it doesn't work.   This is truly odd ... my best thinking at this time is that the app is sensitive to odd/even memory layouts.  For example, by defining a new variable,  or allocating a class, I am moving some data into an word/dword boundary and it works ...

I'll pour over it again and see if I can't find a different recreate.  The pain is that bug is quantum in nature ... when I switch on diagnostics, the problem doesn't manifest itself even though the diagnostics are supposed to be transparent.

liuzfesp, 2017-05-10:
@nkolban do you fixed this issue or not? If not, I'm going to investigate this one..

nkolban, 2017-05-10:
No sir ... I haven't looked at it since it was raised in March.

liuzfesp, 2017-06-07:
Hi @nkolban I just can't reproduce this issue meself, I could you paste following code in your project:
1. The code to init WiFi/LWIP etc, such as when call esp_wifi_init/esp_wifi_start/esp_wifi_connect etc
2. Event handler function.

I suspect this issue is because you call gethostbyname() before the WiFi is connected, but I'm not sure because I can't see the implementation.

liuzfesp, 2017-06-07:
Hi @nkolban since I can't reproduce this issue myself, so I investigate the DNS source code.
The log when problem happens:
L1: dns_enqueue: "example.com": use DNS entry 0
L2: dns_enqueue: "example.com": use DNS pcb 0
L3: dns_send: dns_servers[0] "example.com": request
L4: sending DNS request ID 36047 for name "example.com" to server 0
L5: dns_tmr: dns_check_entries
L6: dns_send: dns_servers[0] "example.com": request
L7: sending DNS request ID 36047 for name "example.com" to server 0
L8: dns_tmr: dns_check_entries
L9: dns_send: dns_servers[0] "example.com": request
L10: sending DNS request ID 36047 for name "example.com" to server 0
L11: lwip_gethostbyname(example.com) failed, err=-6
L12: D (5334) WiFi: Unable to resolve example.com - 210

Log analysis:
L1~L10: it means DNS retries 2 times, the max allowed retry time is 4 (#define DNS_MAX_RETRIES           4）, it means DNS doesn't timeout since it only retry 2 times. Moreover, the log will show "dns_check_entry: example.com, timeout" if DNS timeout, but it doesn't. 

L11: err=-6 means " #define ERR_VAL   -6  /*Illegal value */", the err code is printed in following code:
 
lwip_gethostbyname(const char *name)
{
....
  /* query host IP address */
  err = netconn_gethostbyname(name, &addr);
  if (err != ERR_OK) {
    LWIP_DEBUGF(DNS_DEBUG, ("lwip_gethostbyname(%s) failed, err=%d\n", name, err));
    h_errno = HOST_NOT_FOUND;
    return NULL;
  }
....
}

the place to return the error code is in dns_call_found(), check all places that dns_call_found is called, we found that the only place causes error code -6 should in following code:
static err_t
dns_send(u8_t idx)
{
  err_t err;
  struct dns_hdr hdr;
  struct dns_query qry;
  struct pbuf *p;
  u16_t query_idx, copy_len;
  const char *hostname, *hostname_part;
  u8_t n;
  u8_t pcb_idx;
  struct dns_table_entry* entry = &dns_table[idx];

  LWIP_DEBUGF(DNS_DEBUG, ("dns_send: dns_servers[%"U16_F"] \"%s\": request\n",
              (u16_t)(entry->server_idx), entry->name));
  LWIP_ASSERT("dns server out of array", entry->server_idx < DNS_MAX_SERVERS);
  if (ip_addr_isany_val(dns_servers[entry->server_idx])) {
    /* DNS server not valid anymore, e.g. PPP netif has been shut down */
    /* call specified callback function if provided */
    dns_call_found(idx, NULL);
    /* flush this entry */
    entry->state = DNS_STATE_UNUSED;
    return ERR_OK;
  }
...

So I doubt that dns_server is changed by someone, or entry->server_idx is changed by someone, or there is something wrong in the project which cause memory overwritten etc

... ...
The challenge for me now is can't reproduce this issue myself....

nkolban, 2017-06-07:
Howdy my friend.  Since the code base I was working on was large and I didn't take the time to recreate in an isolated environment, I have nothing more to offer.  Since its now 3 months after raising the issue, and I myself forgot about it, I suggest that we close this issue for now as a "glitch".  Thanks to Github this issue will always remain searchable so if I or someone else find a recurrence, hopefully we can find this issue again and add more clues to the story.

I do appreciate you taking the time to investigate.

Neil

closed 2017-06-07 by nkolban

vguadagnin, 2018-02-19:
Hi nkolban,
￼the problem was corrected in esp-idf release 3.0.
This codes works for me:
void wifi_config(void) {
    tcpip_adapter_init();
    net_event_group = xEventGroupCreate();
    ip_addr_t dnsserver;
    tcpip_adapter_dns_info_t dns;

    if(!flash_param.dhcp) {
        tcpip_adapter_dhcpc_stop(TCPIP_ADAPTER_IF_STA);

        tcpip_adapter_ip_info_t ipInfo;
        inet_pton(AF_INET, (const char *)flash_param.static_ip, &ipInfo.ip);
        inet_pton(AF_INET, (const char *)flash_param.static_gateway, &ipInfo.gw);
        inet_pton(AF_INET, (const char *)flash_param.static_netmask, &ipInfo.netmask);
        tcpip_adapter_set_ip_info(TCPIP_ADAPTER_IF_STA, &ipInfo);

        inet_pton(AF_INET, (const char *)flash_param.static_dns1, &dnsserver);
        dns.ip = dnsserver;
        tcpip_adapter_set_dns_info(TCPIP_ADAPTER_IF_STA, TCPIP_ADAPTER_DNS_MAIN, &dns);

        inet_pton(AF_INET, (const char *)flash_param.static_dns2, &dnsserver);
        dns.ip = dnsserver;
        tcpip_adapter_set_dns_info(TCPIP_ADAPTER_IF_STA, TCPIP_ADAPTER_DNS_BACKUP, &dns);
    }

    ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    ESP_ERROR_CHECK( esp_wifi_set_storage(WIFI_STORAGE_RAM) );
    wifi_config_t wifi_config;
    memcpy(wifi_config.sta.ssid, flash_param.ssid, SSID_LEN);
    memcpy(wifi_config.sta.password, flash_param.passwd, PASSWD_LEN);
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config));
    ESP_ERROR_CHECK(esp_wifi_start());

    esp_wifi_set_ps(DEFAULT_PS_MODE);
    ESP_LOGI(WIFI_TAG, "wifi configuration and initialization OK\r\n");
}