espressif/esp-idf #420: FreeRTOS queue.c - No return in function returning non-void

ammaree, 2017-03-12:
BaseType_t xQueueGenericSend() and BaseType_t xQueueGenericReceive() both have a situation where the compiler complains, issuing warnings.

Not sure if specific to my environment although unlikely since I am using plain vanilla standard ESP-IDF and tools, no setting(s) changed that I am aware of.

Any suggestions?

ammaree, 2017-03-12:
Just checked the original/unmodified and latest FreeRTOS code, should have the same warning issued so problem must be in my setup?

nkolban, 2017-03-13:
What are the messages that are being issued?

projectgus, 2017-03-13:
Do you have an example of a function whose code triggers this warning?

ammaree, 2017-03-13:
@nkolban see clip below
[image]

ammaree, 2017-03-13:
@projectgus 
Actually, my sample code does not use any queues yet. It is based on the SNTP example modified to have 2  separate tasks
blink_task - blinking 2 LEDs and checking for GPIO/switch activity, 500mS cycle time
snmp_task - reading and displaying current time every 10 secs

Any queue activity/usage is as a result of queues defined in esp-idf components

igrr, 2017-03-13:
This (false positive) warning is generated by Eclipse, not by the toolchain. Please see the relevant bug report in Eclipse issue tracker:

https://bugs.eclipse.org/bugs/show_bug.cgi?id=377046

closed 2017-04-05 by igrr

kenmux, 2017-11-03:
I'd like to add return statement at the end of the function body. In this way this issue should be solved completely, no matter what IDE is.