espressif/esp-idf #421: BLE does not scan iBeacons

neilyoung, 2017-03-12:
Hi, 

I'm using a slightly modified version of this iBeacon tracker https://github.com/pcbreflux/espressif/tree/master/esp32/app/ESP32_blufi_beacon_tracker

The modification just covers one line in the initialization code, matching the latest changes in ESP-IDF.

Whereas the scanner does see nearly all BLE devices around, it has two remarkable problems with iBeacons:

1) It provides just a scattered scan of iBeacons, even though the beacons are there and advertising once ever 600 ms. The scanner just comes with one or two scans per minute

2) The RSSI readings are kind of nonsense. I'm having two consecutive readings of one and the same stationary iBeacon. One reads RSSI -27, the other RSSI -84. This is simply impossible.

The board is a Sparkfun ESPThing. 

Anybody having made  a better experience?

neilyoung, 2017-03-12:
Interesting turn in this matter: Up to now I just tested with RadBeacons (Dot). Now I used my iPhone as virtual Estimote beacon. And scans come in. The RSSI values are still nonsense (e.g. -8 very close to the beacon), but at least I receive scans...

heyinling, 2017-03-13:
Hi @neilyoung ,

regarding the RSSI issue, the result I read from `ESP_GAP_SEARCH_INQ_RES_EVT` is consistent. 

Here's the scan result I've got:

```
+BTSCAN:INQ,08:66:98:8c:ce:7e,NULL,-62,1a
+BTSCAN:INQ,24:0a:c4:03:f3:96,UGyJkawBIT,-48,02
+BTSCAN:INQ,7a:a4:60:dc:c8:54,NULL,-70,1a                                                                                                
+BTSCAN:INQ,7b:88:26:5f:1a:6c,NULL,-80,1a
+BTSCAN:INQ,7b:80:8e:e0:de:d7,NULL,-75,1a 
+BTSCAN:INQ,76:3c:0d:b1:51:0a,NULL,-77,1a
+BTSCAN:INQ,08:66:98:8c:ce:7e,NULL,-66,1a    
+BTSCAN:INQ,7b:88:26:5f:1a:6c,NULL,-79,1a     
+BTSCAN:INQ,08:66:98:8c:ce:7e,NULL,-69,1a    
+BTSCAN:INQ,7b:88:26:5f:1a:6c,NULL,-79,1a   
+BTSCAN:INQ,7b:80:8e:e0:de:d7,NULL,-68,1a      
+BTSCAN:INQ,7b:80:8e:e0:de:d7,NULL,-68,1a   
+BTSCAN:INQ,08:66:98:8c:ce:7e,NULL,-63,1a              
+BTSCAN:INQ,76:3c:0d:b1:51:0a,NULL,-76,1a               
+BTSCAN:INQ,08:66:98:8c:ce:7e,NULL,-66,1a            
+BTSCAN:INQ,7b:88:26:5f:1a:6c,NULL,-76,1a              
+BTSCAN:INQ,7b:80:8e:e0:de:d7,NULL,-80,1a    
+BTSCAN:INQ,08:66:98:8c:ce:7e,NULL,-67,1a 
+BTSCAN:INQ,76:3c:0d:b1:51:0a,NULL,-73,1a                 
+BTSCAN:INQ,7b:88:26:5f:1a:6c,NULL,-76,1a               
+BTSCAN:INQ,7b:80:8e:e0:de:d7,NULL,-79,1a              
+BTSCAN:INQ,08:66:98:8c:ce:7e,NULL,-73,1a          
+BTSCAN:INQ,76:3c:0d:b1:51:0a,NULL,-73,1a
+BTSCAN:INQ,7b:80:8e:e0:de:d7,NULL,-80,1a  
+BTSCAN:INQ,7a:a4:60:dc:c8:54,NULL,-67,1a
+BTSCAN:INQ,7b:88:26:5f:1a:6c,NULL,-84,1a             
+BTSCAN:INQ,7b:80:8e:e0:de:d7,NULL,-78,1a       
+BTSCAN:INQ,24:0a:c4:03:f3:96,UGyJkawBIT,-46,02    
+BTSCAN:INQ,7a:a4:60:dc:c8:54,NULL,-68,1a     
+BTSCAN:INQ,08:66:98:8c:ce:7e,NULL,-65,1a         
+BTSCAN:INQ,7b:80:8e:e0:de:d7,NULL,-78,1a   
+BTSCAN:INQ,08:66:98:8c:ce:7e,NULL,-60,1a            
+BTSCAN:INQ,76:3c:0d:b1:51:0a,NULL,-76,1a          
+BTSCAN:INQ,7b:88:26:5f:1a:6c,NULL,-83,1a  
+BTSCAN:INQ,24:0a:c4:03:f3:96,UGyJkawBIT,-44,02    
+BTSCAN:INQ,08:66:98:8c:ce:7e,NULL,-69,1a    
```

neilyoung, 2017-03-13:
@heyinling : Yes, in fact, this looks awesome and as it should be. What board? What beacons? What happens, if you move the board very close to the beacon? I'm reading things like "-8" and "-10" then, complete nonsense and impossible.

neilyoung, 2017-03-13:
Here is my latest result, scanning an Estimote Hardware beacon, which has a very weak battery. An Android sniffer shows approx -77 dBm while laying 5 cm  left of the beacon. The Sparkfun shows this (note the times: I would expect to get ways more callbacks, because the Estimote advertises every 500 ms)

I (3035) ble1: BDA: ea:19:6f:c1:48:39, RSSI -60
I (3645) ble1: BDA: ea:19:6f:c1:48:39, RSSI -61
I (5475) ble1: BDA: ea:19:6f:c1:48:39, RSSI -57
I (9775) ble1: BDA: ea:19:6f:c1:48:39, RSSI -58
I (12225) ble1: BDA: ea:19:6f:c1:48:39, RSSI -58
I (13445) ble1: BDA: ea:19:6f:c1:48:39, RSSI -58
I (14685) ble1: BDA: ea:19:6f:c1:48:39, RSSI -67
I (17745) ble1: BDA: ea:19:6f:c1:48:39, RSSI -61
I (18975) ble1: BDA: ea:19:6f:c1:48:39, RSSI -57
I (19585) ble1: BDA: ea:19:6f:c1:48:39, RSSI -57
I (23875) ble1: BDA: ea:19:6f:c1:48:39, RSSI -58
I (24485) ble1: BDA: ea:19:6f:c1:48:39, RSSI -58
I (25725) ble1: BDA: ea:19:6f:c1:48:39, RSSI -57
I (26335) ble1: BDA: ea:19:6f:c1:48:39, RSSI -57
I (26945) ble1: BDA: ea:19:6f:c1:48:39, RSSI -57
I (28795) ble1: BDA: ea:19:6f:c1:48:39, RSSI -70
I (30025) ble1: BDA: ea:19:6f:c1:48:39, RSSI -59
I (30635) ble1: BDA: ea:19:6f:c1:48:39, RSSI -58
I (32475) ble1: BDA: ea:19:6f:c1:48:39, RSSI -58
I (34925) ble1: BDA: ea:19:6f:c1:48:39, RSSI -64
I (35535) ble1: BDA: ea:19:6f:c1:48:39, RSSI -65
I (37995) ble1: BDA: ea:19:6f:c1:48:39, RSSI -58
I (39225) ble1: BDA: ea:19:6f:c1:48:39, RSSI -56
I (39845) ble1: BDA: ea:19:6f:c1:48:39, RSSI -57
I (41075) ble1: BDA: ea:19:6f:c1:48:39, RSSI -66
I (41685) ble1: BDA: ea:19:6f:c1:48:39, RSSI -62
I (43525) ble1: BDA: ea:19:6f:c1:48:39, RSSI -59
I (44145) ble1: BDA: ea:19:6f:c1:48:39, RSSI -60
I (45375) ble1: BDA: ea:19:6f:c1:48:39, RSSI -56
I (45985) ble1: BDA: ea:19:6f:c1:48:39, RSSI -55
I (47825) ble1: BDA: ea:19:6f:c1:48:39, RSSI -61
I (48445) ble1: BDA: ea:19:6f:c1:48:39, RSSI -61
I (49665) ble1: BDA: ea:19:6f:c1:48:39, RSSI -61
I (50285) ble1: BDA: ea:19:6f:c1:48:39, RSSI -62
I (52725) ble1: BDA: ea:19:6f:c1:48:39, RSSI -56
I (53335) ble1: BDA: ea:19:6f:c1:48:39, RSSI -55
I (55175) ble1: BDA: ea:19:6f:c1:48:39, RSSI -61
I (55785) ble1: BDA: ea:19:6f:c1:48:39, RSSI -60
I (57015) ble1: BDA: ea:19:6f:c1:48:39, RSSI -57
I (57635) ble1: BDA: ea:19:6f:c1:48:39, RSSI -57
I (58865) ble1: BDA: ea:19:6f:c1:48:39, RSSI -56
I (60095) ble1: BDA: ea:19:6f:c1:48:39, RSSI -56

You see, it is somehow constant, but ways too optimistic.

Then these are the results, if I let my iPhone make an Estimote vitrtual beacon. The readings are really following my movements (better if closer, worse if away), but what are these -8, -11 while having the Sparkfun laying on the iPhone? I have never seen such strong readings on every sniffer I used so far. This is simply physically impossible to be true. 

I (365) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -21
I (395) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -21
I (495) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -21
I (555) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -21
I (655) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -21
I (815) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -21
I (845) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -20
I (905) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -19
I (1165) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -20
I (1205) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -19
I (1265) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -21
I (1295) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -20
I (1305) ble1: BDA: ea:19:6f:c1:48:39, RSSI -46
I (1365) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -19
I (1395) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -20
I (1495) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -19
I (1555) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -21
I (1755) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -20
I (1815) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -17
I (1875) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -19
I (1915) ble1: BDA: ea:19:6f:c1:48:39, RSSI -51
[... 112 lines cut ...]
I (11245) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -31
I (11305) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -34
I (11405) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -30
I (11465) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -33
I (11555) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -38
I (11615) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -42
I (11705) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -46
I (11765) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -34
I (11865) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -46
I (11895) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -41
I (11995) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -42
I (12055) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -40
I (12155) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -43
I (12225) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -41
I (12315) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -37
I (12345) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -36
I (12355) ble1: BDA: ea:19:6f:c1:48:39, RSSI -67
I (12445) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -39
I (12505) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -36
I (12665) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -49
I (12765) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -32
I (12915) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -33
I (13045) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -34
I (13205) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -30
I (13395) ble1: BDA: 59:bb:83:e7:c6:dc, RSSI -33

neilyoung, 2017-03-13:
Maybe code sharing helps to come behind:

Here is the code:
```
#include <stdio.h>
#include <stdlib.h>
#include <string.h>

#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "bt.h"

#include "esp_blufi_api.h"
#include "esp_bt_defs.h"
#include "esp_gap_ble_api.h"
#include "esp_bt_main.h"
#include "esp_bt_device.h"

extern void bt_task(void *ignore);

static const char tag[] = "BLE";

static uint8_t ibeacon_prefix[] = {
	 0x02,0x01,0x00,0x1A,0xFF,0x4C,0x00,0x02,0x15
};

void gap_event_handler(esp_gap_ble_cb_event_t event, esp_ble_gap_cb_param_t *param) {
	esp_ble_gap_cb_param_t *p = (esp_ble_gap_cb_param_t *)param;

	if (p->scan_rst.search_evt == ESP_GAP_SEARCH_INQ_RES_EVT) {
		// Check for iBeacon adv prefix. Ignore 3rd byte, this varies from beacon type to beacon type
		p->scan_rst.ble_adv[2]  = 0x00;
		for (int i=0; i < sizeof(ibeacon_prefix); i++) {
			if (p->scan_rst.ble_adv[i] != ibeacon_prefix[i]) {
				return;
			}
		}
		ESP_LOGI(tag, "BDA: %02x:%02x:%02x:%02x:%02x:%02x, RSSI %d", 
			p->scan_rst.bda[0], 
			p->scan_rst.bda[1],
			p->scan_rst.bda[2], 
			p->scan_rst.bda[3], 
			p->scan_rst.bda[4],
			p->scan_rst.bda[5],  
			p->scan_rst.rssi);
	}
} 

void bt_task(void *ignore) {
	esp_err_t ret;

	esp_bt_controller_init();

	ret = esp_bt_controller_enable(ESP_BT_MODE_BTDM);
	if (ret) {
		ESP_LOGE(tag, "%s enable bt controller failed\n", __func__);
		goto end;
	}

	ret = esp_bluedroid_init();
	if (ret != ESP_OK) {
		ESP_LOGE(tag, "%s init bluedroid failed\n", __func__);
		goto end;
	}

	ret = esp_bluedroid_enable();
	if (ret) {
		ESP_LOGE(tag, "%s enable bluedroid failed\n", __func__);
		goto end;
	}
	ret = esp_ble_gap_register_callback(gap_event_handler);
	if (ret != ESP_OK) {
		ESP_LOGE(tag, "esp_ble_gap_register_callback: rc=%d", ret);
		goto end;
	}
	static esp_ble_scan_params_t ble_scan_params = {
		.scan_type              = BLE_SCAN_TYPE_ACTIVE,
		.own_addr_type          = ESP_PUBLIC_ADDR,
		.scan_filter_policy     = BLE_SCAN_FILTER_ALLOW_ALL,
		.scan_interval          = 0x50,
		.scan_window            = 0x30
	};
	ret = esp_ble_gap_set_scan_params(&ble_scan_params);
	if (ret != ESP_OK) {
		ESP_LOGE(tag, "esp_ble_gap_set_scan_params: rc=%d", ret);
		goto end;
	}

	ret = esp_ble_gap_start_scanning(60);
	if (ret != ESP_OK) {
		ESP_LOGE(tag, "esp_ble_gap_start_scanning: rc=%d", ret);
		goto end;
	}
	ESP_LOGI(tag, "Wait for scans...");
end:
	vTaskDelete(NULL);	
} 

void app_main(void) {
	xTaskCreatePinnedToCore(&bt_task, "btTask", 2048, NULL, 5, NULL, 0);
}
```

The code is scanning for 60 seconds and it is examining incoming traces in order to accept just iBeacon advertisements. I have tried to vary esp_ble_scan_params_t to no avail.

Some info regarding the environment:

 ```
(29) boot: ESP-IDF v2.0-rc1-257-g4745895 2nd stage bootloader
I (29) boot: compile time 02:20:42
I (56) boot: Enabling RNG early entropy source...
I (56) boot: SPI Speed      : 40MHz
I (56) boot: SPI Mode       : DIO
I (59) boot: SPI Flash Size : 4MB
I (69) boot: Partition Table:
I (77) boot: ## Label            Usage          Type ST Offset   Length
I (96) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (116) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (137) boot:  2 factory          factory app      00 00 00010000 00100000
I (157) boot: End of partition table

```

This code running on a Sparkfun ESP32Thing does

1) Just displaying scattered scans from an Estimote Hardware beacon with weak battery like so:

I (325) BLE: Wait for scans...
I (2745) BLE: BDA: ea:19:6f:c1:48:39, RSSI -60
I (4585) BLE: BDA: ea:19:6f:c1:48:39, RSSI -58
I (5205) BLE: BDA: ea:19:6f:c1:48:39, RSSI -60
I (13205) BLE: BDA: ea:19:6f:c1:48:39, RSSI -61
I (15045) BLE: BDA: ea:19:6f:c1:48:39, RSSI -67
I (15655) BLE: BDA: ea:19:6f:c1:48:39, RSSI -67

Looks OK, but frequency of scans results is too low and RSSI is too high.

2) Does show this with an iPhone running an virtual beacon Estimote beacon:

I (6445) BLE: BDA: 5a:4d:31:43:d2:21, RSSI -9
I (6515) BLE: BDA: 5a:4d:31:43:d2:21, RSSI -9
I (6545) BLE: BDA: 5a:4d:31:43:d2:21, RSSI -8
I (6605) BLE: BDA: 5a:4d:31:43:d2:21, RSSI -9
I (6635) BLE: BDA: 5a:4d:31:43:d2:21, RSSI -9
I (6695) BLE: BDA: 5a:4d:31:43:d2:21, RSSI -8
I (6735) BLE: BDA: 5a:4d:31:43:d2:21, RSSI -9
I (6795) BLE: BDA: 5a:4d:31:43:d2:21, RSSI -9
I (6865) BLE: BDA: 5a:4d:31:43:d2:21, RSSI -8
I (6895) BLE: BDA: 5a:4d:31:43:d2:21, RSSI -9
I (6955) BLE: BDA: 5a:4d:31:43:d2:21, RSSI -9
I (6985) BLE: BDA: 5a:4d:31:43:d2:21, RSSI -8
I (7045) BLE: BDA: ea:19:6f:c1:48:39, RSSI -59
I (7045) BLE: BDA: 5a:4d:31:43:d2:21, RSSI -9
I (7115) BLE: BDA: 5a:4d:31:43:d2:21, RSSI -9
I (7145) BLE: BDA: 5a:4d:31:43:d2:21, RSSI -8
I (7255) BLE: BDA: 5a:4d:31:43:d2:21, RSSI -9
I (7305) BLE: BDA: 5a:4d:31:43:d2:21, RSSI -8
I (7335) BLE: BDA: 5a:4d:31:43:d2:21, RSSI -9

Frequency is OK, IMHO, but RSSI is nonsense... The Sparkfun lies on top of the iPhone.

3) Does literally show NOTHING at all (OK, maybe one or two in a minute, if in luck), if scanning a RadBeacon Dot.

What am I doing wrong?

TianHao-Espressif, 2017-03-14:
.[0;31mE (531) BT: bta_dm_co_ble_load_local_keys: func not ported
.[0m
.[0;32mI (541) GATTC_DEMO: register callback
.[0m
.[0;32mI (541) GATTC_DEMO: EVT 0, gattc if 4
.[0m
.[0;32mI (541) GATTC_DEMO: REG_EVT
.[0m
.[0;32mI (551) GATTC_DEMO: EVT 0, gattc if 5
.[0m
.[0;32mI (551) GATTC_DEMO: REG_EVT
.[0m
BD_ADDR: .4e:0e:d3:9c:ab:5f:ADV_DATA: .02:01:1a:1a:ff:4c:00:02:15:bd:5f:80:21:33:31:49:ec:9e:df:9d:2d:62:ce:d7:70:00:01:00:01:c8:00:

BD_ADDR: .c6:0e:71:b1:d0:7a:ADV_DATA: .02:01:06:1a:ff:4c:00:02:15:fd:a5:06:93:a4:e2:4f:b1:af:cf:c6:eb:07:64:78:25:27:32:e6:08:c5:02:

BD_ADDR: .4e:0e:d3:9c:ab:5f:ADV_DATA: .02:01:1a:1a:ff:4c:00:02:15:bd:5f:80:21:33:31:49:ec:9e:df:9d:2d:62:ce:d7:70:00:01:00:01:c8:00:

BD_ADDR: .4e:0e:d3:9c:ab:5f:ADV_DATA: .02:01:1a:1a:ff:4c:00:02:15:bd:5f:80:21:33:31:49:ec:9e:df:9d:2d:62:ce:d7:70:00:01:00:01:c8:00:

BD_ADDR: .4e:0e:d3:9c:ab:5f:ADV_DATA: .02:01:1a:1a:ff:4c:00:02:15:bd:5f:80:21:33:31:49:ec:9e:df:9d:2d:62:ce:d7:70:00:01:00:01:c8:00:

BD_ADDR: .4e:0e:d3:9c:ab:5f:ADV_DATA: .02:01:1a:1a:ff:4c:00:02:15:bd:5f:80:21:33:31:49:ec:9e:df:9d:2d:62:ce:d7:70:00:01:00:01:c8:00:

BD_ADDR: .4e:0e:d3:9c:ab:5f:ADV_DATA: .02:01:1a:1a:ff:4c:00:02:15:bd:5f:80:21:33:31:49:ec:9e:df:9d:2d:62:ce:d7:70:00:01:00:01:c8:00:

BD_ADDR: .4e:0e:d3:9c:ab:5f:ADV_DATA: .02:01:1a:1a:ff:4c:00:02:15:bd:5f:80:21:33:31:49:ec:9e:df:9d:2d:62:ce:d7:70:00:01:00:01:c8:00:

Hi, this is iBeacon which esp32 received. I show the BD_ADDR and ADV_DATA.
This iBeacon is sent by "iBeacon Scanner" application of iPhone.

neilyoung, 2017-03-14:
@TianHao-Espressif Thanks for the trace. Unfortunately it does not show the timestamps and I can also not see the measured RSSI values. What kind of development board did you use? I can confirm, that virtual beacons (this is what you have produced with the iPhone app) work as best. But I have serious problems with several kind of hardware beacons. 

Would you mind to use my little code snippet above and re-produce your trace with this.

BTW: This is the trace the iBeacon Scanner app produces here:

I (23955) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -25
I (24035) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -33
I (24105) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -24
I (24135) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -24
I (24195) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -33
I (24225) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -26
I (24295) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -24
I (24355) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -32
I (24385) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -30
I (24555) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -32
I (24585) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -28
I (24655) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -32
I (24685) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -48
I (24745) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -39
I (24775) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -35
I (24825) BLE: BDA: d4:38:ed:41:67:8b, RSSI -54
I (24835) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -30
I (24935) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -27
I (24995) BLE: BDA: c1:3f:fc:fb:03:e5, RSSI -57
I (24995) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -22
I (25035) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -25
I (25095) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -27
I (25125) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -25
I (25195) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -30
I (25255) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -29
I (25285) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -30
I (25385) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -31
I (25425) BLE: BDA: d4:38:ed:41:67:8b, RSSI -60
I (25445) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -30
I (25595) BLE: BDA: c1:3f:fc:fb:03:e5, RSSI -56
I (26195) BLE: BDA: c1:3f:fc:fb:03:e5, RSSI -57
I (26805) BLE: BDA: c1:3f:fc:fb:03:e5, RSSI -58
I (27405) BLE: BDA: c1:3f:fc:fb:03:e5, RSSI -59
I (29605) BLE: BDA: d4:38:ed:41:67:8b, RSSI -55

The virtual beacon produces scans starting with 43:8a...

That looks pretty OK. But if you move the ESP32 very close to the iPhone you read things like this:

I (16635) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -8
I (16705) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -8
I (16725) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -7
I (16795) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -8
I (16825) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -7
I (16885) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -7
I (16985) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -32
I (17045) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -8
I (17145) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -7
I (17175) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -7
I (17235) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -7
I (17335) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -7
I (17435) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -7
I (17495) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -7
I (17535) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -7
I (17625) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -7
I (17685) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -7
I (17755) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -7
I (17785) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -7
I (17845) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -8
I (17875) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -7
I (17975) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -8
I (18035) BLE: BDA: 49:8a:fb:27:92:3a, RSSI -7

Here are several things to be queried: What is the -32 freakout (BTW I would treat this to be the only valid value). And what are these -7, -8 values? You cannot achieve such reception values, even if you glue both devices togehter.

So, yes, virtual beacons work OK, real beacons NOT (with degrees of NOT: RadBeacaon pretty much NOT AT ALL, Estimotes a bit better, but still far away from ordinary scanners)

Please continue the investigations on this issue. Right now the ESP32 on a Sparkfun cannot be used properly for iBeacon scanning. This might be caused by ESP, ESP-IDF or bad Sparkfun desgin. I don't know yet.

TIA.

heyinling, 2017-03-15:
HI @neilyoung ,

for the first issue, adv not come as much as expected, could you please do test with the following scan parameters?
```c
esp_ble_scan_params_t ble_scan_param = {
    .scan_type              = BLE_SCAN_TYPE_PASSIVE,
    .own_addr_type          = ESP_PUBLIC_ADDR,
    .scan_filter_policy     = BLE_SCAN_FILTER_ALLOW_ALL,
    .scan_interval          = 0x640,
    .scan_window            = 0x640
};
```

the adv of ibeacon should come at least once 3 seconds.

neilyoung, 2017-03-15:
Hi @heyinling 

thanks for chiming in again. In fact, something like this was my last attempt last night (to no avail):

```
static esp_ble_scan_params_t ble_scan_params = {
		.scan_type              = BLE_SCAN_TYPE_PASSIVE,
		.own_addr_type          = ESP_PUBLIC_ADDR,
		.scan_filter_policy     = BLE_SCAN_FILTER_ALLOW_ALL,
		.scan_interval          = 10000,
		.scan_window            = 10000
};

```

Now I tried your's with one RadBeacon alive. This is the result:

```
 I (305) BLE: Wait for scans...
I (5545) BLE: BDA 0C:F3:EE:04:1E:AE, RSSI -54, UUID 2F234454-CF6D-4A0F-ADF2-F4911BA9FFA6 MAJ 2, MIN 0, TXPWR -59
I (59695) BLE: BDA 0C:F3:EE:04:1E:AE, RSSI -64, UUID 2F234454-CF6D-4A0F-ADF2-F4911BA9FFA6 MAJ 2, MIN 0, TXPWR -59
E (60305) BT: btc_search_callback  BLE observe complete. Num Resp 6

```

A second attempt with an additional hardware Estimote beacons gave this (an improvement I would say), but RadBeacon did still not appear:

```
I (305) BLE: Wait for scans...
I (515) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -36, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (1115) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -34, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (1705) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -35, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (2925) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -33, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (3505) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -28, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (4115) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -30, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (4705) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -28, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (5315) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -28, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (5905) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -51, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (6525) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -29, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (7135) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -32, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (7705) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -28, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (8975) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -38, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (9525) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -31, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (10115) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -31, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (10715) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -54, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (11315) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -28, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (11925) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -39, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (12545) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -31, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (13125) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -31, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (13735) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -27, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (14435) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -36, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (14945) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -38, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (15545) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -31, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (16185) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -28, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (16855) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -27, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (17365) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -37, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (18095) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -38, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (18575) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -31, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (19175) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -31, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (19775) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -54, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60
I (20395) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -37, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D MAJ 997, MIN 64763, TXPWR -60

```
Pretty good I would say. Not that precise I'm used to see it with an iPhone or an Android device, but at lest it is there once per second minimum.

PS: I have updated my code sample above to match this trace output.

heyinling, 2017-03-15:
HI @neilyoung ,

according to BLE Spec, controller will do scan in 3 channels one by one. it will stay on each channel for scan interval (scan window <= scan interval, means how long controller do scan within scan interval). 

you may try with different scan interval to get better performance. and for your scenario you always need to set scan window = scan interval.

neilyoung, 2017-03-16:
Yes, I did some experiments already, with values varying from the lowest (0x10) up to 10000. The issue is: I does not reveal my RadBeacons. Anybody having such a beacon could run the test script above? Would be great.

neilyoung, 2017-03-17:
@TianHao-Espressif Any other suggestions to try? I'm wondering why this issue doesn't have more attention at all. For me the ESP32 is NOT a good choice if it comes to iBeacon scanning. Not only that the detection rate is low for some real successful hardware beacons, some of them are pretty much undiscoverable. Then as an additional problem there are the nonsense RSSI readings.

TianHao-Espressif, 2017-03-17:
@neilyoung , the RSSI problem have been resolved today. But it will be in ESP-IDF several days later.
And another thing, could provide the purchase URL to buy your RadBeacon, because I have seen multi products of RadBeacon. Thanks.

neilyoung, 2017-03-17:
@TianHao-Espressif Oh, good news. Would you mind to ping me if available?  I purchased my RadBeacons here https://store.radiusnetworks.com/products/radbeacon-dot

neilyoung, 2017-03-22:
The fix is still not available in the most recent version. Any reason for that?

TianHao-Espressif, 2017-03-24:
@neilyoung , it will in the master branch later on.

neilyoung, 2017-03-24:
Hehe, the question is just: When... Have a nice weekend

neilyoung, 2017-03-27:
@TianHao-Espressif Well, I see several changes.

1. The exorbitant high values are not there anymore (-7, -8)
2. The variance of consecutive readings is insanely high: How can a stationary device, laying on top of the ESP32, produce this sequence?

```
I (1497) BLE: BDA 62:ED:BA:91:56:1E, RSSI -39, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (1567) BLE: BDA 62:ED:BA:91:56:1E, RSSI -96, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (1617) BLE: BDA 62:ED:BA:91:56:1E, RSSI -40, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56

```
-39, -96, -40 in a row?

3. I caught an RSSI of -102. Never seen before.
```
I (53837) BLE: BDA 62:ED:BA:91:56:1E, RSSI -102, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
```

`All in all my verdict is - doesn't work for me. 

Below a 60 s scan result, just one virtual beacon (iphone) placed on top of the ESP32, 

Setup is:
```
   static esp_ble_scan_params_t ble_scan_params = {
        .scan_type = BLE_SCAN_TYPE_PASSIVE,
        .own_addr_type = ESP_PUBLIC_ADDR,
        .scan_filter_policy = BLE_SCAN_FILTER_ALLOW_ALL,
        .scan_interval = 0x50,
        .scan_window = 0x30};

```

```

I (417) BLE: Wait for scans...
I (467) BLE: BDA 62:ED:BA:91:56:1E, RSSI -37, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (507) BLE: BDA 62:ED:BA:91:56:1E, RSSI -41, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (567) BLE: BDA 62:ED:BA:91:56:1E, RSSI -43, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (617) BLE: BDA 62:ED:BA:91:56:1E, RSSI -26, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (717) BLE: BDA 62:ED:BA:91:56:1E, RSSI -68, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (867) BLE: BDA 62:ED:BA:91:56:1E, RSSI -26, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (917) BLE: BDA 62:ED:BA:91:56:1E, RSSI -26, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (967) BLE: BDA 62:ED:BA:91:56:1E, RSSI -27, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (1017) BLE: BDA 62:ED:BA:91:56:1E, RSSI -30, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (1067) BLE: BDA 62:ED:BA:91:56:1E, RSSI -26, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (1167) BLE: BDA 62:ED:BA:91:56:1E, RSSI -41, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (1267) BLE: BDA 62:ED:BA:91:56:1E, RSSI -27, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (1307) BLE: BDA 62:ED:BA:91:56:1E, RSSI -63, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (1357) BLE: BDA 62:ED:BA:91:56:1E, RSSI -44, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (1417) BLE: BDA 62:ED:BA:91:56:1E, RSSI -51, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (1467) BLE: BDA 62:ED:BA:91:56:1E, RSSI -45, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (1497) BLE: BDA 62:ED:BA:91:56:1E, RSSI -39, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (1567) BLE: BDA 62:ED:BA:91:56:1E, RSSI -96, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (1617) BLE: BDA 62:ED:BA:91:56:1E, RSSI -40, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (1667) BLE: BDA 62:ED:BA:91:56:1E, RSSI -88, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (1767) BLE: BDA 62:ED:BA:91:56:1E, RSSI -43, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (1807) BLE: BDA 62:ED:BA:91:56:1E, RSSI -66, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (1917) BLE: BDA 62:ED:BA:91:56:1E, RSSI -51, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (1957) BLE: BDA 62:ED:BA:91:56:1E, RSSI -26, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (2017) BLE: BDA 62:ED:BA:91:56:1E, RSSI -27, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (2057) BLE: BDA 62:ED:BA:91:56:1E, RSSI -26, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (2097) BLE: BDA 62:ED:BA:91:56:1E, RSSI -26, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (2217) BLE: BDA 62:ED:BA:91:56:1E, RSSI -26, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (2257) BLE: BDA 62:ED:BA:91:56:1E, RSSI -37, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (2367) BLE: BDA 62:ED:BA:91:56:1E, RSSI -95, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (2397) BLE: BDA 62:ED:BA:91:56:1E, RSSI -39, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (2517) BLE: BDA 62:ED:BA:91:56:1E, RSSI -43, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (2547) BLE: BDA 62:ED:BA:91:56:1E, RSSI -39, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (2667) BLE: BDA 62:ED:BA:91:56:1E, RSSI -26, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
[... 654 lines cut ...]
[... cut ...]
I (60167) BLE: BDA 62:ED:BA:91:56:1E, RSSI -26, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (60247) BLE: BDA 62:ED:BA:91:56:1E, RSSI -42, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (60317) BLE: BDA 62:ED:BA:91:56:1E, RSSI -26, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
I (60407) BLE: BDA 62:ED:BA:91:56:1E, RSSI -25, UUID 8492E75F-4FD6-469D-B132-043FE94921D8, MAJ 9044, MIN 20493, TXPWR -56
E (60427) BT: btc_search_callback  BLE observe complete. Num Resp 6

```

jkhax0r, 2017-04-03:
You should check out #475.  Sounds like it's probably the same issue or very closely related.  Cannot get any scan results when doing an active scan.

@TianHao-Espressif I'd be happy to send my "bad" device and "good" if you can look into this so you can hopefully repeat it.

neilyoung, 2017-04-06:
@TianHao-Espressif Any news on the RadBeacons? And what is your comment regarding my verdict, that your fix didn't entirely remove the problem?

neilyoung, 2017-04-07:
OK, I will let this chip aside. If I scan through the tons of mails I receive a day, talking about problems over problems with nearly every part of software and hardware, that is not really convincing. Looking for alternatives then.

Best

mstewart-triva, 2017-05-04:
Trying to run this example blufi_beacon_tracker but getting error with function esp_bt_controller_init().

it is saying too few arguments.  Anybody know how to fix this?

johnnorbury, 2017-05-04:
Have you recently updated the idf?

If so, try with:

```
    esp_err_t ret;

    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    ret = esp_bt_controller_init(&bt_cfg);
    
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s initialize controller failed\n", __func__);
        return;
    }
```

mstewart-triva, 2017-05-04:
it compiled, but terminal still saying BT: Controller not initialized.

I am new to the esp32, where do you find this information john?

johnnorbury, 2017-05-04:
The code snippet was from the IDF blufi example (esp-idf/examples/bluetooth/blufi/).

The beacon example from @pcbreflux was created some time ago and due to the rapid addition of new features some functions can be outdated. 

I guess you are running this code:
```

void app_main()
{
    esp_err_t ret;

    nvs_flash_init();
    initialise_wifi();

    esp_bt_controller_init();

    ret = esp_bluedroid_init();
    if (ret) {
        BLUFI_ERROR("%s init bluedroid failed\n", __func__);
        return;
    }

    ret = esp_bluedroid_enable();
    if (ret) {
        BLUFI_ERROR("%s enable bluedroid failed\n", __func__);
        return;
    }

  ...
}
```

But with the blufi example code from the IDF we have:

```
void app_main()
{
    esp_err_t ret;

    ESP_ERROR_CHECK( nvs_flash_init() );
    initialise_wifi();

    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    ret = esp_bt_controller_init(&bt_cfg);
    if (ret) {
        BLUFI_ERROR("%s initialize bt controller failed\n", __func__);
    }

    ret = esp_bt_controller_enable(ESP_BT_MODE_BTDM);
    if (ret) {
        BLUFI_ERROR("%s enable bt controller failed\n", __func__);
        return;
    }

  ...
}
```

It also looks like 

> esp_err_t esp_bt_controller_enable(esp_bt_mode_t mode)

now has a parameter, you can see in 

> /esp-idf/components/bt/include/bt.h

The different possible inputs:
```
/**
 * @brief Bluetooth mode for controller enable/disable
 */
typedef enum {
    ESP_BT_MODE_IDLE       = 0x00,   /*!< Bluetooth is not running */
    ESP_BT_MODE_BLE        = 0x01,   /*!< Run BLE mode */
    ESP_BT_MODE_CLASSIC_BT = 0x02,   /*!< Run Classic BT mode */
    ESP_BT_MODE_BTDM       = 0x03,   /*!< Run dual mode */
} esp_bt_mode_t;
```

However, your terminal is specifying the controller isn't "Initialised" in which case a debug message should have popped up from:

`ESP_LOGE(GATTS_TABLE_TAG, "%s enable controller failed\n", __func__);`

TianHao-Espressif, 2017-05-05:
@neilyoung , I'm sorry for delay.  We have tested the RadBeacon devices. It might have something problem on Radbeacon device RF freq deviation. We will debug this.

negativekelvin, 2017-05-05:
This EM9301 chipset maybe not widely used so interesting to see what the issue is.

mstewart-triva, 2017-05-05:
@neilyoung could you post your project for review or on github

jkhax0r, 2017-05-05:
@TianHao-Espressif I'm glad to hear you guys are working on this!  I think this has been at the core of a lot of my problems too and I long suspected it to be RF frequency deviation but never 100% positive!  I even bought a scanner (Ubertooth one) to try to look at frequencies and data sent on individual channels, but it wasn't worth the hassle.  I also bought new ESP-WROOM-32 modules with the FCC logo thinking maybe the ones I populated on my PCBs (w/o the same shield) might be older and mistuned crystals or something.

neilyoung, 2017-05-05:
@mstewart-triva Well, the entire "project" is here on thread. Look for my posting from March 13th
It is just one file. Take whatever esp-idf sample and replace main.cpp by that.

neilyoung, 2017-05-05:
@mstewart-triva OK, updated version here

I think just some init functions have changed. This is the state from the day I left it aside.

```

#include <stdio.h>
#include <stdlib.h>
#include <string.h>

#include "bt.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "freertos/FreeRTOS.h"
#include "freertos/event_groups.h"
#include "freertos/task.h"
#include "nvs_flash.h"
#include "sdkconfig.h"

#include "esp_blufi_api.h"
#include "esp_bt_defs.h"
#include "esp_bt_device.h"
#include "esp_bt_main.h"
#include "esp_gap_ble_api.h"

extern void bt_task(void *ignore);

static const char tag[] = "BLE";

static uint8_t ibeacon_prefix[] = {
    0x02, 0x01, 0x00, 0x1A, 0xFF, 0x4C, 0x00, 0x02, 0x15};

void gap_event_handler(esp_gap_ble_cb_event_t event, esp_ble_gap_cb_param_t *param) {
    esp_ble_gap_cb_param_t *p = (esp_ble_gap_cb_param_t *)param;

    if (p->scan_rst.search_evt == ESP_GAP_SEARCH_INQ_RES_EVT) {
        // Check for iBeacon adv prefix. Ignore 3rd byte, this varies from beacon type to beacon type
        p->scan_rst.ble_adv[2] = 0x00;
        for (int i = 0; i < sizeof(ibeacon_prefix); i++) {
            if (p->scan_rst.ble_adv[i] != ibeacon_prefix[i]) {
                return;
            }
        }

        ESP_LOGI(tag, "BDA %02X:%02X:%02X:%02X:%02X:%02X, RSSI %d, UUID %02X%02X%02X%02X-%02X%02X-%02X%02X-%02X%02X-%02X%02X%02X%02X%02X%02X, MAJ %d, MIN %d, TXPWR %d",
                 p->scan_rst.bda[0],
                 p->scan_rst.bda[1],
                 p->scan_rst.bda[2],
                 p->scan_rst.bda[3],
                 p->scan_rst.bda[4],
                 p->scan_rst.bda[5],

                 p->scan_rst.rssi,

                 p->scan_rst.ble_adv[9],
                 p->scan_rst.ble_adv[10],
                 p->scan_rst.ble_adv[11],
                 p->scan_rst.ble_adv[12],

                 p->scan_rst.ble_adv[13],
                 p->scan_rst.ble_adv[14],

                 p->scan_rst.ble_adv[15],
                 p->scan_rst.ble_adv[16],

                 p->scan_rst.ble_adv[17],
                 p->scan_rst.ble_adv[18],

                 p->scan_rst.ble_adv[19],
                 p->scan_rst.ble_adv[20],
                 p->scan_rst.ble_adv[21],
                 p->scan_rst.ble_adv[22],
                 p->scan_rst.ble_adv[23],
                 p->scan_rst.ble_adv[24],

                 (p->scan_rst.ble_adv[25] << 8) | p->scan_rst.ble_adv[26],

                 (p->scan_rst.ble_adv[27] << 8) | p->scan_rst.ble_adv[28],

                 p->scan_rst.ble_adv[29] - 256

                 );
    }
}

void bt_task(void *ignore) {
    esp_err_t ret;
    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    esp_bt_controller_init(&bt_cfg);

    ret = esp_bt_controller_enable(ESP_BT_MODE_BTDM);
    if (ret) {
        ESP_LOGE(tag, "%s enable bt controller failed\n", __func__);
        goto end;
    }

    ret = esp_bluedroid_init();
    if (ret != ESP_OK) {
        ESP_LOGE(tag, "%s init bluedroid failed\n", __func__);
        goto end;
    }

    ret = esp_bluedroid_enable();
    if (ret) {
        ESP_LOGE(tag, "%s enable bluedroid failed\n", __func__);
        goto end;
    }
    ret = esp_ble_gap_register_callback(gap_event_handler);
    if (ret != ESP_OK) {
        ESP_LOGE(tag, "esp_ble_gap_register_callback: rc=%d", ret);
        goto end;
    }

    static esp_ble_scan_params_t ble_scan_params = {
        .scan_type = BLE_SCAN_TYPE_PASSIVE,
        .own_addr_type = ESP_PUBLIC_ADDR,
        .scan_filter_policy = BLE_SCAN_FILTER_ALLOW_ALL,
        .scan_interval = 0x50,
        .scan_window =  0x30};

    ret = esp_ble_gap_set_scan_params(&ble_scan_params);
    if (ret != ESP_OK) {
        ESP_LOGE(tag, "esp_ble_gap_set_scan_params: rc=%d", ret);
        goto end;
    }

    ret = esp_ble_gap_start_scanning(60);
    if (ret != ESP_OK) {
        ESP_LOGE(tag, "esp_ble_gap_start_scanning: rc=%d", ret);
        goto end;
    }
    ESP_LOGI(tag, "Wait for scans...");
end:
    vTaskDelete(NULL);
}

void app_main(void) {
	nvs_flash_init();
    xTaskCreatePinnedToCore(&bt_task, "btTask", 2048, NULL, 5, NULL, 0);
}
```

neilyoung, 2017-05-05:
@mstewart-triva OK, please give it a try https://github.com/neilyoung/ESP32_blufi_beacon_tracker/tree/master

mstewart-triva, 2017-05-08:
I tried it, Neil, but am getting no beacon readings in my terminal.  It is just showing some intitial set up.

mstewart-triva, 2017-05-08:
For some reason I don't think my bluetooth is being enabled, you know why this could be?

mstewart-triva, 2017-05-08:
All I'm getting in output is as follows:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:3412
load:0x40078000,len:10388
load:0x40080000,len:252
entry 0x40080034
I (761) cpu_start: Pro cpu up.
I (762) cpu_start: Single core mode
I (763) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (775) heap_alloc_caps: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (795) heap_alloc_caps: At 3FFC6E20 len 000191E0 (100 KiB): DRAM
I (816) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (837) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (859) heap_alloc_caps: At 40092B50 len 0000D4B0 (53 KiB): IRAM
I (879) cpu_start: Pro cpu start user code
I (938) cpu_start: Starting scheduler on PRO CPU.
I (994) wifi: wifi firmware version: ba7c292
I (994) wifi: config NVS flash: enabled
I (994) wifi: config nano formating: disabled
I (1010) wifi: Init dynamic tx buffer num: 32
I (1010) wifi: Init dynamic rx buffer num: 64
I (1010) wifi: wifi driver task: 3ffcd494, prio:23, stack:3584
I (1013) wifi: Init static rx buffer num: 10
I (1016) wifi: Init dynamic rx buffer num: 64
I (1021) wifi: Init rx ampdu len mblock:7
I (1024) wifi: Init lldesc rx ampdu entry mblock:4
I (1029) wifi: wifi power manager task: 0x3ffd284c prio: 21 stack: 2560
I (1036) wifi: wifi timer task: 3ffd38dc, prio:22, stack:3584
I (1056) phy: phy_version: 350, Mar 22 2017, 15:02:06, 1, 0
I (1057) wifi: mode : sta (24:0a:c4:00:9f:0a)

t4r4n15, 2017-05-23:
Same issue here, with code from examples on WROOM-32 we have many lost packets. We were changing esp_ble_scan_params_t (scan_interval and scan_window) and things are really messed up (most of the time it works for about 15 seconds and crashes or stops for 2 minutes). Best stability is for default params.

TianHao-Espressif, 2017-06-08:
@neilyoung ， the issues about Radbeacon have been resolved on newest master branch. So did you have re-test it?

neilyoung, 2017-06-08:
@TianHao-Espressif Well, I'm not too crazy about the results. Having one Estimote and five Radbeacons here, and the single Estimote is by far more often scanned, than the others.

Given this is still a valid scanner

https://github.com/neilyoung/ESP32_blufi_beacon_tracker

negativekelvin, 2017-06-09:
@TianHao-Espressif what was the issue with EM9301 chipset?

TianHao-Espressif, 2017-06-13:
@negativekelvin , The phy library have updated.

FayeY, 2017-06-19:
Hi @neilyoung , is this problem solved? Can we close it?

neilyoung, 2017-06-19:
@FayeY Clearly NO. 

Not even that the RadBeacons (I have five dots here among one Estimote) are not scanned as much as often than the Estimote, I think I got two WDT crashes during 1 minute of scans. The scans came in very scattered and got stuck, while the WDT expired.

Setup is
```

    static esp_ble_scan_params_t ble_scan_params = {
        .scan_type = BLE_SCAN_TYPE_PASSIVE,
        .own_addr_type = BLE_ADDR_TYPE_PUBLIC,
        .scan_filter_policy = BLE_SCAN_FILTER_ALLOW_ALL,
        .scan_interval = 0x10,
        .scan_window =  0x10};

```

but I tried other values as well. Trace attached.

[make flash monitor.txt](https://github.com/espressif/esp-idf/files/1084010/make.flash.monitor.txt)

neilyoung, 2017-07-17:
@FayeY @TianHao-Espressif Hi guys, not sure, what's going on, but after a long time I tried it again. My code is still here (unchanged) https://github.com/neilyoung/ESP32_blufi_beacon_tracker, but the results are meanwhile sub zero. 

Now not even Estimote is received correctly, not to talk about the RadBeacon problem. 

```
I (3125) cpu_start: Pro cpu start user code
I (3277) cpu_start: Starting scheduler on PRO CPU.
I (486) cpu_start: Starting scheduler on APP CPU.
I (506) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (686) phy: phy_version: 355.1, 59464c5, Jun 14 2017, 20:25:06, 1, 0
I (726) BLE: Wait for scans...
I (6776) BLE: BDA 0C:F3:EE:04:21:38, RSSI -75, UUID 2F234454-CF6D-4A0F-ADF2-F4911BA9FFA6, MAJ 0, MIN 1, TXPWR -96
I (9436) BLE: BDA 0C:F3:EE:04:21:38, RSSI -76, UUID 2F234454-CF6D-4A0F-ADF2-F4911BA9FFA6, MAJ 0, MIN 1, TXPWR -96
I (15756) BLE: BDA 0C:F3:EE:04:21:38, RSSI -81, UUID 2F234454-CF6D-4A0F-ADF2-F4911BA9FFA6, MAJ 0, MIN 1, TXPWR -96
I (17466) BLE: BDA 0C:F3:EE:04:1E:8F, RSSI -73, UUID 2F234454-CF6D-4A0F-ADF2-F4911BA9FFA6, MAJ 1, MIN 0, TXPWR -70
I (22086) BLE: BDA 0C:F3:EE:04:21:38, RSSI -80, UUID 2F234454-CF6D-4A0F-ADF2-F4911BA9FFA6, MAJ 0, MIN 1, TXPWR -96
I (23176) BLE: BDA 0C:F3:EE:04:1E:8F, RSSI -70, UUID 2F234454-CF6D-4A0F-ADF2-F4911BA9FFA6, MAJ 1, MIN 0, TXPWR -70
I (25716) BLE: BDA 0C:F3:EE:04:21:38, RSSI -79, UUID 2F234454-CF6D-4A0F-ADF2-F4911BA9FFA6, MAJ 0, MIN 1, TXPWR -96
I (26706) BLE: BDA 0C:F3:EE:04:21:38, RSSI -81, UUID 2F234454-CF6D-4A0F-ADF2-F4911BA9FFA6, MAJ 0, MIN 1, TXPWR -96
I (54266) BLE: BDA 0C:F3:EE:04:1E:8F, RSSI -69, UUID 2F234454-CF6D-4A0F-ADF2-F4911BA9FFA6, MAJ 1, MIN 0, TXPWR -70
E (60726) BT: btc_search_callback  BLE observe complete. Num Resp 23

```

Have you abandoned this meanwhile?

neilyoung, 2017-07-19:
BTW: I'm having here now 30 iBeacons from AprilBrothers, a Chinese company. And you know, what? My app scans like hell.

All beacons advertise with full power and 500 ms interval.

Do you have an explanation, why you're having problems with non-Chinese beacons :) ?

Sorry for the long post. It should be made possible to fold/unfold those long quotes...

```
I (0) cpu_start: App cpu up.
I (2822) heap_init: Initializing. RAM available for dynamic allocation:
I (2878) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (2926) heap_init: At 3FFCD998 len 00012668 (73 KiB): DRAM
I (2975) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (3025) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (3077) heap_init: At 40094604 len 0000B9FC (46 KiB): IRAM
I (3126) cpu_start: Pro cpu start user code
I (3277) cpu_start: Starting scheduler on PRO CPU.
I (486) cpu_start: Starting scheduler on APP CPU.
I (506) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (686) phy: phy_version: 355.1, 59464c5, Jun 14 2017, 20:25:06, 1, 0
I (736) BLE: Wait for scans...
I (796) BLE: BDA 12:3B:6A:1A:ED:BA, RSSI -79, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 1, TXPWR -59
I (806) BLE: BDA 12:3B:6A:1A:EF:E6, RSSI -69, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 59119, TXPWR -59
I (816) BLE: BDA 12:3B:6A:1A:EE:11, RSSI -60, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 4590, TXPWR -59
I (866) BLE: BDA 12:3B:6A:1A:ED:FF, RSSI -64, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 65517, TXPWR -59
I (1006) BLE: BDA 12:3B:6A:1A:EF:31, RSSI -58, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 12783, TXPWR -59
I (1016) BLE: BDA 12:3B:6A:1A:EF:DC, RSSI -63, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 8, TXPWR -59
I (1056) BLE: BDA 12:3B:6A:1A:EE:3D, RSSI -60, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 15854, TXPWR -59
I (1056) BLE: BDA 12:3B:6A:1A:EE:08, RSSI -62, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 10, TXPWR -59
I (1106) BLE: BDA 12:3B:6A:1A:ED:A5, RSSI -63, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 42477, TXPWR -59
I (1116) BLE: BDA 12:3B:6A:1A:EE:66, RSSI -63, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 26350, TXPWR -59
I (1126) BLE: BDA 12:3B:6A:1A:EE:4C, RSSI -72, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 19694, TXPWR -59
I (1166) BLE: BDA 12:3B:6A:1A:ED:AD, RSSI -63, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 44525, TXPWR -59
I (1176) BLE: BDA 12:3B:6A:1A:ED:C9, RSSI -65, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 51693, TXPWR -59
I (1186) BLE: BDA 12:3B:6A:1A:ED:D0, RSSI -63, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 4, TXPWR -59
I (1196) BLE: BDA 12:3B:6A:1A:ED:AB, RSSI -66, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 44013, TXPWR -59
I (1236) BLE: BDA 12:3B:6A:1A:EE:34, RSSI -67, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 5, TXPWR -59
I (1366) BLE: BDA 12:3B:6A:1A:ED:FF, RSSI -58, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 65517, TXPWR -59
I (1376) BLE: BDA 12:3B:6A:1A:EE:9C, RSSI -62, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 3, TXPWR -59
I (1496) BLE: BDA 12:3B:6A:1A:F0:0D, RSSI -60, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 3568, TXPWR -59
I (1506) BLE: BDA 12:3B:6A:1A:EF:31, RSSI -70, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 12783, TXPWR -59
I (1516) BLE: BDA 12:3B:6A:1A:EF:2C, RSSI -70, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 7, TXPWR -59
I (1556) BLE: BDA 12:3B:6A:1A:EE:3D, RSSI -64, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 15854, TXPWR -59
I (1556) BLE: BDA 12:3B:6A:1A:EF:76, RSSI -64, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 30447, TXPWR -59
I (1566) BLE: BDA 12:3B:6A:1A:EE:08, RSSI -62, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 10, TXPWR -59
I (1576) BLE: BDA 12:3B:6A:1A:EF:79, RSSI -75, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 31215, TXPWR -59
I (1626) BLE: BDA 12:3B:6A:1A:EE:4C, RSSI -70, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 19694, TXPWR -59
I (1676) BLE: BDA 12:3B:6A:1A:ED:AD, RSSI -62, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 44525, TXPWR -59
I (1686) BLE: BDA 12:3B:6A:1A:ED:D0, RSSI -60, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 4, TXPWR -59
I (1706) BLE: BDA 12:3B:6A:1A:ED:AB, RSSI -65, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 44013, TXPWR -59
I (1736) BLE: BDA 12:3B:6A:1A:EE:34, RSSI -65, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 5, TXPWR -59
I (1816) BLE: BDA 12:3B:6A:1A:EF:E6, RSSI -64, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 59119, TXPWR -59
I (1876) BLE: BDA 12:3B:6A:1A:EE:9C, RSSI -67, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 3, TXPWR -59
I (1926) BLE: BDA 12:3B:6A:1A:ED:A0, RSSI -60, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 41197, TXPWR -59
I (1966) BLE: BDA 12:3B:6A:1A:EF:3D, RSSI -62, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 2, TXPWR -59
I (2006) BLE: BDA 12:3B:6A:1A:F0:0D, RSSI -60, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 3568, TXPWR -59
I (2016) BLE: BDA 12:3B:6A:1A:EF:2C, RSSI -64, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 7, TXPWR -59
I (2066) BLE: BDA 12:3B:6A:1A:EF:76, RSSI -64, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 30447, TXPWR -59
[... cut ...]
I (56976) BLE: BDA 12:3B:6A:1A:ED:A0, RSSI -60, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 41197, TXPWR -59
I (57006) BLE: BDA 12:3B:6A:1A:EF:3D, RSSI -60, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 2, TXPWR -59
I (57056) BLE: BDA 12:3B:6A:1A:EF:31, RSSI -63, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 12783, TXPWR -59
I (57056) BLE: BDA 12:3B:6A:1A:F0:0D, RSSI -60, UUID B5B182C7-EAB1-4988-AA99-B5C1517008D9, MAJ 1, MIN 3568, TXPWR -59
I (57116) BLE: BDA 12:3B:6A:1A:EF:2C,

TianHao-Espressif, 2017-07-19:
@neilyoung , you means ibeacon of AprilBrothers can be scanned by esp32, but RadBeacon and Estimote can not be scanned by esp32, right?

We're doing more test and debug on these similar things.

igrr, 2017-07-19:
@neilyoung unrelated to the main topic, regarding 

> It should be made possible to fold/unfold those long quotes...

Here's how you can do this in Github comments:

    <details>
      <summary>BLE scan log</summary>
      
      ```
      (log goes here)
      ```
    </details>

(make sure you have that empty line between `<summary>` and the backticks!)

The result will look like this:

<details>
 <summary>BLE scan log</summary>

```
(log goes here)
```
</details>

negativekelvin, 2017-07-20:
@neilyoung you would be a hero if you could collapse all these logs like that, ...i've been scrolling for days

So what chipset your April brother beacons use, cc2541?

Can we get a list of chipsets that work well with esp32 and those that don't?

neilyoung, 2017-07-20:
@negativekelvin I feel with you, sorry :) @igrr Thanks for the hint, I will try to edit my posts later again.

It is a DA14580 https://www.dialog-semiconductor.com/products/connectivity/bluetooth-low-energy/smartbond-da14580

All in all a smart 3 chip solution with a CR2450 battery.

neilyoung, 2017-07-20:
@negativekelvin OK, have updated the worst posts... Sorry again.

TianHao-Espressif, 2017-07-20:
@neilyoung , we have fixed this RX packet losing bug, please wait the release/v2.1 version. Thanks.

neilyoung, 2017-08-01:
@TianHao-Espressif It is not fixed with 2.1

TianHao-Espressif, 2017-08-02:
@neilyoung , please check component/bt/lib submodule commit id. It should be 8afa935.
On V2.1 version, we test for serval device(which have the same problem like this), all can receive the ADV.

neilyoung, 2017-08-02:
@TianHao-Espressif 

```
git rev-parse --short HEAD
10942b2

```

Uhm... How this? I made a clean git clone of the entire esp-idf...

neilyoung, 2017-08-02:
Did clone it again. Same result:

```
Cloning into 'components/bt/lib'...
remote: Counting objects: 154, done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 154 (delta 2), reused 6 (delta 2), pack-reused 148
Receiving objects: 100% (154/154), 756.01 KiB | 585.00 KiB/s, done.
Resolving deltas: 100% (87/87), done.
Checking connectivity... done.
Submodule path 'components/bt/lib': checked out '10942b2ff00f7db473c1917f76be93e59730e73f'

```

igrr, 2017-08-02:
10942b2 is the commit ID of BT lib on the master branch of IDF. 8afa935 is the commit ID of BT lib on release/v2.1 branch of IDF. If you want to get v2.1, you need to do `git checkout v2.1 && git submodule update --init`.

negativekelvin, 2017-08-02:
Right but fix RX packets losing bug(8e2ff681) is in master too

neilyoung, 2017-08-02:
@negativekelvin No it seems not. Only @igrr 's hint brought success.

That being said: The problem seems to be fixed.

3 RadBeacons, 60 seconds scan. One seen 234 times, one 109, one 91. OK

negativekelvin, 2017-08-02:
Hmm OK yes v2.1 has `easing syncerr to fix RX packet losing bug` while master has `fix RX packets losing bug(8e2ff681)` so master still has the issue?

TianHao-Espressif, 2017-08-03:
@negativekelvin , Yes.  Both of the branch release/v2.1 and master fix this bug.

negativekelvin, 2017-08-03:
@neilyoung do you still see difference between latest master and latest v2.1?

closed 2017-10-10 by FayeY