espressif/esp-idf #422: i2c matster EEPROM Selective (Random) Read problem

santhoshEsp32, 2017-03-12:
there is problem while working with random read of EEPROM in ESP32 idf,  I have tested EEPROM by i2c scanner  and its detecting also i can write multiple byte to EEPROM, i'am facing problem while reading multiple bye from EEPROM. and its end with timeout error(error code 0x107). one interesting thing is as soon as ESP boot once it will read data then after SDA & SCL signal goes to low(i have probed in DSO) then after even write function also end-up with time out error, if we read other i2c device in same bus it will also end with timeout error. all this happening when we call repeated start after addressing required read location. and this repeated start is required here to stop write operation and start with read process. i need urgent help for this problem. i have also tested nkolban sir DS1307 RTC code in that he is following same and it will work fine. but for EEPROM its not working and i have just tested with  same nkolban RTC code and i have attached text file here. 

[i2c_eeprom issue.txt](https://github.com/espressif/esp-idf/files/836831/i2c_eeprom.issue.txt)

santhoshEsp32, 2017-03-13:
@Neil Kolban sir can you help on this issue?

projectgus, 2017-03-13:
Paging @nkolban ... :)

santhoshEsp32, 2017-03-13:
@projectgus sir can you tell me solution for this problem?

nkolban, 2017-03-13:
Howdy @santhoshEsp32,
Can you explain more about "EEPROM" ... what EEPROM is this?    What is the nature of the I2C slave you are interacting with?   Can you attach a signal analyzer to the bus and validate that the correct I2C signals are being sent and received ... see:

https://www.youtube.com/watch?v=CE4-T53Bhu0

If you are getting timeout errors, then that means that the ESP32 I2C is expecting to read data with an acknowledgement and that acknowledgement is not happening within the timeout period.  We can look at the circuit all day long but without a trace of the signals, we can't go much further.   Fortunately logic analyzers are very cheap ... about $6 (US) ... so I strongly recommend getting one.  Without a logic analyzer, we are debugging in the dark.

santhoshEsp32, 2017-03-13:
Hello Kolban sir !thank you for your reply . I have used at24c32  atmel eeprom and also i have tested fm24l64 FRAM from cypres semiconductor memory in im facing same problem. Some time it will work fine . And each time when i reboot it will read once then its not gona read . As you said its problem with ack mostly if that is the problem then how it can read at 1st boot ? And yes sir i have alredy ordered logic analyser its on the way !. Once again thank you sir !!

nkolban, 2017-03-13:
There is another problem listed here ... https://github.com/espressif/esp-idf/issues/393

This problem sounds similar to yours.  The work around for the other problem that might be applicable to you is to always do a "ping" of the I2C target device when you start.

for example
```
i2c_cmd_handle_t cmd = i2c_cmd_link_create();
ESP_ERROR_CHECK(i2c_master_start(cmd));
ESP_ERROR_CHECK(i2c_master_write_byte(cmd, (i2c_add << 1) | I2C_MASTER_WRITE, 1 /* expect ack */));
ESP_ERROR_CHECK(i2c_master_stop(cmd));
i2c_master_cmd_begin(I2C_NUM_0, cmd, 1000/portTICK_PERIOD_MS);
i2c_cmd_link_delete(cmd);
```

Just do this one time after I2C is initialized.  The chances are that this ping request may fail but that's ok, in that subsequent I2C calls will then work.  Study issue #393 for details.

santhoshEsp32, 2017-03-15:
it was same issue #337 , problem got fixed !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!.I tired before and that time i was not successful because supplied wrong function . thanks @nkolban .:p

nkolban, 2017-03-15:
@santhoshEsp32 Good news.  Make sure you follow github process and if the issue is resolved that you close out this github issue so you don't spend folk's time checking resolved issues.

santhoshEsp32, 2017-03-15:
@nkolban yes i was waiting for your reply !

closed 2017-03-15 by santhoshEsp32