espressif/esp-idf #423: ESP logging library - how to remove colorization characters?

neilyoung, 2017-03-12:
The ESP logging library https://esp-idf.readthedocs.io/en/v1.0/api/log.html seems to decorate log traces with escape characters, which ensure proper colorization if displayed with Putty (and maybe other clients). It is pretty cumbersome to decipher those logs if displayed on an "ordinary" serial monitor 

Example (this is still pretty good readable, but usually the traces are worse):

[0;32mI (28) boot: ESP-IDF v2.0-rc1-257-g4745895 2nd stage bootloader[0m
[0;32mI (28) boot: compile time 16:45:51[0m
[0;32mI (55) boot: Enabling RNG early entropy source...[0m
[0;32mI (56) boot: SPI Speed      : 40MHz[0m

Is there a way to get rid of those "[0;32mI" and friends?

projectgus, 2017-03-13:
Hi Neil,

If you run `make menuconfig`, there's an option under `Component Config` -> `Log output` to "Use ANSI terminal colors in log output". If you disable this then the logging output will be clean ASCII.

Angus

closed 2017-03-13 by projectgus

neilyoung, 2017-03-13:
Thanks. Worked