espressif/esp-idf #424: mbedtls and detecting non-blocking reads during handshaking

nkolban, 2017-03-13:
If we attempt to use a non-blocking socket with mbedtls we receive a failure.  I dug deeply into this and believe I have found the resolution.

If we look at this function:

https://github.com/espressif/esp-idf/blob/master/components/mbedtls/port/net.c#L204

The intent is to determine if an error reported by a previous socket call was caused because the socket is non blocking.  The return code is 0 if the last error was NOT caused by being a non-blocking socket while the return code is 1 if the last error was because we WERE a non-blocking socket. The logic is:

1. If the socket is a blocking socket , return 0
2. If the last error was because we were non-blocking, return 1.
3. Else ... return 0

This sound sensible enough, however there is perhaps a flaw in the implementation.  The we check "if the last error was because we were non-blocking" is to ASK the socket about its last error.  We don't use `errno` because we are multi-threaded and premptable.  Again, on the surface, that took makes sense.  However in test number 1, we perform an `fcntl` against the socket to determine if it is blocking or non-blocking ... and **that** resets the error condition that was present on the socket when the function was called.   What this means is that the original error is lost.

projectgus, 2017-03-13:
Thanks @nkolban, this looks good (and correct) to me.

Are you able to please rewrite the commit message so it summarises the fix? something like:

```
mbedtls port: Fix non-blocking socket detection of MBEDTLS_ERR_SSL_WANT_READ/MBEDTLS_ERR_SSL_WANT_WRITE
Checking non-blocking status clears errno, so read this first.

Closes #424
```

Would be ideal. If you're not comfortable rewriting git history in this way then that's OK, I can rewrite the message when I push it into our internal merge queue.

nkolban, 2017-03-13:
Sorry sir, I'm not good at Git ... :-(

closed 2017-03-15 by igrr