espressif/esp-idf #427: "make flash monitor" doesn't work till manual reboot

eliabieri, 2017-03-14:
If i use the command `make flash monitor` to flash and after monitor the UART, the serial monitor only starts to spit out something after I manually reattached the ESP32 board.
In menuconfig, I configured to reset the board after flashing.
I use the the [following](http://www.watterott.com/de/ESP-WROOM32-Breakout) ESP32 breakout board from Watterott.
Does anyone have an idea how to fix this behavior?

projectgus, 2017-03-14:
Hi elia,
* What host OS are you running on?
* Does anything happen if you press and release the "EN" button on the board, instead of manually removing and reattaching?

eliabieri, 2017-03-14:
I use macOS 10.12.3.
The EN button doesnt work either to make the UART start spitting out data.

forthlightning, 2017-03-14:
I am hitting a similar issue on macOS 10.11.6. Pressing EN and boot hasn't been working recently. 

Physically reattaching the usb and then doing the normal Boot > EN button dance reliably allows the board to flash. To be clear this issue occurs before the board has successfully flashed.

projectgus, 2017-03-16:
As of yesterday there's a new tool, "idf monitor", which adds some new features around monitoring & resetting devices:
http://esp-idf.readthedocs.io/en/latest/idf-monitor.html

This may help with these issues. (For one, the new idf_monitor explicitly tries to reset the board each time it starts up.)

eliabieri, 2017-03-17:
I just tried the new idf monitor. (its very nice 💯 )
The problem however stays. If I hit ctrl-T ctrl-F, the new firmware gets compiled, flashed and then the promt outputs `Leaving...
Hard resetting...`
After that, now serial output gets displayed until I manually reattach the dev board.

projectgus, 2017-03-17:
Thansk @eliabieri . That's bad news.

Unfortunately in that case this sounds like a problem with the reset circuit wiring on the board you have. Our Espressif boards don't have any problem soft-resetting (including from macOS, a number of developers use this OS). Please ask the hardware manufacturer to review their design.

(If you happen to find a modification to the monitor tool which works and doesn't break other boards, we'd be happy to look at merging it.)

closed 2017-03-17 by projectgus

eliabieri, 2017-03-17:
You are right. I try the ESPRESSIF DEV-KIT C now. Thanks for the help anyway.

awatterott, 2017-03-18:
@eliabieri @projectgus The problem is a CP2102N driver issue under MacOS and Silabs is currently looking after it. Here are further infos: http://community.silabs.com/t5/Interface/CP2102N-Strange-Issue/td-p/190476

projectgus, 2017-03-20:
Thanks for filling us in @awatterott. Very glad it's been tracked down.

awatterott, 2017-04-14:
The CP2102N issue happens (on Mac and Linux), when the serial port is closed and re-opened immediately. As workaround a short pause between the uploading and monitoring seems to fix it and the waiting can be done in the esptool.py upload tool: https://esp32.com/viewtopic.php?f=13&t=1287#p7637

mlinaje, 2018-04-20:
I had the same problem on Windows 10. It's for sure a CP2102N driver issue. I detailed the problem and the solution/workaround (same as for Linux/MacOS) at https://www.silabs.com/community/interface/forum.topic.30.15.html/cp2102n_strange_issu-Sd3I

projectgus, 2018-05-30:
Hi @mlinaje,

Sorry this got a bit lost on this closed issue. If you're still needing to patch esptool.py with the latest SiLabs Windows drivers then can you please open an issue with some more details on the esptool repo: http://github.com/espressif/esptool/issues

The forum post suggests that introducing a 0.1s delay was enough to work around the issue with the older MacOS drivers, in which case we can probably add this in esptool without causing any major inconveniences.