espressif/esp-idf #428: esp_ble_gatts_set_attr_value not saving characteristic value

h-vishal, 2017-03-15:
Start a GATT service and add a characteristic using `esp_ble_gatts_add_char` with an initial characteristic value then in the program logic set the characteristic value to a new value using `esp_ble_gatts_set_attr_value`. Read the characteristic using `esp_ble_gatts_get_attr_value` and you will get the initial characteristic value. If you want to just have a connected client modify the value you can use the `esp_attr_control_t` option of `esp_ble_gatts_add_char` to have the value saved automatically and access it using `esp_ble_gatts_get_attr_value`, however a large number of use cases require the program logic to set the characteristic value and although you can work around this by maintaining your own database of values this is replicating the functionality provided by `esp_ble_gatts_set_attr_value` and `esp_ble_gatts_get_attr_value`.

heyinling, 2017-03-15:
Hi @h-vishal 
`esp_ble_gatts_set_attr_value` `esp_ble_gatts_get_attr_value` only work with attribute created by `esp_ble_gatts_create_attr_tab`. you can refer to example in `esp-idf/examples/bluetooth/gatt_server_service_table`.

for attribute created by `esp_ble_gatts_add_xxx`, you need to manually reply when receive read/write events in gatts callback.

h-vishal, 2017-03-15:
So we have two sets of values? One for setting the initial value in `esp_ble_gatts_add_char` then another set for the read/write response like I mentioned, this seems redundant.

heyinling, 2017-03-15:
I think for your use scenario, just set NULL to `char_val` and `control`. Then handling read write is entirely up to application.

JonathanChenNew, 2017-03-16:
@heyinling 
HI heyinling!
"esp_ble_gatts_set_attr_value esp_ble_gatts_get_attr_value only work with attribute created by esp_ble_gatts_create_attr_tab."
I create the attribute with  "esp_ble_gatts_create_attr_tab",and "esp_attr_control_t" is "ESP_GATT_RSP_BY_APP",just like this:
`	 
static const uint8_t UpdEnd[1]={0x0a};

 /****UpdEnd****/
	// UpdEnd Characteristic Declaration
	[PRO_UPDATE_IDX_UPDATE_END_DECL]            =
	{{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_declaration_uuid, ESP_GATT_PERM_READ,
			sizeof(uint8_t),sizeof(uint8_t), (uint8_t *)&char_prop_read_write}},

	// UpdEnd Characteristic Value
	[PRO_UPDATE_IDX_UPDATE_END_VAL]             	=
	{{ESP_GATT_RSP_BY_APP}, {ESP_UUID_LEN_16, (uint8_t *)&UpdateProfilechar4UUID, ESP_GATT_PERM_READ|ESP_GATT_PERM_WRITE,
			sizeof(UpdEnd), sizeof(UpdEnd), (uint8_t *)UpdEnd}},`
" esp_ble_gatts_get_attr_value "works well,but "esp_ble_gatts_set_attr_value" does not work.
My Code:
`	case ESP_GATTS_WRITE_EVT: {
		ESP_LOGI(UPDATE_GATTS_TAG, "GATT_WRITE_EVT, conn_id %d, trans_id %d, handle %d\n", param->write.conn_id, param->write.trans_id, param->write.handle);
		ESP_LOGI(UPDATE_GATTS_TAG, "GATT_WRITE_EVT, value len %d, value %08x\n", param->write.len, *(uint32_t *)param->write.value);
		LOG_ERROR("write value = %d\n",param->write.value[0]);
		esp_ble_gatts_set_attr_value(param->write.handle,param->write.len,param->write.value);
		esp_ble_gatts_send_response(gatts_if, param->write.conn_id, param->write.trans_id, ESP_GATT_OK, NULL);

		break;
	}`
thanks!

xcguang, 2017-04-11:
Hi @JonathanChenNew 
Does it work now?

JonathanChenNew, 2017-04-19:
@xcguang 
NO.
But i found the mistake:The function "esp_ble_gatts_set_attr_value" does not work.the code about this function is empty. so , i use my own code!

Yulong-espressif, 2017-06-11:
@JonathanChenNew Thank you so much for find the issue on the IDF, we have fixed this bug in the new version of SDK, you can get it from the github. Any question please let me know. Thanks.

closed 2017-07-13 by FayeY

davidtrc, 2017-09-06:
It is still not working.
I declare a char like:

```
/// Heart Rate Sensor Service - Heart Rate Object Size Characteristic, write&read
static const uint16_t hr_obj_size_uuid = 0x2AC0;
static uint8_t hr_obj_size_val[4] = {0x00, 0x00, 0x00, 0x00};
```

After in the esp_gatts_attr_db_t database:

```
// Heart Rate Object Size Characteristic Declaration
    [HRS_IDX_HR_OBJ_SIZE_CHAR]          = 
    {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_declaration_uuid, ESP_GATT_PERM_READ,
      CHAR_DECLARATION_SIZE, CHAR_DECLARATION_SIZE, (uint8_t *)&char_prop_read_write}},
                                                    
    // Heart Rate Object Size Characteristic Value
    [HRS_IDX_HR_OBJ_SIZE_VAL]             = 
    {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&hr_obj_size_uuid, ESP_GATT_PERM_READ|ESP_GATT_PERM_WRITE,
      4*sizeof(uint8_t), sizeof(hr_obj_size_val), (uint8_t *)hr_obj_size_val}},
```

And when calling:

` esp_ble_gatts_set_attr_value(health_thermometer_handle_table[HTS_IDX_HT_OBJ_SIZE_VAL], sizeof(ht_obj_size_val), &ht_obj_size_val);`

After increasing the value of hr_obj_size_val, the value when reading remains always the same (first assigned).