espressif/esp-idf #429: can't get `examples/bluetooth/gatt_server` to work?

edmund-huber, 2017-03-15:
I built and flashed the `examples/bluetooth/gatt_server` to the ESP32. I'm using the ESP32 Sparkfun "Thing", for reference. The startup from reset looks (?) fine:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:3464
load:0x40078000,len:7828
load:0x40080000,len:252
entry 0x40080034
I (28) boot: ESP-IDF v2.0-rc1-258-g0bbf02b 2nd stage bootloader
I (28) boot: compile time 12:24:22
I (62) boot: Enabling RNG early entropy source...
I (62) boot: SPI Speed      : 40MHz
I (62) boot: SPI Mode       : DIO
I (73) boot: SPI Flash Size : 4MB
I (86) boot: Partition Table:
I (97) boot: ## Label            Usage          Type ST Offset   Length
I (119) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (143) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (166) boot:  2 factory          factory app      00 00 00010000 00100000
I (189) boot: End of partition table
I (202) boot: Disabling RNG early entropy source...
I (219) boot: Loading app partition at offset 00010000
I (1259) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512) 
I (1260) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x2abc4 (175044) map
I (1276) boot: segment 2: paddr=0x0004abd4 vaddr=0x3ffc0000 size=0x01edc (  7900) load
I (1306) boot: segment 3: paddr=0x0004cab8 vaddr=0x40080000 size=0x00400 (  1024) load
I (1330) boot: segment 4: paddr=0x0004cec0 vaddr=0x40080400 size=0x12290 ( 74384) load
I (1391) boot: segment 5: paddr=0x0005f158 vaddr=0x400c0000 size=0x00000 (     0) load
I (1392) boot: segment 6: paddr=0x0005f160 vaddr=0x00000000 size=0x00ea8 (  3752) 
I (1414) boot: segment 7: paddr=0x00060010 vaddr=0x400d0018 size=0x629ec (403948) map
I (1442) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1464) heap_alloc_caps: At 3FFCB9A0 len 00014660 (81 KiB): DRAM
I (1485) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAM
I (1507) heap_alloc_caps: At 40092690 len 0000D970 (54 KiB): IRAM
I (1527) cpu_start: Pro cpu up.
I (1539) cpu_start: Starting app cpu, entry point is 0x40080d38
I (0) cpu_start: App cpu up.
I (1571) cpu_start: Pro cpu start user code
I (1632) cpu_start: Starting scheduler on PRO CPU.
BTDM CONTROLLER VERSION: 010101
btip start
 Sta .bss scheduom  ffb800  tU.3[0mff70, len 00007f70
BTDM ROM VERSION 0101
BD_ADDR: 24:0A:C4:00:96:72
NVDS MAGIC FAILED
RF Init OK with coex
I (311) phy: phy_version: 329, Feb 22 2017, 15:58:07, 1, 0
Enable Classic BT
Enable Low Energy
E (311) BT: Startup BTU

E (311) BT: GATTS_CreateService:  handle of service handle1
E (331) BT: bta_dm_co_ble_load_local_keys: func not ported

I (341) GATTS_DEMO: REGISTER_APP_EVT, status 0, app_id 0

E (351) BT: Partial data write into ADV
I (351) GATTS_DEMO: CREATE_SERVICE_EVT, status 0,  service_handle 40

I (351) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 40

I (361) GATTS_DEMO: ADD_CHAR_EVT, status 0,  attr_handle 42, service_handle 40

I (371) GATTS_DEMO: the gatts demo char length = 3

I (371) GATTS_DEMO: prf_char[0] =11

I (381) GATTS_DEMO: prf_char[1] =22

I (381) GATTS_DEMO: prf_char[2] =33

I (381) GATTS_DEMO: ADD_DESCR_EVT, status 0, attr_handle 43, service_handle 40

I (391) GATTS_DEMO: REGISTER_APP_EVT, status 0, app_id 1

I (401) GATTS_DEMO: CREATE_SERVICE_EVT, status 0,  service_handle 44

I (411) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 44

I (411) GATTS_DEMO: ADD_CHAR_EVT, status 0,  attr_handle 46, service_handle 44

I (421) GATTS_DEMO: ADD_DESCR_EVT, status 0, attr_handle 47, service_handle 44
```

I open up `bluetoothctl`, and attempt to pair with the device:

```
$ bluetoothctl 
[NEW] Controller 88:1F:A1:19:FF:DF laptop [default]
[NEW] Device 24:0A:C4:00:96:72 ESP_GATTS_DEMO
[bluetooth]# power on
[CHG] Controller 88:1F:A1:19:FF:DF Class: 0x00010c
Changing power on succeeded
[CHG] Controller 88:1F:A1:19:FF:DF Powered: yes
[bluetooth]# agent on
Agent registered
[bluetooth]# default-agent
Default agent request successful
[bluetooth]# pair 24:0A:C4:00:96:72 
Attempting to pair with 24:0A:C4:00:96:72
[CHG] Device 24:0A:C4:00:96:72 Connected: yes
```

After the `pair` command in `bluetoothctl`, I get new output from the ESP32:

```
E (36651) BT: smp_br_connect_callback is called on unexpected transport 2

E (36651) BT: MTU request PDU with MTU size 517

E (36651) BT: Call back not found for application conn_id=3
I (36661) GATTS_DEMO: SERVICE_START_EVT, conn_id 0, remote 88:1f:a1:19:ff:df:, is_conn 1

I (36661) GATTS_DEMO: SERVICE_START_EVT, conn_id 0, remote 88:1f:a1:19:ff:df:, is_conn 1
```

There is a pause of about 10 seconds. Then I get this new output in `bluetoothctl`:

```
Failed to pair: org.bluez.Error.AuthenticationFailed
[CHG] Device 24:0A:C4:00:96:72 Connected: no
```

And this new output over serial from the ESP32:

```
E (68641) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0013
E (68641) BT: btm_sec_disconnected clearing pending flag handle:0 reason:19

E (68641) BT: bta_gattc_mark_bg_conn unable to find the bg connection mask for: 88:1f:a1:19:ff:d
```

I tested `bluetoothctl` by pairing with my Android phone -- this worked fine. Am I missing something about how I should be setting things up in order for this example to work?

The error `E (68641) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0013` seems to indicate (HCI_ERR_PEER_USER) that the `laptop` device closed the connection. The `bluetoothctl` output seems to further indicate that `laptop` closed the connection because auth failed. But I have no idea if I am interpreting this correctly.

edmund-huber, 2017-03-15:
This seems to be the same problem described in https://github.com/espressif/esp-idf/issues/250 .

negativekelvin, 2017-03-15:
I think pairing/bonding is still not supported.  Did you try just connect?

edmund-huber, 2017-03-15:
Thanks, good suggestion:
```
[bluetooth]# power on
Changing power on succeeded
[bluetooth]# devices
Device 24:0A:C4:00:96:72 ESP_GATTS_DEMO
Device 4C:A5:6D:F0:B4:38 SAMSUNG-SM-G900V
[bluetooth]# connect 24:0A:C4:00:96:72 
Attempting to connect to 24:0A:C4:00:96:72
Failed to connect: org.bluez.Error.Failed
```

There's no additional output in the service log.

VallaiPakkam, 2017-04-04:
Hi, i have the same problem.. Could u able to pair ?

edmund-huber, 2017-04-09:
As @negativekelvin mentioned, pairing isn't supported. However, I was able to use Android's BLE facilities to connect and swap characteristics with the esp32 .. so everything is fine!

closed 2017-04-09 by edmund-huber

HITMAnsOFT, 2017-05-08:
@edmund-huber May I ask, what version of Android did you use to connect to the gatt server? I'm having problems with 6.0.1.