espressif/esp-idf #432: ESP-IDF monitor cannot neither be opened nor closed from VSCode

neilyoung, 2017-03-15:
Probably not your fault, but running 

`make flash monitor
`

run as build command under Visual Studio Code just runs compile and flash. The monitor does not start. It terminates with 

```
termios.error: (102, 'Operation not supported on socket')
make: *** [monitor] Error 1

```
If one is running 

`make monitor
`

from integrated VSCode terminal he is unable to terminate it by CTRL-]

I'm sure Atom will show the same behaviour (not tested).

I'm aware, that these problems might be related to VSCode. But because both commands run well in a Mac OSX terminal, I thought it might be of interest for you.

projectgus, 2017-03-16:
That's unfortunate, thanks for reporting it. Apparently IDF Monitor runs under Eclipse on OS X, so it's possible it will work under Atom.

It will depend a lot on how faithfully VSCode (or any other IDE/host) emulates a terminal, or whether it treats "make" as a non-interactive process (not an unacceptable assumption, really).

There might be something that can be done to detect non-terminal environments and disable some of the keyboard shortcut behaviour (which relies on terminal support), while keeping the basic monitor functionality.

projectgus, 2017-03-16:
Just to double-check: I'm guessing this is with the new idf_monitor functionally that was added yesterday? Is it the same if you "make simple_monitor" instead?

neilyoung, 2017-03-16:
No, I guess it is the old one, because I don't currently have the "make simple_monitor" command. Need to upgrade later on. Will let you know.
Thanks for now

neilyoung, 2017-03-16:
Well, the description on the ESP-IDF website regarding the new monitor seems to be a bit ahead the reality. I have cloned the github repositories again, but there is no 

`make simple_monitor
`

So I can't really say, whether there is the "old monitor" or the "new monitor" running. At least also the new monitor seems to require CTRL-[ to be terminated. On my MacBook this is "CTRL-ALT-6", but this combination is not working in the VSCode terminal yet.

Only unplugging the ESP32 dev board terminates the monitor.

neilyoung, 2017-03-17:
I can confirm: It does not work with the latest IDF release (the one which contains the new monitor). Same behaviour as initially reported.

projectgus, 2020-04-23:
Hi @neilyoung,

Thanks for reporting this initially and being patient about it.

Since this issue was opened we have developed a new VSCode extension dedicated to IDF that includes monitoring functionality along with a lot of other features:
https://github.com/espressif/vscode-esp-idf-extension/

This is now the best option for using VSCode with ESP-IDF including monitoring device output.

closed 2020-04-23 by projectgus