espressif/esp-idf #433: Core Dump Analysis

forthlightning, 2017-03-15:
Hi,

I am trying to use espcoredump.py with a corefile saved into flash.

When I use menuconfig to set coredump destination to flash my serial output looks like this:
```
--- forcing DTR inactive
--- forcing RTS inactive
--- Miniterm on /dev/tty.SLAB_USBtoUART  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:2488
load:0x40078000,len:7128
load:0x40080000,len:252
entry 0x40080034
I (1376) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1376) heap_alloc_caps: At 3FFC8F70 len 00017090 (92 KiB): DRAM
I (1386) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAM
I (1407) heap_alloc_caps: At 4009ADB0 len 00005250 (20 KiB): IRAM
I (1428) cpu_start: Pro cpu up.
I (1440) cpu_start: Single core mode
I (1453) cpu_start: Pro cpu start user code
E (3498) esp_core_dump: No core dump partition found!
I (3545) cpu_start: Starting scheduler on PRO CPU.
I (3597) app_main: hello world
```

Note at 3498 there is a "No core dump partition found."

[This](https://github.com/espressif/esp-idf/blob/b8e2edc99f369eaa069e478604c47907972b2585/docs/core_dump.rst) page of documentation says a partition is automatically created if the default configuration is used. I would like to confirm that this is the case, because the serial output does not reflect that any partition is created.

Then when I run this command 
`python $COREDUMP_PATH/espcoredump.py --port=/dev/tty.SLAB_USBtoUART dbg_corefile /build/solsense32.elf`

I get this output 
```
espcoredump.py v0.1-dev
esptool.py v2.0-beta1
Connecting........
Uploading stub...
Running stub...
Stub running...
16 (100 %)
16 (100 %)
Read 16 bytes at 0x110000 in 0.0 seconds (13.6 kbit/s)...
Hard resetting...

A fatal error occurred: Invalid start magic number!
```

igrr, 2017-04-05:
Could you please upload your sdkconfig file into a gist and post a link here?

forthlightning, 2017-04-05:
https://gist.github.com/forthlightning/0a4f53ca844e4233730e81fecb3db644

igrr, 2017-04-05:
OK, so you are using a custom partition table, ss_partition.csv. When one of the default partition tables is used, IDF will indeed automatically adjust it to add a core dump partition, when menuconfig option to use core dump to flash is selected. Quoting the documentation page you have linked above:

> When using default partition table files which are provided with ESP-IDF it automatically allocates necessary space on flash, But if user wants to use its own layout file together with core dump feature it should define separate partition for core dump as it is shown below:

forthlightning, 2017-04-05:
Ok, I think this sdkconfig may be from more recently than the original issue, 21 days ago. 

In any case I am not seeing this particular issue, thanks!

closed 2017-04-05 by forthlightning