espressif/esp-idf #434: 'mbedtls_net_recv' timeout

arsinio, 2017-03-15:
I know I've run into this problem before (and solved it), but for some reason it's come up again so I figured I should mention it...

Anyways, when attempting to connect to AWS IoT, I get the following error:
```
int tmpRet = mbedtls_ssl_read(&netClientIn->tls.sslContext, byteOut, 1);
// returns -76
```

So i decided to investigate a little further...if I enable mbedTLS debugging (through `make menuconfig`) I get the following debugging information:
```
ssl_tls.c::6484  2  => read
ssl_tls.c::3738  2  => read record
ssl_tls.c::2222  2  => fetch input
ssl_tls.c::2380  2  in_left: 0, nb_want: 5
ssl_tls.c::2404  2  in_left: 0, nb_want: 5
ssl_tls.c::2405  2  ssl->f_recv(_timeout)() returned -76 (-0x004c)
ssl_tls.c::3767  1  mbedtls_ssl_fetch_input() returned -76 (-0x004c)
```

I _believe_ ssl->f_recv ultimately works it's way back to `mbedtls_net_recv` in `net.c` and specifically the following code:
```
ret = (int) read( fd, buf, len );

    if ( ret < 0 ) {
        if ( net_would_block( ctx ) != 0 ) {
            return ( MBEDTLS_ERR_SSL_WANT_READ );
        }

        error = mbedtls_net_errno(fd);
        if ( error == EPIPE || error == ECONNRESET ) {
            return ( MBEDTLS_ERR_NET_CONN_RESET );
        }

        if ( error == EINTR ) {
            return ( MBEDTLS_ERR_SSL_WANT_READ );
        }

        return ( MBEDTLS_ERR_NET_RECV_FAILED );
    }
```

The problem is that at some point `ret` becomes `-1` which triggers this if block, and `net_would_block( ctx )` returns 0, and `mbedtls_net_errno(fd);` returns 0, leading to the catch-all `return ( MBEDTLS_ERR_NET_RECV_FAILED );`.

In reality, if you read `errno` (actual errno) after the `read` you'll see that it's set to `EAGAIN`. So _one_ possible solution would be to make the following modification to `net_would_block`:
```
int error = mbedtls_net_errno(ctx->fd);
// ******** begin modification ********
#if defined EAGAIN
    if( (error == 0) && (errno == EAGAIN) ) error = EAGAIN;
#endif
// ******** end modification ********
```

projectgus, 2017-03-16:
Thanks for reporting this. I think we fixed a similar bug yesterday with non-blocking sockets:
https://github.com/espressif/esp-idf/commit/829800f27290b1c9b0972e05b7e51b87180cba48

This seems like the same thing. I'll audit this entire file and fix this and other places where calling fcntl() (or other LWIP functions) resets errno.

arsinio, 2017-03-16:
Sounds reasonable...I'm not terribly familiar with with a lot of these function calls, but if that _is_ the case, this line in `net_would_block` is likely the culprit:

```
if ( ( fcntl( ctx->fd, F_GETFL, 0) & O_NONBLOCK ) != O_NONBLOCK ) {
```

which is called before:
```
int error = mbedtls_net_errno(ctx->fd);
```

Thanks for looking at this! Let me know if I can help!

projectgus, 2017-03-16:
That's it I think. Anywhere which calls net_would_block() has to read errno beforehand, because it's going to be reset afterwards.

nkolban, 2017-03-16:
This appears to be a duplicate of https://github.com/espressif/esp-idf/issues/424

projectgus, 2017-03-22:
It's the same underlying cause (fcntl() resets errno), but it's in a different place.

arifmansoori74, 2017-03-23:
Keep retrying mbedtls_ssl_read until you get a positive return code.

arsinio, 2017-03-23:
@arifmansoori74 

I'm not sure that is an option...`mbedtls_net_recv` returns `MBEDTLS_ERR_NET_RECV_FAILED` which I _believe_ is what ultimately gets returned from `mbedtls_ssl_read`. According to the docs for `mbedtls_ssl_read`:

>If this function returns something other than a positive value or MBEDTLS_ERR_SSL_WANT_READ/WRITE or MBEDTLS_ERR_SSL_CLIENT_RECONNECT, then the ssl context becomes unusable, and you should either free it or call mbedtls_ssl_session_reset() on it before re-using it for a new connection; the current connection must be closed.

arifmansoori74, 2017-03-24:
@arsinio 

Yes, thats correct from what I understand too from the documentation. However, I tried this approach upon seeing another implementation and it just worked. I am able to do a successful handshake even though intermittently I get the -76 return code. Also works flawlessly with mbedtls_ssl_read.

Or, you may also try using the timeout feature and use mbedtls_net_recv_timeout function instead of mbedtls_net_recv while setting up the mbedtls BIO. Use mbedtls_ssl_conf_read_timeout to set the timeout milliseonds prior.

projectgus, 2018-02-19:
Looks like this issue was fixed 10 months ago in PR #511 commit a523aa3ef538da39334d1c6a1819ec19e57dcc9e but this issue has not been closed to match.

For future reference, error -76 is  -0x004C which is `MBEDTLS_ERR_NET_RECV_FAILED`. This return value essentially means that the underlying socket read failed for some reason, maybe to do with the network or because the other end hung up unexpectedly. It means you need to reset the SSL session and start again with a new socket.

If you think ESP-IDF is doing the wrong thing here then you could try logging the value of error at this point:
https://github.com/espressif/esp-idf/blob/master/components/mbedtls/port/net_sockets.c#L368

Please open a new issue if you think ESP-IDF is doing the wrong thing, however if the other end closes the socket then there unfortunately isn't much we can do about it.

closed 2018-02-19 by projectgus