espressif/esp-idf #435: Wrong data type for scan initialization in BT samples

nkolban, 2017-03-16:
If we look in some of the BT samples ... for example ...

https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_client/main/gattc_demo.c#L75

We see that the property called `own_addr_type` is being assigned the value `ESP_PUBLIC_ADDR`.  Unfortunately, this value is from a typedefed enum that is not one of the possible allowable values for this data type.  I believe that the correct value should be `BLE_ADDR_PUBLIC`.

negativekelvin, 2017-03-16:
```
/// Own BD address source of the device
typedef enum {
    /// Public Address
    ESP_PUBLIC_ADDR,
    /// Provided random address
    ESP_PROVIDED_RND_ADDR,
    /// Provided static random address
    ESP_GEN_STATIC_RND_ADDR,
    /// Generated resolvable private random address
    ESP_GEN_RSLV_ADDR,
    /// Generated non-resolvable private random address
    ESP_GEN_NON_RSLV_ADDR,
    /// Provided Reconnection address
    ESP_PROVIDED_RECON_ADDR,
} esp_ble_own_addr_src_t;
```

```
/// Own BD address source of the device
typedef enum {
    /// Public Address
    BD_ADDR_PUBLIC,
    /// Provided random address
    BD_ADDR_PROVIDED_RND,
    /// Provided static random address
    BD_ADDR_GEN_STATIC_RND,
    /// Generated resolvable private random address
    BD_ADDR_GEN_RSLV,
    /// Generated non-resolvable private random address
    BD_ADDR_GEN_NON_RSLV,
    /// Provided Reconnection address
    BD_ADDR_PROVIDED_RECON,
} esp_bd_addr_type_t;
```

```
/// BLE device address type
typedef enum {
    BLE_ADDR_TYPE_PUBLIC        = 0x00,
    BLE_ADDR_TYPE_RANDOM        = 0x01,
    BLE_ADDR_TYPE_RPA_PUBLIC    = 0x02,
    BLE_ADDR_TYPE_RPA_RANDOM    = 0x03,
} esp_ble_addr_type_t;
```
`#define BLE_ADDR_PUBLIC         0x00`
`uint8_t own_addr_type`
`UINT8 own_addr_type = BLE_ADDR_PUBLIC;`
`esp_ble_addr_type_t     own_addr_type;`
`tBLE_ADDR_TYPE              own_addr_type;`

xiewenxiang, 2017-04-28:
Hi,

Thanks for your valuable advice, we will revise it in the futurre release.

`static esp_ble_scan_params_t ble_scan_params = {
    .scan_type              = BLE_SCAN_TYPE_ACTIVE,
    .own_addr_type          = BLE_ADDR_TYPE_PUBLIC,
    .scan_filter_policy     = BLE_SCAN_FILTER_ALLOW_ALL,
    .scan_interval          = 0x50,
    .scan_window            = 0x30
};`

closed 2017-06-13 by FayeY