espressif/esp-idf #442: Intermitent crash using BLE and Wifi

torntrousers, 2017-03-20:
Initially raised this over at ESP32/Arduino which was closed saying try here: https://github.com/espressif/arduino-esp32/issues/273#issuecomment-287709206

I've a sketch scanning for BLE advertisements and for each sending them off over Wifi with an MQTT publish. It runs ok for a while by after an hour or 3 or 48 will crash. Using the ESP Exception Decoder tool it shows:

0x400855b8: xTimerGenericCommand at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839
0x40083108: update_written_pages at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/./flash_mmap.c line 291
: (inlined by) spi_flash_mark_modified_region at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/./flash_mmap.c line 266
0x400855b8: xTimerGenericCommand at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839
0x40084782: xPortGetCoreID at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4402
: (inlined by) xTaskRemoveFromEventList at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 3045
0x40083e3b: xQueueGenericSendFromISR at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1998
0x400856a9: prvProcessExpiredTimer at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839
0x400856c2: prvProcessExpiredTimer at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839

And for another one of the crashes:
.E (1905257) BT: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x200c
Guru Meditation Error: Core 0 panic'ed (Interrupt wdt timeout on CPU0)
Register dump:
PC : 0x40085724 PS : 0x00060034 A0 : 0x800848ee A1 : 0x3ffd41c0
A2 : 0x3ffd3d34 A3 : 0x00060023 A4 : 0x00060020 A5 : 0x3ffeebe0
A6 : 0x00000003 A7 : 0x00060e23 A8 : 0x00000066 A9 : 0x00000066
A10 : 0x00060023 A11 : 0x00000000 A12 : 0x3f417a34 A13 : 0x0000001f
A14 : 0x00000001 A15 : 0x00000001 SAR : 0x00000004 EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000 LBEG : 0x400014fd LEND : 0x4000150d LCOUNT : 0xfffffffb

Backtrace: 0x40085724:0x3ffd41c0 0x400848ee:0x3ffd41e0 0x40083fa7:0x3ffd4200 0x40085815:0x3ffd4240 0x4008582e:0x3ffd4280

CPU halted.

shows:
0x40085724: prvProcessTimerOrBlockTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839
0x40085724: prvProcessTimerOrBlockTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839
0x400848ee: vTaskEnterCritical at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4402
0x40083fa7: xQueueGenericReceive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1998
0x40085815: prvProcessReceivedCommands at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839
0x4008582e: prvTimerTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839

Any ideas?

torntrousers, 2017-03-21:
Another one today:

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Register dump:
PC      : 0x40085724  PS      : 0x00060234  A0      : 0x800848ee  A1      : 0x3ffd41c0 
A2      : 0x3ffd3d34  A3      : 0x00060223  A4      : 0x00060220  A5      : 0x3ffeec50 
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x00000000  A9      : 0xb33f0000 
A10     : 0x00060223  A11     : 0x00000000  A12     : 0x3f417a34  A13     : 0x0000001f 
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x00000005 
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb 

Backtrace: 0x40085724:0x3ffd41c0 0x400848ee:0x3ffd41e0 0x40083fa7:0x3ffd4200 0x40085815:0x3ffd4240 0x4008582e:0x3ffd4280

CPU halted.

Which decoded gives:

Decoding 8 results
0x40085724: prvProcessTimerOrBlockTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839
0x40085724: prvProcessTimerOrBlockTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839
0x400848ee: vTaskEnterCritical at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4402
0x40083fa7: xQueueGenericReceive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1998
0x40085815: prvProcessReceivedCommands at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839
0x4008582e: prvTimerTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839

Even without a fix would it be possible for the ESP32 to restart instead of "CPU halted."

projectgus, 2017-03-21:
We've seen this crash recently when the timer task overflows its stack (the normal stack overflow detection fails to kick in, unfortunately). There are two options:

* Use less stack in the timer callback. This includes either not calling printf() or enabling the "nano printf" configuration item in menuconfig under "Components" -> "ESP32 Specific".
* Increase the timer task stack size. There is a PR open to add this as a menuconfig option #444 . In the meantime, you cam manually edit the `configTIMER_TASK_STACK_DEPTH` entry in FreeRTOSConfig.h.

> Even without a fix would it be possible for the ESP32 to restart instead of "CPU halted."

This is configured under menuconfig also: "Components" -> "ESP32 Specific" -> "Panic handler behaviour". The default should be to reset rather than halt.

torntrousers, 2017-03-21:
Thanks! What would be a sensible value to try for increasing configTIMER_TASK_STACK_DEPTH? 4096?
(i did try to workout what the current value is but not sure I can, what is XtExcFrameSize?)

torntrousers, 2017-03-21:
Well tried 4096 for configTIMER_TASK_STACK_DEPTH and it crashed again after a few hours. Have now up'd it to 32768...

projectgus, 2017-03-22:
4KB should be more than enough. Are you getting the exact same crash?

Does any of your code, or the MQTT code, etc. use FreeRTOS timers (xTimer___ APIs)?

torntrousers, 2017-03-23:
4K was the same crash and now again with the 32k setting. It lasted about 32 hours this time then:

scan completed, restarting ...
................................E (100032657) BT: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x200c
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Register dump:
PC      : 0x40085724  PS      : 0x00060634  A0      : 0x800848ee  A1      : 0x3ffd41c0 
A2      : 0x3ffd3d34  A3      : 0x00060623  A4      : 0x00060620  A5      : 0x3ffeebe0 
A6      : 0x00000003  A7      : 0x00060423  A8      : 0x00000068  A9      : 0x00000068 
A10     : 0x00060623  A11     : 0x00000000  A12     : 0x3f417a34  A13     : 0x0000001f 
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x00000005 
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb 

Backtrace: 0x40085724:0x3ffd41c0 0x400848ee:0x3ffd41e0 0x40083fa7:0x3ffd4200 0x40085815:0x3ffd4240 0x4008582e:0x3ffd4280

CPU halted.

which decodes to what looks like just the same:

Decoding 8 results
0x40085724: prvProcessTimerOrBlockTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839
0x40085724: prvProcessTimerOrBlockTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839
0x400848ee: vTaskEnterCritical at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4402
0x40083fa7: xQueueGenericReceive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1998
0x40085815: prvProcessReceivedCommands at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839
0x4008582e: prvTimerTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839

My code doesn't use xTimer___ APIs and i dont think the MQTT code does either as its the standard Arduino [pubsub cllient](https://github.com/knolleary/pubsubclient).

I do have quite a lot of prints in the code which i think you mentioned earlier, and its quite busy picking up BLE advertisments from a couple of devices every second or so.

torntrousers, 2017-03-23:
Also I should say the IDF code I have isn't the very latest, its from back on 23rd of Feb. Is there anything relevant in the latest?

torntrousers, 2017-03-26:
I noticed there usually appears to be some Wifi connection problem immediately before this crash, an MQTT publish fails as its lost connection with the broker and tries to reconnect, then crash. 

So now I just do esp_restart() if an mqtt publish fails. Seems a bit hacky but its been working ok for some days now. Regularly gets the fail and restarts so is down for some seconds but otherwise no crash and halt.

torntrousers, 2017-03-30:
Up for 5 days but now even with the regular esp_restart() another crash that looks just the same:

Reconnecting MQTT client to air0i6.messaging.internetofthings.ibmcloud.com
.E (3709274) BT: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x200c
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Register dump:
PC      : 0x40085724  PS      : 0x00060c34  A0      : 0x800848ee  A1      : 0x3ffd41c0 
A2      : 0x3ffd3d34  A3      : 0x00060c23  A4      : 0x00060c20  A5      : 0x3ffeebe0 
A6      : 0x00000003  A7      : 0x00060a23  A8      : 0x00000066  A9      : 0xb33fffff 
A10     : 0x00060c23  A11     : 0x00000000  A12     : 0x3f417a68  A13     : 0x0000001f 
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x00000005 
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb 

Backtrace: 0x40085724:0x3ffd41c0 0x400848ee:0x3ffd41e0 0x40083fa7:0x3ffd4200 0x40085815:0x3ffd4240 0x4008582e:0x3ffd4280

CPU halted.
Decoding 8 results
0x40085724: prvProcessTimerOrBlockTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839
0x40085724: prvProcessTimerOrBlockTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839
0x400848ee: vTaskEnterCritical at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4402
0x40083fa7: xQueueGenericReceive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1998
0x40085815: prvProcessReceivedCommands at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839
0x4008582e: prvTimerTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./timers.c line 839

jack0c, 2017-05-30:
I tried to debug. Bug I can't reproduce. Did you use dual-core mode? To say the truth, the dual-core version is not so stable until idf2.0.  I suggest you to use the latest version, at lease idf2.0.

torntrousers, 2017-05-31:
Hi @jack0c what do you mean you can't reproduce - do you mean you've been able to successfully run both BLE and Wifi for days on end and it works fine for you with no crashes? Would you share your code?

jack0c, 2017-06-03:
1.It's not so easy to write code like what you described. So I just run some sample code in latest version, but I did not use Arduino.
2.My question was do you use dual-core? If you do, I suggest to use latest idf version.
3.Please give the codes , .elf file or SHA of commit to me, than I can do more.
   The problem is the debug info "__LINE__" did not match my code.

closed 2017-11-21 by FayeY