espressif/esp-idf #443: SpiFlashChip g_rom_flashchip provides wrong values

martinayotte, 2017-03-20:
(just a recap of the issue described in https://esp32.com/viewtopic.php?f=14&p=6739#p6722)
The "SpiFlashChip g_rom_flashchip" structure in sdk/include/esp32/rom/spi_flash.h provides wrong/dummy values since g_rom_flashchip.deviceId is returning 1540ef while esptool.py is returning 1640c8.

chmorgan, 2017-10-18:
Any updates on this issue? I'd really like to use the flash chip deviceId as a unique hardware identifier.

projectgus, 2017-10-19:
This is being worked on but it's not complete yet, unfortunately.

>  I'd really like to use the flash chip deviceId as a unique hardware identifier.

This is probably not what you want, as the flash JEDEC ID is a part number identifier not a serial number. So all ESP32s with the same flash chip part will have the same deviceId.

If you want a non-secret unique identifier, I'd suggest using the MAC address or something derived from the MAC address.

If you want a secret unique identifier, you can use the ESP32 RNG to produce a string of random bytes and then store these in efuse BLK3 (which is internal to the ESP32 and not used for anything else).

This issue is the wrong place to discuss unique identifiers, but if you post on https://esp32.com then I can help you further there.

chmorgan, 2017-10-19:
@projectgus ahh this is very helpful. I misunderstood what deviceId is and it makes sense that it could be the part id, not a unique serial number. Sorry for the noise here.

FayeY, 2017-11-30:
See https://esp32.com/viewtopic.php?f=14&p=6739#p6722

jack0c, 2018-03-21:
@projectgus Can we close this issue?

igrr, 2018-03-21:
We need to merge MR1315 first @jack0c  :)

igrr, 2019-03-01:
This has been implemented a while ago in https://github.com/espressif/esp-idf/commit/b1dcb52fec4d427cc831240d675e6804d9aebc6f#diff-e52b7631da27a0e2aecacf4a59cd9a62R122, sorry that we forgot to update this issue.

closed 2019-03-01 by igrr

X-Ryl669, 2019-10-19:
Please notice that most flash provide a command 4B to get the flash unique id (typically 64bits or 128bits long, depending on the chip). This is not supported (or documented?) for GD25LQ32C chip, but it's supported for GD25LQ64C. I don't know what chip is being installed on WROOM32 modules, but that might worth supporting it.

X-Ryl669, 2019-10-19:
BTW, the thread on the forum did exactly that, they are sending `4B` command, so I *guess* it also works on GD25LQ32C, it's just not in the datasheet. The remark from @projectgus is not correct through, and I think there is a mix of name in the code.

The RDID command (`9F`) query the manufactured ID + device information (mainly the size and the possible access mode). Yet, the code calls it `device_id` which is confusing since there is another command `90` that does get the device ID (but as @projectgus stated, it's the same for all chip with the same reference). 
Finally, there's a unique ID in the flash chip that's independent from the device information or device ID (via command `4B`). I would have expected this to be implemented (not that hard to add since we already have `device_id`) since it would save to waste BLK3 precious bits just to store a unique identifier. 

In all cases, if such ID are used for any cryptographic source, they are not safe, since they are likely continuous, and as such, one could figure out the "keys" for the 2 products based on their serial number (for example: if they differ by 2, there's a high chance that the MAC's also differ by 2, and so will the flash chip)

X-Ryl669, 2019-10-19:
Please have a look to [this commit](https://github.com/X-Ryl669/esp-idf/commit/10dec4e861723303a1a6769f1abea7b4f2cffb7a) for implementation of the idea. I can't send a PR here, since I'm still waiting on you merging my last 2 bugfixes and I don't want to bootstrap again 2 month ago.

X-Ryl669, 2019-10-19:
Just a remark, I've tested on 2 devices I've bought at the same time (so likely very close in production) and they returns very different flash UID (but it's consistent, it's always the same value after reboot, either by physical reboot or software reboot). I've no idea what such UID should look like (there is absolutely 0 documentation on the content of these id in the SPI's flash datasheet), so finally, they might look like they are safe to use for a true "pseudo random" UID.

Can you guys confirm they are different on your side with 2 WROOM32 modules that follows in production ?

projectgus, 2019-11-04:
HI @X-Ryl669 ,

> Can you guys confirm they are different on your side with 2 WROOM32 modules that follows in production ?

We have a number of flash chip vendors whose parts are used in production. As our manufacturing team's qualification tests don't include verifying the 4Bh command, you can't rely on this command working on all Espressif modules with the same part number. Sorry.

Going back one step, can you explain exactly what you need to use this for? As you explained, it's a bad choice for any kind of "secret" identifier as it's trivial to recover. There is already a unique MAC address for a non-secret identifier, and if you want a non-secret value that is different to the MAC then you could hash the MAC address bytes (with or without some additional fixed bytes if you want to avoid it being trivially derived from the MAC). Another option is to generate a random identification number and write it to a partition in flash, if you need to reserve efuse BLK3 for another purpose.

If using flash encryption then the recommended approach would be to generate the ID and then store it an encrypted NVS partition.

X-Ryl669, 2019-11-04:
That's a lot of options, indeed!
Let's sum up a bit:

| Unique ID source | Usable for identifying a product | Usable in crypto-secure env | Easily computable |
|---|---|---|---|
| MAC address | Yes | No | Yes (any packet sniffer can deduce it) |
| Flash UID | Yes | Low security (could be sniffed on the SPI bus, but requires product disassembly) | No |
| Hash(Flash) or Hash(MAC) | Yes | No to Low | No additional protection compared to no hashing (since it's trivial to test for usual algorithms, even if using a salt) |
| Unique Key in eFuse | Yes | Yes | No (but a burden to set up and maintain) |
| Encrypted Flash & NVS | Yes | Yes | No (a burden to set up, but probably inevitable for a safe product release) |

projectgus, 2019-11-04:
Disagree on a few points:

* Flash UID is no more secure against physical readout than generating a random number and storing it in unencrypted flash, or storing a key in efuse. In all cases, an attacker with physical access can read this out - including via the serial link (although esptool doesn't have support for sending this RDUID command it has support for sending arbitrary SPI flash commands so it would be a simple job for someone to add it). 

  The only significant difference is that Flash UID is harder to clone to a new device (would probably need to binary patch the firmware to clone a device, instead).
* For Hash(MAC), agree an attacker who knows one MAC can enumerate the full MAC range and come up with a lot of possible Hash(MAC) IDs to test against. Unclear if the RDUID flash properties have the same characteristic of being enumerable or not, might depend on the manufacurer.

To repeat the question from the last comment: **What do you need this to do? What's your threat model?**

X-Ryl669, 2019-11-04:
In my case, I'm looking a unique ID that not dump-able (and that's large enough not to be enumerated, like 64bits or more). This is used to derive some cryptographic keys (that are used in turn for programming NFC tags). Upon production, I must know the *per-device* "root" key (stored in a database) so I can assert if the information in the NFC tag is correct or not.

In my *specific case*, I guess an encrypted flash is the solution to use (but that means that I must use some kind of production firmware specific for writing the keys, then flash another firmware that's only reading it, without any programming capacity). 

This is additional work compared to "just" using the FlashUID that's *already* here (no additional step while producing the product). But as you said, if it's accessible via serial link, then there's no additional protection compared to eFuse (yet, it's still better than the MAC address since it's never sent on the network).