espressif/esp-idf #448: esp_restart(); and ESP.restart(); not working

forthlightning, 2017-03-21:
Hello, I'm trying to restart my device remotely on occasion. Interpreting JSON from a cloud message. 

Instead of device restarting, wifi turns off and doesn't turn back on even if I use WiFi.begin()

I realize this is a bit of a mixture between idf and arduino methods.

```
W (55504) mqtt_callback: restarting device by command
I (55510) wifi: state: run -> init (0)
I (55514) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
I (55519) wifi: pm stop, total sleep time: 0/34390394

I (55525) wifi_event_handler: event: 5
mbedtls_ssl_write returned -0x4e
[E][WiFiClientSecure.cpp:140] write(): 113

Cleaning SSL connection.
I (55540) wifi_event_handler: SYSTEM_EVENT_STA_DISCONNECTED
I (55544) wifi_event_handler: event: 3
I (55548) wifi_event_handler: SYSTEM_EVENT_STA_STOP
```

The print statement at 55504 directly precedes the esp_restart(); command.

I upped the priority on the parent task so I don't think that is the problem. Do I have to pause all the relevant tasks or clear out the queues or something like that before it works?

projectgus, 2017-03-22:
There was a bug along these lines recently, but it should be fixed in current IDF master. What version do you have? I noticed you mentioned WiFi.begin(), are you using IDF or Arduino?

>  Do I have to pause all the relevant tasks or clear out the queues or something like that before it works?

No, that shouldn't be necessary. We recommend calling esp_wifi_stop() to cleanly disassociate from the AP (at the moment this happens automatically, but this behaviour may change in the future.)

Can you provide a code sample showing the function where you call esp_restart()?

Angus

FayeY, 2017-05-01:
Hi, forthlightning, is this problem resolved now?

TimXia, 2017-05-08:
@forthlightning Please try the latest IDF to see whether this problem still exsit.

closed 2017-07-07 by FayeY