espressif/esp-idf #449: Wifi/BLE scan coexistence

jkhax0r, 2017-03-22:
I'm seeing an issue that seems pretty critical and after spending a good 10 hours investigating figured I'd make sure I'm not crazy...

Currently we have setup BLE scanning and WIFI STA connection.  If I scan for BLE advertisements from my device (a remote control), which sends at ~25-35ms intervals, I often will see some very sporadic intervals between reception, some ranging as high as 500ms.  This is with a scan interval/window of 0x50/0x30 per and lots of other combinations as well as different settings for PASSIVE/ACTIVE scanning.   This is with strong RSSI the entire time.  I have checked in nRF Connect on Android and confirmed regular intervals of ~25ms coming from my device.  Also this same device (remote control) was used with a previous product with no problems.  

I can turn off all initialization of the WIFI AP and STA (but still leave "WIFI" options checked in make menuconfig) and the behavior is virtually the same, so I don't think its specific to any custom WIFI code I have added.  I have not yet tried to completely disable WIFI in menuconfig b/c I got some errors the first time I tried

My question if this behavior is expected?   Should I be able to scan near full duty (while having WIFI enabled) and expect updates a bit faster?

There's still more I will continue to try, but I wanted to probe the experts to see if you guys had any thoughts.  I'll be happy to collect and post my code although its damn close to the gatt client example and a few others I have found searching around the forums and web for possible solutions.  As I mentioned, I'm committed to getting this working since its critical to my product, so will continue to investigate.  I'd even be happy to send one of my devices to a developer if needed.  

Thanks in advance for any help!

jkhax0r, 2017-03-22:
Here's some screenshots of my advertiser info.  

[image]
[image]

neilyoung, 2017-03-22:
@jkhax0r Not something which will help you, but IMHO there is a lot of problems around BLE scanning (please refer also to this: https://github.com/espressif/esp-idf/issues/421): Missing scans, wrong RSSI values, scattered scan results. I also suspected a coexistence problem, but all these BLE problems are also present, if Wifi is completely off here.

What kind of ESP32 dev board are you using?

However, I doubt somehow, that you will receive _any_ single advertisement frame in case it is sent at 30 ms interval. I'm not receiving that either, and my beacons advertising at 500-600 ms. But I have never seen a stack, which would fire at this rate (not on Android, iOS or OSX).

jkhax0r, 2017-03-23:
@neilyoung We've built some custom PCBs that has the ESP-WROOM-32.  I will go and check on our sparkfun thing dev board too.  

One thought I had is that maybe the scan cycles through one of the 3 advertising channels after each scan interval and perhaps only one is working, or that 2 out of the 3 are too far out of band or something.  Just a wild ass guess b/c I seem to often get intervals at exactly 3x the scan interval when I setup my phone to advertise at 100ms.  NOTE: This also might explain your symptoms (issue #421) where certain devices just will not show up.  So is there some kind of NVRAM frequency tuning or calibration that might be slightly out of whack, or even just the crystal???  But its probably much more fundamental b/c this doesn't explain wifi connectivity issues impacted by scan interval/window ratio.  

Also weird is that I often get advertisements that are spaced at 0 or 1ms ticks apart.  Unless that's some anomaly with the RTOS tasks and priorities, I can't really explain it.  I'll go and print the data (I randomize it on each advert) and see if maybe that's a repeated packet or not.

I'm not expecting 30ms (obviously not with a a scan window that shuts off for 30+ms), but hoping for at least 1 in 4 of these adverts to give around 100ms or even 200ms in the weak worst signal case.  Anything else is easily perceivable.  I have seen this function properly on the 8-bit, 32Mhz, 8051, TI CC2540 chipset, so I don't think its unreasonable to expect.  And my phone (Galaxy S7) was hauling them in at this rate when I took those screenshots (although I didn't explicitly check for the largest gap, i didn't see even one miss out of about 10 I briefly inspected, and the cool plot that nRF Connect shows had at least 10 dots every second for the 5-10 seconds I sampled).

BTW, I don't mean to complain, as I absolutely love this CPU and environment and am already restructuring and planning for my company to create a whole suite of products around it.   Just hoping a dev might chime in with some thoughts...

neilyoung, 2017-03-23:
@jkhax0r Thanks for the detailed response. Yes, I think 100-200ms should be achievable. The crystal is for sure one suspect, but what concerns me a more is your note regarding interferences between Wifi and BLE: This in fact is not that unlikely as it seems. I know that from some Broadcomm chips (combined BLE and Wifi), built into some Motorola Android devices. They definitely cannot operate simultaneously because each one tramples down the other. So maybe we have the same case here?  If you are sure with your observations, then the hardware design has to be blamed and it would be a big CON for all ideas around ESP32 and BLE.

Other than that I'm still waiting for the promised RSSI fix and I hope our friends at Expressif are checking the issue with the RadBeacon. My preliminary verdict is: Right now the ESP32 is not a good choice if you plan to do _real_ BLE stuff and maybe one should better stick with ESP8266 (or others) and TI, which is supposed to work well in some products known to me.

Siwykrzywy, 2017-05-23:
Same issue here, with code from examples on WROOM-32 we have many lost packets. We were changing esp_ble_scan_params_t (scan_interval and scan_window) and things are really messed up (most of the time it works for about 15 seconds and crashes or stops for 2 minutes). Best stability is for default params.
We have mean interval at about 800ms from iBeacon (kontkat.io) set at 350ms.

neilyoung, 2017-05-24:
@Siwykrzywy If you would like to test with my test program: https://github.com/neilyoung/ESP32_blufi_beacon_tracker

Smeedy, 2017-10-19:
Hi there,

Any updates on this topic? I do have the same issues and miss a lot of advertisement data. What is the best strategy for the esp_ble_scan_params_t and the total BLE scan?

Thanks,
Martijn

Yulong-espressif, 2017-12-16:
@Smeedy I think we have fix this bug in the new version of SDK, please update it. Thanks.

closed 2017-12-21 by FayeY