espressif/esp-idf #452: tcp connect timeout too long

chenkaiyao, 2017-03-24:
If I try to do a TCP connect to an "unplugged" IP, I block my socket too long. Is there some features or opts.h values to use to reduce this timeout?

forthlightning, 2017-03-27:
I am seeing this also, sometimes extended connection time causes my tasks to hit the watchdog timer.

liuzfesp, 2017-05-10:
Hi @chenkaiyao @forthlightning for TCP socket:
1. The default socket mode is blocking mode, it means it will blocked the calling task until the connection is setup (3-hand-shaking is completed) or timeout or error happens. In your example, if connected an "unplugged" IP, socket tries to send the TCP sync packets, the max retry time is CONFIG_TCP_SYNMAXRTX (default is 6), if all 6 retry fails, then the connect API returns. The tcp sync retry is based on a backoff timer, the backoff timer increased as the retry counter increased, e.g. maybe the retry happens after 1, 1, 1, 1, 1, 1 seconds etc, then after 6 seconds, the API returns.
 
2. You can configure the socket to unblocking mode with fcntl(socket_id, F_SETFL, O_NONBLOCK) mode if you expect the connect API returns immediately, you need to fully understanding unblocking socket programming model before you decide to use unblocking socket

3. As for @forthlightning 's watchdog timer, I think it has nothing with blocked socket, because when the socket is in blocked status, it will not hold the CPU

closed 2017-06-19 by FayeY