espressif/esp-idf #454: Eclipse Error

sfranchin63, 2017-03-24:
Hello guys,
I'm trying to compile the example hello-world with eclipse IDE in win7, but always receive  this error:
/usr/bin/bash: C:/msy32/home/Stefanof/esp-idf/tools/windows/eclipse_make.sh: No such file or directory
I checked the file and directory exist.
Can someone help me.
Thanks
Stefano
[image]

ESP32DE, 2017-03-25:
@sfranchin63 
[please read the windows setup for eclipse](http://esp-idf.readthedocs.io/en/latest/eclipse-setup-windows.html#eclipse-windows-setup)

[you have forget to](http://esp-idf.readthedocs.io/en/latest/eclipse-setup-windows.html#project-properties) 
- uncheck use default build command
- add python command for the shell..

here is a picture
[image]

hope this helps

best wishes
rudi ;-)

sfranchin63, 2017-03-26:
Hi Rudi, 
I have made change you suggested, but it have failed.
In order to set Eclipse I followed the instructions (Setup Build Variables):
https://exploreembedded.com/wiki/Setting_up_Eclipse_for_ESP-IDF
[image]
[image]
Thanks
Stefano

projectgus, 2017-03-27:
There's a typo in the path, `msy32` -> `msys32` and it should work.

sfranchin63, 2017-03-27:
Thanks for your suggestions and with the change of msys32 now it all works .

projectgus, 2017-03-28:
Glad to hear you got it working.

closed 2017-03-28 by projectgus