espressif/esp-idf #455: Crash on reboot while wifi is connecting

rojer, 2017-03-24:
Every now and then i get a crash and core dump when the device is supposed to reboot.
So yes, reboot happens, but it's not exactly clean:

```
I (4063) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
I (4623) wifi: state: init -> auth (b0)
I (4633) wifi: state: auth -> assoc (0)
Tick
esp32_fs_deinit      Unmounting FS
I (4643) wifi: state: assoc -> run (10)
I (4643) wifi: state: run -> init (0)
I (4653) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
wifi_event_handler   WiFi STA: disconnected, reason 8
mgos_system_restart  Restarting
event_handler        event: 3
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x4009983a  PS      : 0x00060330  A0      : 0x800e571e  A1      : 0x3ffc0dd0
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00060320  A5      : 0x3ffc0e68
A6      : 0x0000000f  A7      : 0x00000000  A8      : 0x80083bd1  A9      : 0x00000001
A10     : 0x00000003  A11     : 0x00060323  A12     : 0x00060320  A13     : 0x9d00caa6
A14     : 0x00000000  A15     : 0x3ffb1148  SAR     : 0x00000020  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000044  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace: 0x4009983a:0x3ffc0dd0 0x400e571e:0x3ffc0df0 0x400927be:0x3ffc0e20 0x40094659:0x3ffc0e50
...
```

Looks like wifi was connecting when the time came to reboot.
The code that performs reboot is [here](https://github.com/cesanta/mongoose-os/blob/5b498824a184a2a7a732ee6fcbc2e4bb2494ac04/fw/platforms/esp32/src/esp32_hal.c#L35)

The backtrace at the bottom symbolizes to: `ieee80211_sta_is_running <- eapol_txcb <- ppProcTxDone <- ppTask`
Looks like `esp_restart()` stopped wifi but then TX callback arrived and touched some state that was already gone (`EXCVADDR` suggests a `NULL` struct pointer dereference).

rojer, 2017-03-24:
i should not that event 3 is `SYSTEM_EVENT_STA_STOP`

TimXia, 2017-04-12:
What did you do when received event 3?

rojer, 2017-04-12:
nothing, just [print](https://github.com/cesanta/mongoose-os/blob/7a1e7e62feaab8de8c747d0eaf28e046a7a967ad/fw/platforms/esp32/src/esp32_main.c#L64).

TimXia, 2017-04-12:
Did you use the latest IDF?

rojer, 2017-04-12:
i was at the time when this was filed. we are currently using IDF at https://github.com/espressif/esp-idf/commit/65baf502b7777655c7e8088311fdb709cba23109

TimXia, 2017-05-08:
@rojer Please try the latest IDF to see whether the problem still exist.

liuzfesp, 2017-05-10:
Hi @rojer I think this issue is fixed in latest idf.

rojer, 2017-05-10:
yes, i haven't seen this in a while. let's consider this fixed.
OTOH, i keep seeing #341 sometimes and have just seen it happen again - with latest, most bleeding edge IDF (a41ac2d21d90fc69f8426eb6036d2d1e1274ec28).

closed 2017-05-10 by rojer