espressif/esp-idf #460: No documentation on BLE characteristic value_type

nkolban, 2017-03-27:
We can make a call to esp_ble_gattc_read_char() to read a BLE characteristic.  This causes an ESP_GATTC_READ_CHAR_EVT to be raised when the value is ready.  In the data structure returned, there is a field called `value_type`.  See:

http://esp-idf.readthedocs.io/en/latest/api/bluetooth/esp_gattc.html#_CPPv2N24esp_ble_gattc_cb_param_t25gattc_read_char_evt_param10value_typeE

Unfortunately there doesn't appear to be any documentation on the values that `value_type` can take and hence I am stuck on how to interpret a returned characteristic value.

joicetm, 2017-05-29:
Hi Kolban,

I am also in a similar situation, have you find any solutions ?

nkolban, 2017-05-29:
Sorry, I wandered off to study other things and don't think I ever came back to this.

joicetm, 2017-05-30:
Thanks for your reply. 
FYI,
Its now working for me, have a look at the structure definitions on esp_gattc_api.h
url: https://github.com/eerimoq/simba-esp32/blob/master/inc/bt/bluedroid/api/include/esp_gattc_api.h
try to read the value type, value length and the pointer to actual value.

YelongHuang, 2017-06-01:
@nkolban Thanks for your advertise, the value type has two type value which is GATTC_READ_VALUE_TYPE_AGG_FORMAT &　GATTC_READ_VALUE_TYPE_VALUE, if the value type is GATTC_READ_VALUE_TYPE_AGG_FORMAT means the the unit of the value is define according to spec Part G 3.3.36, this type of value should define in the Characteristic Aggegate Format descriptor, and the value of GATTC_READ_VALUE_TYPE_VALUE is for the generally value type. We will add it to the note docs in the next version of the SDK.

Yulong-espressif, 2017-06-11:
@nkolban Thank you so much for your suggetion, that is really helpful for us. We have change it in the new version of SDK, you can get it from the github. Thanks.

closed 2017-06-23 by nkolban