espressif/esp-idf #461: ESP-IDF can not use ADC2 channels！

FFtust, 2017-03-27:
The ADC driver API currently only supports ADC1 channels，but if I want to use ADC2 channels , how should I do?

ESP32DE, 2017-03-28:
@FFtust 
recommend you this steps:
first, download the [technical reference](http://espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf) for esp32 

study the register and compare the thing with adc1.

then [check the header](https://github.com/espressif/esp-idf/blob/master/components/driver/include/driver/adc.h) for adc

right, you see, there is just in time only support adc1

then look up [in the c file](https://github.com/espressif/esp-idf/blob/master/components/driver/rtc_module.c) for , where adc is at home
and right again, you see, that adc1 is only supported just in time.

if you can not wait for official support adc2 then you must do this steps:

1) look up which pins are have ad2 supported function

2) add in adc header file your enum for this as adc2 type pins , example for new adc2_channel_t

3) add in adc header file 3 (api) function declaration  
- esp_err_t adc2_config_width(adc_bits_width_t width_bit);
- esp_err_t adc2_config_channel_atten(adc2_channel_t channel, adc_atten_t atten);
- int adc2_get_voltage(adc2_channel_t channel);

4) add in rtc_module.c the definition for this, a good example are the adc1 function'see

...
[it looks like this then](https://github.com/ESP32DE/esp-idf/commit/f467f68ff50d679ac4004eea2dcaa812d50912a7) 

everything too much trouble?,  [then here you are welcome to test this repo](https://github.com/ESP32DE/esp-idf)

btw, pullrequest is done.
https://github.com/espressif/esp-idf/pull/466

hope this helps
best wishes
rudi ;-)

YanMinge, 2017-03-30:
Well done, just need it.

OldCharter, 2017-04-01:
For me adc2 works fine until I enable wifi. adc2 stops working then. Same issue when using the esp32-arduino libraries. Could this be the reason why adc2 isn't available in esp-idf?

igrr, 2017-04-01:
Exactly! WiFi driver uses some of the ADC2 circuitry. So the code in the PR can indeed be used when WiFi/BT are disabled, otherwise controlling ADC2 from software may interfere with WiFi.

Supporting simultaneous WiFi and ADC2 operation is not something we have plans to support, for now. We will however add an ADC2 driver which may be used if WiFi is not in use.

CarlosGS, 2017-04-01:
Quoting @20leunam: _"Then we will chop off the antenna, and plug an external bluetooth module to TX/RX!"_ :)
I agree it is a pity to discard so many analog pins. This will limit many applications of ESP32.

Do you think it may at least be possible to live-disable WiFi momentarily, in order to measure ADC2, and re-enable wireless it afterwards? Or is the adc2 locked at boot time?

igrr, 2017-04-01:
Yes, this should be possible. We'll need to make sure that WiFi/PHY initialization code resets all the ADC2 registers into the expected state during initialization (this may not be the case now), but aside from that, disabling WiFi (using esp_wifi_stop/esp_wifi_deinit) to do the measurement should be possible.

ESP32DE, 2017-04-01:
Would like to do that very much, too.
Only we unfortunately do not know what exactly is done under the api commands,   esp_wifi_stop/esp_wifi_deinit Since they are packed in the lib.
So sometimes we can only wait or do it ourselves.
or pack third party code now in libs too :)
honest: libs are bad if src is not available and counterproductive
It is my personal opinion

negativekelvin, 2017-04-01:
> I agree it is a pity to discard so many analog pins. This will limit many applications of ESP32.

Yes possibly down to two if you use pre amp and 32khz. Would be nice to have even low rate async adc2 read possible in cooperation with the WiFi driver.

CarlosGS, 2017-04-29:
I am trying to read from ADC2 after enabling and disabling wifi. So far I've tested multiple combinations of esp_wifi_stop, esp_wifi_deinit, esp_phy_rf_deinit to disable the wifi driver.
The test set up is a pull down resistor in ADC GPIO26, which properly reads as "0" until wifi is started.

Since @igrr mentioned restoring back ADC registers to default state, I've also force re-run the [ADC initialization routines in Arduino-ESP32](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-adc.c#L85), after wifi had been disabled.
This did not help: reading back from ADC2 pins always returns 4095. Maybe I'm missing other ADC registers?
Using rtc_gpio_init had no effect either.

I've run out of ideas. Please let me know if you think of other ways to achieve this. I'd be happy to test them!

CarlosGS, 2017-05-05:
I'm making a connected sensor that uses most of the ADC pins, and for the moment my approach is going to consist in:
- Read from the ADC.
- Then turn wifi ON to send the values.
- Finally self-restart the board, with wifi disabled (restoring the possibility to read from ADC2).
This is very ugly. So I am hoping it will be possible to achieve it in a clean way at some point :)

HardwireIO, 2017-05-11:
Can we have a clarification about ADC2, please?

Is it a matter of an hardware limitation and WiFi cannot work together with ADC2, or simply there's no support for ADC2 right now?

Can we expect to have ADC2 available in the future or it's simply impossible?

igrr, 2017-05-11:
ADC2 can be used together with WiFi on a time sharing basis. I.e. WiFi stack uses ADC2 for its own purposes, but also allows the application to poll ADC2 using some API. Such scheme is not implemented yet, and will not be part of the upcoming release.

CarlosGS, 2017-05-12:
What about using esp_wifi_stop/esp_wifi_deinit to do the measurement, do you advise to wait until the proper driver is ready?

frax84, 2017-06-05:
Hello,
i would like to know if the same limitation is valid for BT too or just WiFi. I'm asking this because of a similar issue 

> Fyi, we're a bit hesitant to make ADC2 available because it is also used for pretty important control loops inside the WiFi (and maybe BT, I don't remember) drivers. We want to make sure there's no chance of these and the user ADC calls to interfere with each other.

amuzyka-grinn, 2017-07-19:
Hello,
@igrr  could you specify which ADC2 channels are used by WiFi. On our custom board, we met with a problem of establishing a connection with AP (APs scan is working fine). At the same time, we are using pins with ADC2 func for other purposes like uart, digital out and in. We tried wifi example on our board (is not working) and Esp Wroom (is working). I know there might be several other reasons, but it helps cut off ADC2 related suspicions. Thanks in advance!

igrr, 2017-07-19:
None of the ADC2 channels connected to external pads are used by WiFi driver. There is a mux which is used to connect other internal signals to ADC2 input. Digital pins connected to ADC2 pins should not have any effect on WiFi.

enitalp, 2017-08-30:
> "Such scheme is not implemented yet, and will not be part of the upcoming release."
So igrr, any hope it will be implemented any time soon. It's really, really important for us. Thanks

Yuggo1, 2017-10-03:
Hi, i want to use the adc1 and adc2 channels from esp32 integrated with the bluetooth, i'm using the spp_counter from bluekitchen as a base code, this is the repo with the code:
https://github.com/Yuggo1/ESP_Gloves
The code to support the adc2 i've got from here ESP32DE/esp-idf@f467f68
and it works when using it without the bluetooth, but i'm getting the same problem from this issue using the bluetooth, does anyone know if ADC2 conflicts with bluetooth using spp?

Best Regards

whizkid79, 2017-10-30:
Sorry if this is a stupid question, but d30f3e7 only contains test code as far as I see. Are there further changes to use adc2 in connection with wifi?

negativekelvin, 2017-10-31:
Doesn't seem like what would expected from [coexistence](https://github.com/espressif/esp32-wifi-lib/commit/a3c8f349fe3c9e5be6cf74b813a1dc505fe38dc1)

igrr, 2017-10-31:
That's not coexistence with WiFi yet. This just allows using ADC2 when WiFi
is disabled, and prevents collisions between app and wifi driver if someone
tries to use ADC2 with WiFi enabled. Full coexistence is still being
developed.

whizkid79, 2017-10-31:
Is there another issue I can watch to get notified on coexistence?
Couldn't find one.

igrr, 2017-10-31:
Let's keep this one open until coexistence with WiFi is available.

whizkid79, 2017-10-31:
Thanks a lot @igrr

surumadurum, 2017-11-16:
I know it's annoying, but: Is there a time-estimate on when coexistence will be available?
Are we talking about weeks or months?
Thanks!

MichaelEL65, 2017-12-12:
I'm finding my use limited by this issue too. I wasn't using WiFi for my project but BLE, and apparently it runs into the same problem.

whizkid79, 2017-12-21:
Anything new here? Would be a nice christmas present ;)

dubkov, 2018-05-21:
Hi! I've just got stuck with exactly the same problem with BLE. I have 16 ADCs running from ADC1 and ADC2 banks. They all work as expected till the moment when I initialize BLE - then suddenly all 10 ADCs from bank ADC2 stop working but ADCs from 1st bank work as usual. It is crucial to me to have 16 ADCs and BLE to function alltogether. Is there any chance to get them work simultaniously?

whizkid79, 2018-10-23:
Anything new on this topic?

superman13, 2019-01-04:
Hi, I have a problem like this one... 
When I power up my module i receive valid data from ADC2.
After data was received I start wifi and send them via wifi and going to deep sleep.
After waking up I read permanently 4095 till I trigger an reset...
ULP is also working
PS. Before going to deep sleep I call esp_wifi_stop();

Alvin1Zhang, 2019-03-22:
@liuzfesp Hi, would you help share if any updates? Thanks.

CarlosGS, 2021-07-25:
In order to fix https://github.com/espressif/arduino-esp32/issues/102#issuecomment-881694700 we would just need a method that allows to reset the ADC2 without having to reset the whole board. Any chance to achieve this, or is it a silicon bug?

joko2020, 2021-07-28:
The Issue ** still exist until 2021 ** more than 4 year..
today i implemented suggestion from [https://github.com/espressif/arduino-esp32/issues/102#issuecomment-611307887](url)  but still did not worked
`
#include <WiFi.h>
#include "soc/sens_reg.h" // needed for manipulating ADC2 control register
uint64_t reg_b; // Used to store ADC2 control register
#define PIN 27 // Substitute xx with your ADC2 Pin number
int value;
int adc2[]={25,26,27,14,12,13,15,2,4};
const char* ssid     = "private";
const char* password = "private";

void setup() {
// Save ADC2 control register value : Do this before begin Wifi/Bluetooth
reg_b = READ_PERI_REG(SENS_SAR_READ_CTRL2_REG);
//Wifi.Begin(); // or similar wifi init function or Bluetooth begin()
WiFi.begin(ssid, password);

while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());

}

void loop() {
for(int i=0;i<8;i++)
{
WRITE_PERI_REG(SENS_SAR_READ_CTRL2_REG, reg_b);
SET_PERI_REG_MASK(SENS_SAR_READ_CTRL2_REG, SENS_SAR2_DATA_INV);
value = analogRead(adc2[i]);
Serial.print(adc2[i]);
Serial.print("=");
Serial.print(value);
Serial.print(",");
}
Serial.println("");
delay(500);
}

`

Icarus113, 2025-03-14:
Supported, [adc oneshot read example](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/adc/oneshot_read/main/oneshot_read_main.c#L116)

commit d30f3e7a (2017-10-30): example(adc2): add example and test case for adc2.

closes #461
closed 2025-03-14 by Icarus113