espressif/esp-idf #462: Enabling RMT/semaphores in complex code can cause analogread to return values 2x too big

marcmerlin, 2017-03-28:
As first reported here https://github.com/MartyMacGyver/ESP32-Digital-RGB-LED-Drivers/issues/7
using the RMT neopixel driver causes analogread to return 4096 for 50%-100% and proper values from 0 to 50%.
Sadly if you are sampling data that needs to go in the 50 to 100% range (joystick), it's impossible to get usable data.
If i comment out just these 3 lines of code, then analogread works properly again:
//RMT.conf_ch[RMTCHANNEL].conf1.mem_rd_rst = 1;
//RMT.conf_ch[RMTCHANNEL].conf1.tx_start = 1;
//xSemaphoreTake(ws2812_sem, portMAX_DELAY);

projectgus, 2017-03-28:
Are you seeing this running with Arduino or on IDF?

marcmerlin, 2017-03-28:
arduino, in my IoTuz code.

projectgus, 2017-03-28:
Please open the issue here: https://github.com/espressif/arduino-esp32/issues/ as the ADC software implementations (along with various other subsystem parts) are different.

If you can reproduce the issue under IDF by itself then feel free to reopen here.

marcmerlin, 2017-03-28:
Ooops, sorry I was under the wrong project, moved accordingly, thanks for the note.
https://github.com/espressif/arduino-esp32/issues/290

closed 2017-03-28 by marcmerlin