espressif/esp-idf #465: How to get scan response

Nicholas3388, 2017-03-28:
I found that there is a ESP_BLE_EVT_SCAN_RSP event define in esp_gap_ble_api.h, but I didn't find a sample to show how to get scan response. When ESP_GAP_SEARCH_INQ_CMPL_EVT event comes, I try to print scan_rst.ble_evt_type, but there is no ESP_BLE_EVT_SCAN_RSP event comes. Can anyone provide a sample to show how to get scan response?

heyinling, 2017-03-29:
HI, 
you can refer to the process of `ESP_GAP_BLE_SCAN_RESULT_EVT` in [gattc demo](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_client/main/gattc_demo.c#L304)

FayeY, 2017-05-03:
Hi Nicholas3388, is this problem solved? Can we close this issue?

Nicholas3388, 2017-05-03:
@FayeY  I think there is no process for ESP_BLE_EVT_SCAN_RSP in the gattc demo. Can you provide a sample to show how to handle ESP_BLE_EVT_SCAN_RSP event?

FayeY, 2017-05-03:
@Nicholas3388 , OK, we are going to work on it. Thanks for your interest in ESP32!

xiewenxiang, 2017-05-03:
@Nicholas3388 
If you want get scan response : 
```
void esp_gap_cb(esp_gap_ble_cb_event_t event, esp_ble_gap_cb_param_t *param)
{
    switch (event) {
    case ESP_GAP_BLE_SCAN_PARAM_SET_COMPLETE_EVT:
        break;
    case ESP_GAP_BLE_SCAN_START_COMPLETE_EVT:
        break;
    case ESP_GAP_BLE_SCAN_RESULT_EVT: {
        esp_ble_gap_cb_param_t *scan_result = (esp_ble_gap_cb_param_t *) param;
        switch (scan_result->scan_rst.search_evt) {
        case ESP_GAP_SEARCH_INQ_RES_EVT:
                printf("ADV DATA:");
                for(int i = 0; i < scan_result->scan_rst.adv_data_len ;i++){
                    printf("%02x",scan_result->scan_rst.ble_adv[i]);
                }
                printf(", SCAN RSP DATA:");
                for(int i = 0; i < scan_result->scan_rst.scan_rsp_len ;i++){
                    printf("%02x",scan_result->scan_rst.ble_adv[scan_result->scan_rst.adv_data_len + i]);
                }
                printf("\n");
            break;
        case ESP_GAP_SEARCH_INQ_CMPL_EVT:
            break;
        case ESP_GAP_BLE_ADV_START_COMPLETE_EVT:
            break;
        default:
            break;
        }
        break;
    }
    default:
        break;
    }
}
```

closed 2017-05-16 by Nicholas3388