espressif/esp-idf #467: Enhancement: fix pathing for windows_install_prerequisites.sh

madscientist42, 2017-03-29:
Right now it's conglomerating all sorts of things in the msys32 profile.d file that probably shouldn't be in there and using an explicitly fixed pathlist- which is a no-no in a profile.d entry.

Best answer, since this gets ran for each invocation of the shell, is to presume a concatenation will ALWAYS occur for the $PATH variable and do the following...

export PATH="$PATH:<foo>"

Where <foo> is the entirety of the path rules you want to extend things with.  As it is, you're sort-of breaking all the profile PATH setup files that follow you and doing a LOT more complicated scripting work than is needed.

madscientist42, 2017-03-29:
Proposed patch for the enhancement requested, attached...
[windows_scripting_fixes.txt](https://github.com/espressif/esp-idf/files/879538/windows_scripting_fixes.txt)

projectgus, 2017-03-29:
Hi @madscientist42,

Thanks for sending this.

Regarding accidentally expanding $PATH in the context of the setup script rather than the profile script, that's a serious bug that we will fix ASAP. Thanks for pointing it out.

Regarding the other changes, I appreciate the desire to simplify the setup experience. However I think the patch has are some expectations about the way people will use this script that aren't necessarily how we expect they will always use it. It'd be easier to discuss this in-context as a github pull request, but I'll go through the changes here:

* Passing `-o` to unzip means the script may clobber existing files without prompting the user. Prompting is the cautious approach here, if someone has an existing installation they may not expect that this script will modify it. (We don't support using this script to upgrade existing installations, although I expect it will probably work.)

Regarding IDF_PATH:
* The setup guide works on the expectation that the developer may not have a 'git' command line installed at the beginning of the process (MSYS2 doesn't come with git out of the box). The [setup guide provides a link to download that single file in 'raw' format from github](http://esp-idf.readthedocs.io/en/latest/windows-setup-scratch.html#configure-toolchain-environment-from-scratch), and then once git is installed ESP-IDF can be cloned using it (or using a git GUI tool, if that's what the developer prefers).
* Developers will quite often want more than one ESP-IDF directory, which is a (subtle and not very clear reason) why that step is left as a manual one. This way if you they to mess around with IDF_PATH, they know where/how it's set.
* (You can also theoretically have more than one toolchain, but this is less common.)

* I don't really understand why you split `windows_install_prerequisites.sh` into two. Is it just the name of the script, or for readability?

madscientist42, 2017-03-30:
Regarding -o to unzip: You're clobbering them ANYWAY :-D    If you're pouring it in that way, you're going to get asked if you want to overwrite anyhow- and NOT clobbering them is a **DANGEROUS** action, to be honest- because a new version may put a mix of desired and undesired stuff in.  You're either dealing with separate toolchains, or you're updating/replacing the current one, depending on the use model there.  Having it be not the way I did it is a confusing step for a novice dev.  :D

Regarding IDF_PATH observations:

- This doesn't presume git (though I'd think pouring it into MSYS2's environment counts as a Windows requirement...  :D )  As it stands, I don't think I dealt with that aspect of things in my patch offered. 

- The patch presumes a _**default**_ IDF_PATH.  You can always export a new one- which would be what one would do if they're changing things like you're suggesting there.  They're not going to be editing their profile.d entry except for the case where they're changing their default IDF- which you'd just simply run the script in that case.  No fuss, no muss.  Let a tool help you get the right one for the desired IDF path- run the script in the one you want default.

- Theoretically more than one toolchain presumes a level where the both of us likely are (I do OpenEmbedded and OpenWRT distribution work as part of my day job and after hours putzing...)  They're not going to need past the initial hand-hold we're talking about here and they wouldn't really be using much of those sorts of scripts, even on Windows...  >:-D

- I split it apart because it's really a function that you would use for the new IDF install.  You're...unlikely...to be doing pre-requisites install on MSYS2 more than the first time.  It probably needs to be a prerequisites install, a toolchain install/update, and the other script I made, to be honest- but that's me.

madscientist42, 2017-03-30:
Anything past the basics there is presuming something that the dev is advanced enough that they need only be told that the rest of the framework is needing <X> in the advanced use cases, and your scripting there is actually in the way past setting up prereqs. 

It sets up things like pathing that needs to not be there except in the default main use case, for example.  An advanced user using multiple path stories, etc. will have something within their makefiles or build scripting that will force toolchain selection outside of the path rules you're talking about here and force IDF_PATH regardless of the environment set up by profile.d or profile in MSYS2.

As such, it probably needs to be broken up further into a "do default setup" that does the whole process, and then three phase scripts like the one I broke off, the toolchain setup/install, and the prerequisites.  Make it as painless as possible for a first-time user, make it as painless and useful as possible for someone who's more advanced.  And the bunch for the most advanced use cases just need good documentation, which is slightly lacking right now on this subject- a bit of how-to's for that sort of thing so that someone trying to step up to this level so that if you follow <X> steps, this will work right out of box.

madscientist42, 2017-03-30:
As for the Github pull request, I probably ought to learn how to properly do that- because I suspect for a while, at least, I'll be off and on having these sorts of suggestions/discussions with you guys...  :-D

projectgus, 2017-03-30:
> As such, it probably needs to be broken up further into a "do default setup" that does the whole process, and then three phase scripts like the one I broke off, the toolchain setup/install, and the [...]

We agree, which is why we have two paths (these used to be munged up in the same doc, but we've started splitting them out and @krzychb has some plans to split them further):

Basic Windows setup: http://esp-idf.readthedocs.io/en/latest/windows-setup.html

Advanced Windows setup: http://esp-idf.readthedocs.io/en/latest/windows-setup-scratch.html

If you're running the `windows_install_prerequisites.sh` script then you're already on the alternative "harder than it has to be" set of steps.

Would you suggest a shorter/simpler set of steps for the "easier" Windows steps (where you download the .zip file)?

Or are there reasons/shortcomings that you think are making people choose the harder set of steps (windows_install_prerequisites.sh), even though they just want to get up and running ASAP?

Angus

madscientist42, 2017-03-31:
I'm not going to tell you your business.  I will, however, offer insights from 3+ decades of doing this class of stuff.  This includes being a principal build engineer for Qualcomm's FOSS endeavors.

1. What seems "easy" is more along the lines of "easy" for the people making the toolset.  It's no different than here. 

2. A monolithic install doesn't work terribly well for someone who has MSYS2 already installed.  It makes things messier and more complicated.  As an observation, with this class of device, only about 1/3rd of your actual IDF customers are going to use this at best.  What you're calling "easy" is just that.  For any developers with MSYS2 installed for any other reasons, you just made their life difficult and they won't go down that path (*waves hand*)

3. In order to sell this to developers like myself, you need two tiers within what you're calling the "advanced" path.  Something you can probably rightly call the "intermediate" (and probably most commonly used) and "advanced" that properly is that.

4. "Intermediate" should be a series of automation scripts that let you gracefully, within the dev environment on at least Windows, install/update prerequisites for your toolchain and the IDF, install/update the toolchain easily and safely for a single instance of the IDF, update the IDF correctly (since the IDF will be poured somewhere from an archive bundle or from GitHub...), and instantiate a default configuration that is useful for the developer for that install of the IDF (specifically, a path to your toolchain install for that instance of the IDF, and the path to the IDF install itself.).  You can bundle all of those together with a first-time run script.  This allows someone to have a good single instance of the IDF on their machine that works every time within MSYS2.  This should be the norm for most use cases of your product and allows someone with the bundled set install, drive it the "right" ways once they've installed the monolithic bundle.  

5. A slightly more sophisticated way for the previous is to allow the user to select which IDF version (released version lines in the sand, pre-releases that are properly tagged (v2.0-rc2...), and bleeding edge stuff that you're offering)

6. "Advanced" needs to be a proper documentation of how to drive it in the most sophisticated use-cases.  Literally how one does things to have multiple toolchains, multiple IDF's, and suggestions for automating that sort of configuration's selection either as a per-build or a gear-shift script in MSYS2 and other OSes.

Ideally, you want to have this for each OS possibility.  For OSX/Linux, it'd be mostly POSIX type scripting that did git pulls, proper toolchain fetches, updates, etc.  You can state the prerequisites for those users and not need/have a script for them- because there's enough differences in packaging that you couldn't do it like you do for MSYS2 on Windows.  The update phases for the toolchain and the IDF are so similar that you can have a common scripting for those, and maybe even the profile.d part.  Not sure about that one, because I can't tell you if OSX or *BSD do that like most Linux distributions do.  It's easy enough to find out.  If so, you have three of the four basic phases available to automate and make "easy" for most user.

You want it, though, in the four phases I describe for the intermediate step.

- prerequisites install/update
- toolchain install/update
- IDF update
- profile.d like setup of defaults.

On a first install, you want to do all of them.  If you want to track latest of IDF, where the IDF has changed but the toolchain hasn't, you just want to run the IDF update process.  Yes, you can issue the commands; they're documented and straightforward.  You know, though, that as professionals, we're intrinsically "lazy"- if I can issue one script command that encapsulates two or three git commands that takes less typing, we can get on to programming again quicker.  :-D 

Let me mull a bit on this.  I can give you some good examples on all of this for you to contemplate.

projectgus, 2017-04-04:
Thanks for the suggestions and the feedback. I agree with pretty everything you're saying here.

One point to make is that, I guess, the expectation we have is that most Windows developers won't be familiar with MSYS2 or with Bash command lines at all. So in your particular case, I think being an existing MSYS2 user who understands what that means and what having multiple msys sysroots installed may mean probably puts you in the "Advanced" tier by default. I'm not saying we can't do more to make things more comfortable for MSYS2 users to use IDF, but as a demographic I suspect you're in the minority among the Windows install base - and probably also the demographic who is most able to understand the existing install script and figure out how to adapt it to their needs. :)

The other suggestions about having better tooling, one click "run this to install", etc. is all excellent. I couldn't agree more. So far, our situation has been that of a small team trying to roll out the best cross-OS support we can with the time & resources we have available. Eventually, producing something like what you're describing would be fantastic. We have some ambitions around developing an "IDF package manager" so it may be possible to roll prerequisite support, toolchain versions, etc into this - at least as much as we can without eclipsing/reinventing existing OS-level package management systems.

madscientist42, 2017-04-04:
Oh, I think you'll find I'm a Lazy Guru level user here.  Client in one of my cases insists on Windows, so I have to do it there.  Linux setup's been a dream.  Done right for what it's worth.  Windows...it needs a bit of refinement and I've been trying to approach it from the perspective of a novice but still a professional that would be doing a bit  more than just pouring the all-in-one into the system.  I think I have an answer or three there.  Revolves around MSYS2 and a bit of scripting.  Since you need that anyhow, and you're telling them about pacman, there is a good possibility of a tool that welds your own pacman repository off off of GitHub's stuff.  If they're going to use the IDF as you've currently designed it, the devs will, by default, have to get at least passingly conversant with bash and MSYS2's functions anyhow.  Unless you're going to go whole-hog supporting Windows through PlatformIO, you're faced with that reality.  Might as well make it go well for all, right?  :-D

feedthecat, 2017-04-06:
Madscientist is right. I re-installed over an existing msys32 and it failed. I found the fix here https://github.com/espressif/esp-idf/issues/200

projectgus, 2017-04-10:
@madscientist42 Adding an msys repository could be a nice hybrid fix. Keen to hear the other suggestions you have.

> the devs will, by default, have to get at least passingly conversant with bash and MSYS2's functions anyhow

At the moment, yes. If you use Eclipse then you can do everything "the Windows way" up to but not including "make menuconfig", as this currently has to be run from the MSYS terminal.

@feedthecat:
> Madscientist is right. I re-installed over an existing msys32 and it failed. I found the fix here #200

Can you give any more details about the failure you saw? I'm not able to reproduce, updating an older install via windows_install_prerequisites.sh succeeds for me. There are two things to note:

* The first time MSYS updates itself, it gets stuck in a situation it can't continue (lots of errors like `cygheap base mismatch detected`...) This is mentioned in the docs under [During the initial update step...](http://esp-idf.readthedocs.io/en/latest/windows-setup-scratch.html). There isn't any easy fix for this that I know of, MSYS sometimes updates itself to the point where it can't continue. Closing the window and re-opening, then re-running the update, allows it to complete.
* The unzip prompts to replace files, as there is already a toolchain installed. Answering 'A' (All) allows it to complete.

madscientist42, 2017-04-11:
@projectgus 

> At the moment, yes. If you use Eclipse then you can do everything "the Windows way" up to but not including "make menuconfig", as this currently has to be run from the MSYS terminal.

Until you have a "plugin" framework, like might be trying to gel in the PlatformIO world, on Eclipse, you will never have it "the Windows Way".  It must all be point and click from within the IDE in question for that to count- and the moment you're having to do a "make menuconfig" you're off into the weeds there on that notion.  I'd opine that you're more likely to get it there with PlatformIO/Atom than you will with Eclipse without stepping up to the plate and doing an Eclipse plugin yourselves.  (I think we both know what THAT answer is...  :-D )  

This doesn't get into the "Windows Way" not working terribly well in a CI or production server world.  Even VS has command line tools for that purpose.  If you're going to be forced to that anyhow...why not make it the smoothest experience possible?

Re your answer to feedthecat: 

This is why I did what I did in my proposed patch on the unzip part.  You're telling them to do an "A" answer anyhow in most cases.  It's confusing to make people do extra steps that would be taken care of and simply documented on- if they're following advice, the concern about overwrite just went out the window anyhow...  Why present them a useless choice that will pretty much be the same as if you'd just done it yourself in the scripting?

madscientist42, 2017-04-11:
Let me frame in at least the first couple of ideas for you guys as pull requests so this can work with the right workflow on GitHub and what you're geared up for.  We can discuss it all, pros/cons on it and any other suggested items on other pull requests.

feedthecat, 2017-04-12:
I think maybe I didn't install over the existing msys32. I'm new to msys32, and I wanted to keep some of the work I'd done exactly as it was, so I renamed the old install, and installed the new version into a new msys32 directory. The result was some of the install didn't go (python, etc.), presumably it detected python was already installed because of the old install, but couldn't find it from within the new install - I first found the problem when trying to compile one of the samples.

projectgus, 2018-12-04:
Closing this as the issue with PATH in the windows_install_prerequisites.sh script is long solved, and we're moving ESP-IDF away from requiring MSYS2 in order to work around a lot of the other issues touched on here - the [IDF Tools installer for CMake](https://docs.espressif.com/projects/esp-idf/en/latest/get-started-cmake/) sets up an MSYS2-free environment, and eventually it can be extended to also "git clone" ESP-IDF and set up the IDF_PATH variable automatically.

closed 2018-12-04 by projectgus