espressif/esp-idf #469: Issue with Bluetooth and UART0

rajeshchavan88, 2017-03-29:
We are using esp32 devkit for our experiments. While experimenting we have observed that if i turn the Bluetooth functionality(basically esp_bt_controller_init()), UART0 stops receiving data. Neither uart_rx_one_char() or uart_rx_one_char_block() functions work as expects. However the transmit functionality of UART0 is intact, we can see the messages sent by ESP32 SDK( messages sent using ESP_LOG() APIs ). 
If i do not turn on Bluetooth functionality UART simply works fine.
While browsing online i read somewhere that BT RF is affecting the UART functionality (not able to find the thread now).
I am curious to know if this is known silicon issue with simultaneous functionality of UART0 and Bluetooth?

Thanks and Best Regards
- RChavan

madscientist42, 2017-03-30:
Interested in this.  Have you looked at what UART1 is doing in this same regard?  I'd love to know so I can have a workaround for one of my projects, if it does work right as opposed to how UART0 seems to be working for you.

rajeshchavan88, 2017-03-31:
Now I have configured UART1 of ESP32 and same issue is getting observed. UART transmit functionality of ESP32 is intact while reception hangs when BT turned ON. UART functionality is critical for our project and hence just wanted to confirm, if this is really an issue or if yes is there some workaround available?

madscientist42, 2017-03-31:
Annoying.  I was about to work on something once I got a bit of a success in the code I was porting over that needed this ability as well.  

Need to know if we've got a silicon or IDF bug here, guys- and be handed workarounds if it is.  I need UART support to bridge two ESP32's with BT/BLE running on them.

madscientist42, 2017-03-31:
SPI might work, but it's clumsier for what I'm attempting.

madscientist42, 2017-04-04:
What's the timeline for this?  I've got a few projects where this would be useful...  I know, I'm being a bit of a pest, but...  >;-D

rajeshchavan88, 2017-04-04:
@TianHao-Espressif  I'm bit unclear here. Does this mean we can not use UART0 and bluetooth simultaneously? Just for information what kind of BT chip we have inside ESP32 and does ESP-IDF(and/or silicon implementation) uses the same pins/pads to communicate with BT peripheral which are also configured for UART?

TianHao-Espressif, 2017-04-06:
@rajeshchavan88 , There's something unclear, do you want to use UART0 as HCI when bluetooth enable? or just user UART0 do another things when bluetooth enabled?

TianHao-Espressif, 2017-04-06:
@rajeshchavan88 , Maybe the interrupt number of UART0 receive do something conflict with bluetooth's (Bluetooth use 1/5/8). Please check that.

dpgeorge, 2017-04-06:
> Maybe the interrupt number of UART0 receive do something conflict with bluetooth's (Bluetooth use 1/5/8). Please check that.

I had a similar problem with BT+UART0 not working together and this was indeed the issue: both BT and UART0 use ISR number 5.  To fix it I used uart_isr_register to register the handler, rather than the low-level function xt_set_interrupt_handler.

rajeshchavan88, 2017-04-06:
@TianHao-Espressif  I want to use UART0 as simple UART and at the same time I want BT is also enabled.
@dpgeorge I will try out your solution and will check if it works for me. Thanks

jkhax0r, 2017-04-13:
Any news on if this works?  Basically can we use 2 UART ports (one for serial/log and one for comms), while BT is in use?

TianHao-Espressif, 2017-04-13:
BT doesn't effect UART. rajeshchavan88's problem is interrupt number conflict with bt.

TianHao-Espressif, 2017-05-03:
@rajeshchavan88 ,  sorry for disturbing you. So your problem is fixed?

rajeshchavan88, 2017-05-03:
Hello @TianHao-Espressif, as mentioned by @dpgeorge BT and UART  does use same interrupt number. after changing the interrupt number, both BT and UART are working fine. Thanks ! :)

closed 2017-05-03 by rajeshchavan88