espressif/esp-idf #470: Partition table needs factory partition

forthlightning, 2017-03-29:
[This partition table documentation](http://esp-idf.readthedocs.io/en/latest/partition-tables.html) says that we don't need to include a factory partition if we have ota_0 and ota_1.

So I make a partition table like so:
```
# Name,   Type, SubType, Offset,  Size
nvs,      data, nvs,     0x9000,  0x6000
phy_init, data, phy,     0xf000,  0x1000
ota_0,    app,  ota_0,   0x10000, 1M
ota_1,    app, ota_1,    ,        1M
spiffs,   data, nvs,     ,        0x80000
```

and I do `make partition_table`

and I get

```
*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
ota_0,app,ota_0,0x10000,1M,
ota_1,app,ota_1,0x110000,1M,
spiffs,data,nvs,0x210000,512K,
*******************************************************************************
```

But when I `make flash monitor`, I get this:

```
--- idf_monitor on /dev/tty.SLAB_USBtoUART 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:2488
load:0x40078000,len:7128
load:0x40080000,len:252
0x40080000: _WindowOverflow4 at ??:?
entry 0x40080034
0x40080034: _WindowOverflow4 at ??:?
E (44) boot: nothing to load
user code done
```

If I just rename the partition to "factory" then everything is fine, but then OTA will not work correct? 

Is this user error or what? I am trying to save as much room for SPIFFS as possible.

Thanks much!

negativekelvin, 2017-03-29:
You could initialize the otadata section to point to ota0 or you could modify the bootloader. But it does seem like a good idea for the bootloader to boot the first firmware partition by default and not necessarily require it to be factory.

forthlightning, 2017-03-30:
ooo my bad... I forgot the otadata partition!

just tried this partition.csv and it worked.

```
# Name,   Type, SubType, Offset,  Size
nvs,      data, nvs,     0x9000,  0x4000
phy_init, data, phy,     0xd000,  0x1000
otadata,  data, ota,     0xe000,  0x2000
ota_0,    app,  ota_0,   0x10000, 1M
ota_1,    app,  ota_1,   ,        1M
spiffs,   data, nvs,     ,        1M
```

closed 2017-03-30 by forthlightning

negativekelvin, 2017-03-31:
I also missed that in your post. Good to see that it already defaults to ota0 if no factory.