espressif/esp-idf #471: running make gives "error: option `porcelain' takes no value"

MrBuddyCasino, 2017-03-30:
After pulling the latest master version I get this now when running make:

`master$ make
error: option 'porcelain' takes no value
usage: git status [<options>] [--] <pathspec>...

    -v, --verbose         be verbose
    -s, --short           show status concisely
    -b, --branch          show branch information
    --porcelain           machine-readable output
    --long                show status in long format (default)
    -z, --null            terminate entries with NUL
    -u, --untracked-files[=<mode>]
                          show untracked files, optional modes: all, normal, no. (Default: all)
    --ignored             show ignored files
    --ignore-submodules[=<when>]
                          ignore changes to submodules, optional when: all, dirty, untracked. (Default: all)
    --column[=<style>]    list untracked files in columns`

The build still works, but its quite annoying.

MrBuddyCasino, 2017-03-30:
Environment: 
git version 2.8.3
macOS Sierra 10.12.3

negativekelvin, 2017-03-30:
You have this change?
https://github.com/espressif/esp-idf/commit/dbb6d20aac0be134987ca73d45a73bdc248605a8

MrBuddyCasino, 2017-03-30:
Yup that fixed it. :-)

closed 2017-03-30 by MrBuddyCasino