espressif/esp-idf #472: vRtosStatsReport() returns incorrect task info

ammaree, 2017-03-30:
In my effort to understand the tasks created by ESP-IDF I found what I believe is a bug. The situation only occurs once and only when uxTaskGetSystemState() is called immediately after the last task has been created.

The situation can be simulated easily and the results are consistent
xTaskCreatePinnedToCore(&blink_task, "TskBlink", 2048, NULL, 5, NULL, 1) ; xTaskCreatePinnedToCore(&sntp_task, "TskSntp", 2048, NULL, 5, NULL, 1) ;
while(1) { vRtosStatsReport() ; vTaskDelay(pdMS_TO_TICKS(10000)) ; }

Num1 is the number of uxTaskGetNumberOfTasks() + 2.
Num2 is the value returned by uxTaskGetSystemState().
Both values are obtained right at the top of the function.

On the first call to vRtosStatsReport() Num2 = 14 (or 15) and the second last task created "TskBlink" exists twice in the array filled with uxTaskGetSystemState(), sometimes 1 of the 2 idle tasks are also duplicated.

On the second call to vRtosStatsReport() Num2 = 13 and the second last task created "TskBlink" exists only once in the array filled with uxTaskGetSystemState().

After the second call the task state remains consistent and correct, no matter how long the code runs.
Inserting a vTaskDelay(pdMS_TOTICKS(1)) before the while(1) is sometimes sufficient to make the problem go away.

Console output below:

I (1687) event: ip: 192.168.1.76, mask: 255.255.255.0, gw: 192.168.1.2
Num1=14 Num2=14
------Name------|MCU|F# | Pc/Pb | S | LoStk |
ipc0| 0 | 1 | 24/24 | B | 4,176 |xIdx=13
ipc1| 1 | 2 | 24/24 | B | 3,632 |xIdx=8
main| 0 | 3 | 1/ 1 | R | 8,992 |xIdx=2
IDLE| 0 | 4 | 0/ 0 | P | 3,504 |xIdx=4
IDLE| 1 | 5 | 0/ 0 | P | 3,456 |xIdx=3
Tmr Svc| 0 | 6 | 1/ 1 | B | 7,008 |xIdx=7
tiT| X | 7 | 18/18 | B | 4,720 |xIdx=6
eventTask| 0 | 8 | 20/20 | B | 1,584 |xIdx=12
ppT| 0 | 9 | 23/23 | B | 5,776 |xIdx=11
pmT| X |10 | 21/21 | B | 9,024 |xIdx=9
rtT| X |11 | 22/22 | B |11,920 |xIdx=10
TskBlink| 1 |12 | 5/ 5 | P | 7,008 |xIdx=0
TskBlink| 1 |12 | 5/ 5 | B | 7,008 |xIdx=5
TskSntp| 1 |13 | 5/ 5 | P | 5,612 |xIdx=1

Num1=14 Num2=13
------Name------|MCU|F# | Pc/Pb | S | LoStk |
ipc0| 0 | 1 | 24/24 | B | 4,176 |xIdx=11
ipc1| 1 | 2 | 24/24 | B | 3,632 |xIdx=12
main| 0 | 3 | 1/ 1 | R | 8,992 |xIdx=0
IDLE| 0 | 4 | 0/ 0 | P | 3,504 |xIdx=2
IDLE| 1 | 5 | 0/ 0 | P | 3,456 |xIdx=1
Tmr Svc| 0 | 6 | 1/ 1 | B | 7,008 |xIdx=6
tiT| X | 7 | 18/18 | B | 4,016 |xIdx=5
eventTask| 0 | 8 | 20/20 | B | 1,584 |xIdx=7
ppT| 0 | 9 | 23/23 | B | 5,776 |xIdx=10
pmT| X |10 | 21/21 | B | 3,968 |xIdx=9
rtT| X |11 | 22/22 | B |11,920 |xIdx=8
TskBlink| 1 |12 | 5/ 5 | B | 6,496 |xIdx=3
TskSntp| 1 |13 | 5/ 5 | B | 2,176 |xIdx=4

Andre

igrr, 2017-06-20:
Fixed in https://github.com/espressif/esp-idf/commit/1c798b0e.

closed 2017-06-20 by igrr

ammaree, 2017-06-20:
Much appreciated…

Fixed in 1c798b0<https://github.com/espressif/esp-idf/commit/1c798b0e>.