espressif/esp-idf #473: rmt_driver_uninstall() reports error when executed for an RX channel

devsaurus, 2017-03-30:
I get error logs when executing `rmt_driver_uninstall()` for an RX channel.
This happens due to the check in `rmt_set_tx_thr_intr_en()` on parameter `evt_thresh`. It is set to `0xffff` by `rmt_driver_uninstall()` which is out of range.

Happens most likely also for a TX channel (where the abort is more serious) but I haven't tried that.

nilay994, 2017-05-15:
What are the logs about ?

If I am understanding this correctly, 
During `rmt_driver_uninstall`, the check in `rmt_set_tx_thr_intr_en()` for parameter `evt_thresh` is skipped and should not print any log when`evt_thresh` is out of limit.

devsaurus, 2017-05-15:
Yes, was fixed by #492, closing.

closed 2017-05-15 by devsaurus