espressif/esp-idf #474: esp-idf make flash problem

icelik1, 2017-03-30:
Dear all I have problem about make flash my project. I can make menuconfig and i can set the menu but when i run make flash make return this messages.
`icelik@icelik-ubuntu:~/esp/myapp$ make flash
CC intr_alloc.o
/home/icelik/esp/esp-idf/components/esp32/./intr_alloc.c: Assembler messages:
/home/icelik/esp/esp-idf/components/esp32/./intr_alloc.c:697: Error: unknown opcode or format name 'xsr.INTENABLE'
/home/icelik/esp/esp-idf/components/esp32/./intr_alloc.c:700: Error: unknown opcode or format name 'wsr.INTENABLE'
/home/icelik/esp/esp-idf/components/esp32/./intr_alloc.c:715: Error: unknown opcode or format name 'xsr.INTENABLE'
/home/icelik/esp/esp-idf/components/esp32/./intr_alloc.c:718: Error: unknown opcode or format name 'wsr.INTENABLE'
/home/icelik/esp/esp-idf/make/component_wrapper.mk:176: recipe for target 'intr_alloc.o' failed
make[1]: *** [intr_alloc.o] Error 1
/home/icelik/esp/esp-idf/make/project.mk:386: recipe for target 'esp32-build' failed
make: *** [esp32-build] Error 2
`
thanks your advice

Spritetm, 2017-03-30:
Sounds like something has gone terribly wrong with your toolchain. Were you able to compile and flash things before? If so, what did you change since then? If not, which instructions did you follow to set up esp-idf and the toolchain?

madscientist42, 2017-03-30:
Seeing something like that (Which I did just yesterday, while I was working towards using a Git pull of the IDF for the 2.0 branch...), you probably need to re-install with a known line in the sand (tarballs?) of the toolchain and the IDF and clean house.  Once I re-installed everything and found the line in the sand (some of the latest commits in the 2.0 pre-release branch _**WILL**_ mess you up short-term, so watch out...) it all started building correctly, including the make flash parts.

icelik1, 2017-03-30:
Dear Spritetm, 
I follow thatin the  http://esp-idf.readthedocs.io/en/latest/linux-setup.html instructions and when i install i had never get any error  message.
Dear madscientist42 i will try re-install and re-comment. 
thanks your advices

projectgus, 2017-03-30:
Hi @icelik1 ,

If you haven't yet deleted the broken install, could you please try running "make V=1" and upload the full log output to gist.github.com or similar? If multiple people have seen this kind of error then we'd like to fix/avoid it if possible.

Angus

icelikTTT, 2017-03-31:
Dear All,
I reinstall all toolchain but it does not work and return same error messages.  What i am doing wrong?

igrr, 2017-04-05:
Could you please paste the output of these two commands?
```
xtensa-esp32-elf-gcc --version
cat /dev/null | xtensa-esp32-elf-gcc -c -x assembler-with-cpp - -Wa,--version
```

igrr, 2017-05-03:
@icelik1 could you please attach the output of the commands mentioned above? Thanks.

OSesveren, 2017-05-03:
@igrr I have the same problem..

xtensa-esp32-elf-gcc --version

xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-61-gab8375a) 5.2.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-----
cat /dev/null | xtensa-esp32-elf-gcc -c -x assembler-with-cpp - -Wa,--version

GNU assembler (crosstool-NG crosstool-ng-1.22.0-61-gab8375a) 2.25.1
Copyright (C) 2014 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or later.
This program has absolutely no warranty.
This assembler was configured for a target of `xtensa-esp32-elf'.

OSesveren, 2017-05-03:
The problem is caused by the Turkish language settings. The compilation is complete when it is in English.

igrr, 2017-05-03:
Which OS version are you using?

OSesveren, 2017-05-03:
Ubuntu 17.04

phantomxe, 2017-08-24:
On Linux Mint Serena and Turkish language environment, needs some extra commands.
For make
> Exec=env LANG=en_US.UTF-8 make

closed 2018-01-11 by FayeY