espressif/esp-idf #475: gatt client open issue

Nicholas3388, 2017-03-31:
Hi,
I want to connect to a BLE device using ESP32. I use PASSIVE scan type for scanning, and I can scan my BLE device's advertising. But when I try to invoke esp_ble_gattc_open to connect to the device, I got the following error: 
[0;31mE (2201) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x003e[0m
[0;31mE (2201) BT: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x003e[0m
[0;31mE (2201) BT: btm_sec_disconnected clearing pending flag handle:0 reason:62

esp_ble_gattc_open return 0. It seems that gatt open without problem. But when the ESP_GATTC_OPEN_EVT event occurs, the following error output:

GATTC_DEMO: ESP_GATTC_OPEN_EVT conn_id 0, if 4, status 133, mtu 0

The status 133 means ESP_GATT_ERROR. There is no more info to help debug. Can anyone provide some suggestion to solve it?

P.S. the target device to connect is nRF51822, and I can connect to it successfully via BLE debug tool on Android phone.

jkhax0r, 2017-03-31:
I am having the exact same issue.  I'll keep working on it but please let me know if you figure anything out

Nicholas3388, 2017-04-01:
@jkhax0r Hi, I can connect to nRF51822 by calling esp_ble_gattc_close to close the connection, and then reopen gattc by calling esp_ble_gattc_open. This is not a perfect solution, but it works. If this is a bug, I hope the espressif staff can fix it for a new version of esp-idf.

Nicholas3388, 2017-04-01:
@jkhax0r  If you have another better solution to connect to the device, please do let me know. Thanks;)

Nicholas3388, 2017-04-02:
Still no one can provide some advice on this issue?

jkhax0r, 2017-04-02:
I've made little progress.  Can you do me a favor... try setting the scan parameters to ACTIVE.  Can you even see your device?  

With my device I can't.  Best I can tell, I get the initial advertising packet, but, for a "connectable" or "scannable" device, the stack does not send it through to the callback (see components/bt/bluedroid/stack/btm/btm_ble_gap.c line 2623) until the scan request is sent and scan response is seen.   

So basically the stack is not sending, or my device is not replying to the scan request/response packets.  I could have sworn that yesterday it was seeing my device during an ACTIVE scan, but very very rarely. 

What device or demo kit do you have (i.e. ESP-WROOM-32, Sparkfun Thing board, etc)?

And for what its worth, my device scans and connects no problem in nRF Connect (android scanner app).  Its a Bluegiga BLE112 module, using script code, so there's very little room for me to screw up the code relating to the BT stack.

Similarly, when I "clone" my device in nRF Connect and then setup a virtual advertiser, the ESP32 will connect no problem.  

So somehow the ESP32 just does not play well with certain devices.  

I suspect this is all related to #421 too.

Nicholas3388, 2017-04-03:
@jkhax0r Hi, thanks for your response, My device is ESP-WROOM-32. If I change the scan type to ACTIVE, the ESP32 is not able to scan the target device. The target device (nRF51822) is developed by someone else, I cannot get its sources, all I known is that it is advertising ibeacon and it will response when it received a scan request. 

I can connect to the target and then search all services and characteristics it provided, if I use some BLE debug app on Android phone without problem. But I cannot access them by ESP32, the issue just as I post previously. This is what I am confused. 

I used BLE packet sniffer to catch packets sent between ESP32 and the target device. I found that no scan request is sent by ESP32, so there is no scan response. When I sniff the packets between  Android phone and the target, a scan request is sent by Android and follow by a scan response sent by the target, then the connection is established between the two devices.

Nicholas3388, 2017-04-03:
If scan response is necessary for connection establishing, I don't know why the scan request is not sent when the scan type is ACTIVE in this case.

jkhax0r, 2017-04-03:
That is my exact situation. I have a custom board with esp-wroom-32.  Seems
like a clear bug.

I do not have a sniffer but best I could tell either no scan request was
being sent/received or no scan response was sent/received when in active
scanning, So it never even sees the device.  I'm sure this is related to
our connection problems too when using passive scanning.

If I change the device to unscannable/unconnectable it is found, as I would
expect since no scan request is sent.

What is real interesting is that I got another one of my peripheral devices
and programmed it with the exact same code and it works well with the same
esp32 board.  Scanning, connecting, and everything.

If anyone from EspressIF is open, I'd be happy to send my "bad" device if
you guys can work on this. It seems like it's down beneath the hci level
either in the library or hardware because I enabled all debug messages in
the stack and the only difference I could see is that the second hci event
for the scan response just never comes with my "bad" device.  I can also
send my good device if it helps.

I'll try on a few more tomorrow when I get into the office to see if I can
find a trend.  I have 100s of them.  Again, I'm sure this is related to
#421 as well as another issue I posted re: sporadic scanning (but of an
unconnectable/non scannable device at the time).

Nicholas, what sniffer do you use and would you recommend it?

jkhax0r, 2017-04-03:
Oh and this is a ble113 based design
  Just to make sure my code wasn't somehow to blame, I programmed it with Bluegigas heart rate demo with the same bad results.

Nicholas3388, 2017-04-04:
@jkhax0r  Hi, my sniffer is an nRF51822 dev-board, bought from Taobao China. The sniffer is easy to use, but I'm afraid you cannot buy it, if you are not in China. I bought the sniffer from the following link: https://item.taobao.com/item.htm?spm=a230r.1.14.40.mB3Ga1&id=545629969442&ns=1&abbucket=13#detail

Nicholas3388, 2017-05-16:
@jkhax0r  Hi, is there any progress for this issue?

Yulong-espressif, 2017-06-03:
@Nicholas3388 Can you show more detail of the log? 

#ifndef BTM_INITIAL_TRACE_LEVEL
#define BTM_INITIAL_TRACE_LEVEL             BT_TRACE_LEVEL_DEBUG
#endif

#ifndef L2CAP_INITIAL_TRACE_LEVEL
#define L2CAP_INITIAL_TRACE_LEVEL           BT_TRACE_LEVEL_DEBUG
#endif

#ifndef GATT_INITIAL_TRACE_LEVEL
#define GATT_INITIAL_TRACE_LEVEL             BT_TRACE_LEVEL_DEBUG
#endif

Can you change the GATT, L2CAP & BTM trace level from BT_TRACE_LEVEL_WARNING to BT_TRACE_LEVEL_DEBUG in the bt_trace.h file. When you changing this and then catch the log again.

And have you have a BLE Capturing packet tools? If you have, mabey you can show the log of the BLE air packect with the log together.

FayeY, 2017-06-09:
Hi @Nicholas3388, is there any update?

Nicholas3388, 2017-06-12:
@FayeY  Nothing new, problem is still not solved

wzaggle, 2017-07-19:
Having same problem connecting to Minew beacons if I try to connect to them.

wzaggle, 2017-07-19:
Having the same problem trying to connect to a Minew brand iBeacon to get the battery level.

jkhax0r, 2017-08-07:
There's sort of two issues here but quick update is that I am also still seeing this connect error "reason=0x003e" on a BLE113 (CC2540 chipset) based device.  Happens maybe 1 in 3 times.  If I just close and reopen it eventually works.

Weijian-Espressif, 2017-09-25:
hi @Nicholas3388 , you can refer https://github.com/espressif/esp-idf/issues/770.

FayeY, 2017-10-10:
@Nicholas3388 , did the above suggestion help you resolve the issue?

closed 2017-10-11 by Nicholas3388

Parthesh, 2017-10-23:
Hi @Nicholas3388  Did you find the complete solution of the above problem? If so please share with me.
Thank you.