espressif/esp-idf #477: spi_device_transmit doesn't return when changing host ID

MrSurly, 2017-03-31:
I'm working on [implementing HW SPI support](https://github.com/micropython/micropython-esp32/issues/58) for MicroPython on ESP32.

I've run into an issue where `spi_device_transmit` never returns.

1.  Initialize bus, add device
1.  Send data (works)
1.  remove device, free bus
1.  change host from 1 to 2
1. Initialze bus (with different host), add device
1. Send data (`spi_device_transmit` never returns)

I've run into an issue with regards to using an SPI host, then deconfiguring it, reconfiguring using a different host, and then the IDF freezes on `spi_device_transmit`.

    spi_bus_config_t                bus_config;
    spi_device_interface_config_t   device_config;
    spi_device_handle_t             spi;
    spi_host_device_t               host = 1;
    int                             dma = 1;

    memset(&bus_config, 0, sizeof(spi_bus_config_t));
    memset(&device_config, 0, sizeof(spi_device_interface_config_t));

    bus_config.miso_io_num = -1;
    bus_config.mosi_io_num = 26;
    bus_config.sclk_io_num = 25;
    bus_config.quadwp_io_num = -1;
    bus_config.quadhd_io_num = -1;

    device_config.clock_speed_hz = 50000;
    device_config.mode = 0;
    device_config.spics_io_num = -1;
    device_config.queue_size = 1;
    device_config.flags = SPI_DEVICE_TXBIT_LSBFIRST | SPI_DEVICE_RXBIT_LSBFIRST;

    assert(spi_bus_initialize(host, &bus_config, dma) == ESP_OK);
    assert(spi_bus_add_device(host, &device_config, &spi) == ESP_OK);

    struct spi_transaction_t transaction = {
        .flags = SPI_TRANS_USE_TXDATA | SPI_TRANS_USE_RXDATA,
        .length = 16,
        .tx_buffer = NULL,
        .rx_buffer = NULL,
        .tx_data = {0x04, 0x00}
    };

    printf("before first xmit\n");
    assert(spi_device_transmit(spi, &transaction) == ESP_OK);
    printf("after first xmit\n");

    assert(spi_bus_remove_device(spi) == ESP_OK);
    assert(spi_bus_free(host) == ESP_OK);

    // change things up!
    #if 0
        bus_config.mosi_io_num = 4;
        printf("changing mosi to %d\n", bus_config.mosi_io_num);
    #endif
    #if 1
        host = 2;
        printf("changing host to %d\n", host);
    #endif
    
    assert(spi_bus_initialize(host, &bus_config, dma) == ESP_OK);
    assert(spi_bus_add_device(host, &device_config, &spi) == ESP_OK);

    printf("before second xmit\n");
    assert(spi_device_transmit(spi, &transaction) == ESP_OK);
    printf("after second xmit\n");

Spritetm, 2017-04-01:
I'll look into this, I have the SPI code open for surgery anyway.

MrSurly, 2017-04-01:
> I'll look into this, I have the SPI code open for surgery anyway.

Thank you.  For MicroPython, we use this commit hash for ESP-IDF:

`c06cc31d85cc700e1dbddbe527d4282c4bc5845a`

MrSurly, 2017-04-03:
Okay, I looked into this a bit.

Two things:

* No data is ever transmitted on the second call to `spi_device_transmit`
* This line in `spi_device_get_trans_result` never returns: 
``` r=xQueueReceive(handle->ret_queue, (void*)trans_desc, ticks_to_wait);```

This would tend to indicate that the ISR isn't running the first time to initiate the xfer; why it does this only when changing host -- I don't know =P

MrSurly, 2017-04-27:
@igrr Thanks!

igrr, 2017-12-14:
Sorry for lack of updates, this has been fixed some time ago in https://github.com/espressif/esp-idf/commit/9314bf0d376fc3dfdc5b954376a40f74fd4a1d78.

closed 2017-12-14 by igrr