espressif/esp-idf #478: SPI: pin routing remains after disabling SPI

MrSurly, 2017-03-31:
This was discovered at the same time as #477, and uses the same test code.

1.  Initialize bus, add device (MOSI = 26)
1.  Send data (works)
1.  remove device, free bus
1.  change MOSI pin (MOSI = 4)
1. Initialze bus , add device
1. MOSI data appears on both pin 26 *and* pin 4!

Spritetm, 2017-04-01:
I would say that's not necesarily a bug: I think for the general case it would be a bit overkill to remember whatever the pins were connected to only so we can restore these defaults. Reverting them to GPIO is an option, maybe.

MrSurly, 2017-04-01:
> Reverting them to GPIO is an option, maybe.

Rather new to the ESP-IDF -- I cobbled this together from various sources -- but something like this?

    pin_num = 26; // MOSI in the example above
    gpio_set_direction(pin_num, GPIO_MODE_INPUT);
    gpio_matrix_out(pin_num, SIG_GPIO_OUT_IDX, false, false);
    PIN_FUNC_SELECT(pin_num, PIN_FUNC_GPIO);

Presumably, making sure the pin output isn't connected to the SPID "D" (aka MOSI) signal should be enough.  The fact that the pin input is still routed there shouldn't matter much; I assume the SPI "D" input circuit isn't used?  In any case, if you reconfigure via `spi_master.c`, then this will be re-routed anyway.

MrSurly, 2017-04-03:
Re-opening -- There doesn't seem to be any macros that will allow these pins to be used as GPIO after being configured for SPI.

MrSurly, 2017-04-03:
Okay, got this working (sheesh)

closed 2017-04-03 by MrSurly