espressif/esp-idf #479: bt_controller_tasks hangs if Controller to Host Flow Control is enabled before Host Buffer Size is set

mringwal, 2017-04-02:
When I try to enable HCI Host Controller to Host Flow Control but don't send the HCI Host Buffer Size command first, I get this log:

Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x40018cdc  PS      : 0x00060430  A0      : 0x80019218  A1      : 0x3ffcc070
A2      : 0x00000000  A3      : 0x3ffb8154  A4      : 0x3ffae710  A5      : 0x3ffae710
A6      : 0x00000014  A7      : 0x00000001  A8      : 0x80018cd0  A9      : 0x3ffcc050
A10     : 0x00000000  A11     : 0x3ffb9a20  A12     : 0x00000000  A13     : 0x3ffb8360
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace: 0x40018cdc:0x3ffcc070 0x40019218:0x3ffcc0a0 0x40017576:0x3ffcc0c0 0x40019d14:0x3ffcc0e0 0x40055b50:0x3ffcc100 0x400e413a:0x3ffcc120 0x400e4192:0x3ffcc140 0x400e3ba1:0x3ffcc160

0x400e413a: r_rw_schedule at ??:?
0x400e4192: btdm_controller_schedule at ??:?
0x400e3ba1: bt_controller_task at /Projects/esp32/esp-idf/components/bt/./bt.c:206

If Controller to Host Flow Control command requires Host Buffer Size to be set first, you could check and return a Command Complete with error, e.g. UNSUPPORTED FEATURE OR PARAMETER VALUE (0x11) or INVALID HCI COMMAND PARAMETERS (0x12)

This refers to components/bt/lib at 9a4bb1d, but there's no issue tracker on that submodule.

Yes, I'll just change the command order to avoid this hang :)

jack0c, 2017-05-18:
This should be equal to issue #480.

mringwal, 2017-05-18:
It is definitely related. Let's see if the fix mentioned in #480 fixes this as well. In any case, calling Host Buffer Size first seems like a good idea anyway.

FayeY, 2017-06-09:
#644

mringwal, 2017-08-26:
Just checked: the issue has been fixed by one of the Bluetooth Controller updates, closing it now. Thanks!

closed 2017-08-26 by mringwal