espressif/esp-idf #480: Bluetooth Controller: Memory Capacity Exceeded on Host Number Completed Packets?

mringwal, 2017-04-02:
I'm trying to use Controller to Host Flow Control. Host Buffer Size is set to 10 ACL buffers a 1000 bytes each. BTstack sends a Host Number of Completed packet for the current handle. 

In a Central test, 6 ACL packets are received and 6 HCI Host Num Completed Packets Commands are sent. After the last command, I receive a Command Complete even with status "Memory Capacity Excceeded". 

Under what conditions is this status reported? I could imagine that this error is thrown if more packets are reported as completed then have been received, but I don't see such an error in the packet log: one HCI Host Num Complete command for each received ACL packet.

I'll attach a log from the ESP32 both as ASCII:
[memory-capacity-exceeded.txt](https://github.com/espressif/esp-idf/files/888632/memory-capacity-exceeded.txt) as well as PacketLogger:
[memory-capacity-exceeded.pklg.zip](https://github.com/espressif/esp-idf/files/888635/memory-capacity-exceeded.pklg.zip), which can be opened in WireShark. 

p.s. the last ATT Read By Type Response is incorrect, but it is a well formed ACL packet and should not confuse the Controller to Host Flow Control. I'll look after that next.

mringwal, 2017-04-06:
Following up, it's not just that I get a irritating error code, the Bluetooth controller hangs up right after that when BTstack sends the next HCI Host Number Completed packet.

...
[00:00:03.678] CMD => 35 0C 05 01 00 00 01 00
[00:00:03.682] EVT <= 0E 04 00 35 0C 07
[00:00:03.705] EVT <= 13 05 01 00 00 01 00
[00:00:03.736] ACL <= 00 20 05 00 01 00 04 00 13
[00:00:03.736] EVT <= A0 03 00 00 00
[00:00:03.736] CMD => 35 0C 05 01 00 00 01 00
[00:00:03.737] ACL <= 00 20 1B 00 17 00 04
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x40018cdc  PS      : 0x00060930  A0      : 0x80019218  A1      : 0x3ffcc070
A2      : 0x00000000  A3      : 0x3ffb8154  A4      : 0x3ffae710  A5      : 0x00000000
A6      : 0x00000050  A7      : 0x00000001  A8      : 0x80018cd0  A9      : 0x3ffcc050
A10     : 0x00000000  A11     : 0x3ffbb7b0  A12     : 0x0000002c  A13     : 0x3ffb8360
A14     : 0x00000038  A15     : 0x00000001  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace: 0x40018cdc:0x3ffcc070 0x40019218:0x3ffcc0a0 0x40017576:0x3ffcc0c0 0x40019d14:0x3ffcc0e0 0x40055b50:0x3ffcc100 0x400e413a:0x3ffcc120 0x400e4192:0x3ffcc140 0x400e3ba1:0x3ffcc160

0x400e413a: r_rw_schedule at ??:?
0x400e4192: btdm_controller_schedule at ??:?
0x400e3ba1: bt_controller_task at /Projects/esp32/esp-idf/components/bt/./bt.c:206

TianHao-Espressif, 2017-04-12:
@mringwal , if Host Buffer Size is set to 10, what's the complete event status?

mringwal, 2017-04-12:
@TianHao-Espressif the command complete event status for the Host Buffer Size command is OK/0. I'll be on IRC most of today and tomorrow - European time if you have further questions/ideas/etc..

mringwal, 2017-04-19:
@TianHao-Espressif, @projectgus hi guys! I think I found a bug in the Host Controller to Host Flow Control implementation of ESP32's component/bt/lib.

It looks like each HCI Host Num Completed Packets Command reduced the number of HCI Commands the Host can send. That's not correct, since this command is special (Core v5.0, Vol 2, Part E: HCI, 7.3.40). As a consequence, BTstack gets the HCI Command Complete Event with Memory Capacity Exceeded when sending roughly the 6th HCI Command.

I've added a quick hack to send a HCI Read BD ADDR command each time before sending a HCI Host Num Completed Packets to get a HCI Command Complete Event back that indicates current the num hci hci commands that we can send. In fact, this goes down by one after each HCI Host Num Completed Packets. Here are some log files in text & .pklg format (for WireShark(). 
[controller_to_host_flow_control_error.zip](https://github.com/espressif/esp-idf/files/937854/controller_to_host_flow_control_error.zip)

Please let me know if you can see the problem in the BT Controller code and when you have an updated version.

I guess Host Controller to Host Flow control wasn't tested with Bluedroid much... ;)

Edited snippet from the log

[00:00:03.976] CMD => 09 10 00
[00:00:03.979] EVT <= 0E 0A 03 09 10 00 C4 8B 00 C4 0A 24
3..

[00:00:03.990] CMD => 35 0C 05 01 00 00 01 00
[00:00:04.024] CMD => 09 10 00
[00:00:04.052] EVT <= 0E 0A 02 09 10 00 C4 8B 00 C4 0A 24
2..

[00:00:04.093] CMD => 35 0C 05 01 00 00 02 00
[00:00:04.115] CMD => 09 10 00
[00:00:04.118] EVT <= 0E 0A 01 09 10 00 C4 8B 00 C4 0A 24
1..

crash!

TianHao-Espressif, 2017-04-20:
Thank you for the information. We will do something about it.

mringwal, 2017-04-28:
hi there. Just wanted to check back and saw that the title was updated. Since the command only needs to update an internal counter, it shouldn't be hard to fix. If you have a version to test (e.g. based on the last public commit + the fix for this), I'm happy to give it a try!

mringwal, 2017-05-11:
@TianHao-Espressif Any idea of an ETA? Can you tell me if this is just a minor bug not decrementing the num commands, or is it a bigger issue and the flow control isn't implemented?  

Like, if ACL packets cannot be sent to the host, the Bluetooth Link Layer has to pause the remote from sending more packets, etc.. Thanks for any feedback on his issue!

TianHao-Espressif, 2017-05-18:
@mringwal , this bug is fixed. Please wait the master branch or release branch update.
Thank you very much.

mringwal, 2017-05-18:
Excellent! Thanks for taking care of this.

I'm "watching" the components/bt/lib project. If you remember, ping me here.

I'll test and validate the fix as soon as I get the update.

mringwal, 2017-05-22:
@TianHao-Espressif Thanks for the update. With the current master, the command counting is working properly and I'm closing this one. I still get the BT Controller thread to crash, but will create a new issue once I have more data.

closed 2017-05-22 by mringwal

TianHao-Espressif, 2017-05-24:
@mringwal, It updated.