espressif/esp-idf #482: Deep sleep consumption

SirRibeiro, 2017-04-03:
In Deep sleep mode is it normal for the ESP32 to be consuming 5mAh? 
Using 3.7V. The code I use is just your example deep_sleep.
It seems a little excessive, does not it?

Spritetm, 2017-04-03:
mAh? I presume you mean 5mA? Yes, that is excessive. What hardware are you using?

protocentralashwin, 2017-04-03:
I hope you have taken into consideration the current leakage due to the
voltage regulator, battery management and other power components of the
hardware that you are using.

SirRibeiro, 2017-04-03:
@Spritetm Yes of course sorry.
Now that you speak, maybe that's it, I'm using the watterott electronic ESP-WROOM32-breakout development board.
I'll try without all of this.
Thanks!

SirRibeiro, 2017-04-04:
I've tried now, without all the peripherals, and the consumption in deep sleep is just 5 to 7 micro amperes.
Amazing you guys are the best!
Many thanks
@Spritetm 
@protocentralashwin

closed 2017-04-05 by igrr