espressif/esp-idf #483: ULP Example not compiling

lmagalhaes1337, 2017-04-03:
Hey,

I've just followed the Linux Setup Guide and I'm trying to compile the ULP Example present in the examples/system/ulp folder and I'm always getting the same error:

CPP pulse_cnt.S
ULP_AS pulse_cnt.ulp.o
make[1]: esp32ulp-elf-as: Command not found
/home/esp32/esp/esp-idf/components/ulp/component_ulp_common.mk:40: recipe for target 'pulse_cnt.ulp.o' failed
make[1]: *** [pulse_cnt.ulp.o] Error 127
/home/esp32/esp/esp-idf/make/project.mk:386: recipe for target 'main-build' failed
make: *** [main-build] Error 2

Can anyone tell me what I'm doing wrong please?

igrr, 2017-04-03:
You need to get ULP coprocessor toolchain to compile this example, please check this page: http://esp-idf.readthedocs.io/en/latest/ulp.html#installing-the-toolchain

Platform specific getting started guides don't mention this yet, but you need to get esp32ulp-binutils to compile ULP programs.

lmagalhaes1337, 2017-04-03:
That was the answer to my problem. Thank you ! Although I still have problems with the assembly code.

closed 2017-04-03 by lmagalhaes1337

Noo-Noo, 2017-11-15:
Hello,
I have the same Problem.

Noo-Noo, 2017-11-15:
Hi, 
I have the same problem. 
I downloaded the toolchain extracted the toolchain into a directory but I don´t know how to add the path to the bin/ directory of the toolchain to the PATH environment variable.

Can anybody help a beginer?

krzychb, 2017-11-15:
Hi @Noo-Noo,
Sure! What is your OS?

Noo-Noo, 2017-11-15:
Thank you, my OS is Windows 7 and I used this guide:
https://esp-idf.readthedocs.io/en/latest/get-started/index.html

lmagalhaes1337, 2017-11-15:
For the ULP you have to follow this guide as well:

https://esp-idf.readthedocs.io/en/latest/api-guides/ulp.html

Noo-Noo, 2017-11-15:
I tried to follow this guide as well but I don´t know how to add the path to the bin/ directory of the toolchain to the PATH environment variable.

Installing the toolchain

ULP coprocessor code is written in assembly and compiled using the binutils-esp32ulp toolchain.

1. Download the toolchain using the links listed on this page: https://github.com/espressif/binutils-esp32ulp/wiki#downloads

2. Extract the toolchain into a directory, and add the path to the bin/ directory of the toolchain to the PATH environment variable.

krzychb, 2017-11-15:
#### How to add the path to the bin/ directory of the ULP toolchain to the PATH environment variable?

Applies to the ESP32 toolchain installed on Windows / MSYS2 following standard instructions under http://esp-idf.readthedocs.io/en/latest/get-started/windows-setup.html and ULP toolchain from https://github.com/espressif/binutils-esp32ulp/wiki#downloads

1.  Unzlip the ULP toolchain `esp32ulp-elf-binutils-win32-...` into MSYS2's `opt` directory. This will be typically `C:\msys32\opt` directory. I use this directory, since it also contains the ESP32 toolchain `xtensa-esp32-elf`
2.  Open `esp32_toolchain.sh` file in `C:\msys32\etc\profile.d` directory for editing and add the path to the 'bin/' directory of the ULP toolchain. The modified file will look as follows:

    ```
    # This file was created by ESP-IDF windows_install_prerequisites.sh
    # and will be overwritten if that script is run again.
    export PATH="$PATH:/opt/xtensa-esp32-elf/bin:/opt/esp32ulp-elf-binutils/bin"
    ```
3.  Save the file and restart MSYS2 console.
4.  Now you should be able to compile ULP programs.  Check it out following the standard instruction under https://esp-idf.readthedocs.io/en/latest/get-started/index.html like for any other ESP32 examples by compiling, e.g. https://github.com/espressif/esp-idf/tree/dec4a868d0303f53b438125b1b087f2f5e44a575/examples/system/ulp_adc

Good luck :smile:  :+1:

Noo-Noo, 2017-11-15:
It worked :-)
Thank you very much!!!

Aleixtejada, 2017-12-20:
@krzychb Thanks for the explanation!