espressif/esp-idf #484: ULP Assembly code issues

lmagalhaes1337, 2017-04-03:
Hi there,

I'm having some issues with the assembly code in the ULP coprocessor. I have made a small assembly code:

`#include "soc/rtc_cntl_reg.h"
#include "soc/rtc_io_reg.h"
#include "soc/soc_ulp.h"

        .bss
        .global loop_counter
loop_counter:
        .long 0

        .text
        .global entry
entry:
        move r3, loop_counter
        ld r3, r3, 0
        move r2, loop_counter
        //and r1, r3, 0x0011
        st r1, r2, 0
        jump wake_up
        halt

        .global wake_up
wake_up:
         wake
         halt`

The loop_counter variable is started with the value 7. When the line of code is "st r3, r2, 0", the value the main program prints is 7.

Modifying the code to
and r1, r3, 0x0011
st r1, r3, 0

Makes the main program print 1, which is wrong.

Modifying the code to
add r1, r3, 0x0001
st r1, r3, 0

makes the main program print 24, which, again, is wrong. Finally, modifying the code to:

sub r1, r3, 0x0001
st r1, r3, 0

makes the main program print 65526. Anyone has any idea of what might be going on? I've been trying to edit registers with no success and I tried to do something a little easier and that had an output that I could see.

igrr, 2017-04-05:
> Modifying the code to
and r1, r3, 0x0011
st r1, r3, 0

Sorry, it is not clear to me which lines exactly you are modifying... Does it happen after you do `ld r3, r3, 0`? If so, `r3` no longer contains the address of `loop_counter`, so `st r1, r3, 0` instruction you use in subsequent tests doesn't make much sense.

Could you please update the issue report to include the relevant part of the main application (or a link to a gist), and clarify what is being modified in the assembly part?

lmagalhaes1337, 2017-04-06:
Here you go, what I'm using to see the error: https://gist.github.com/lmagalhaes1337/c37c49eee3b1910985816fb1dd23854b

My putty output for the test value was 14. And it only works once for some reason... after that, everytime the ULP wakes the SoC you get a deepsleep reset and a TG0WDT_SYS_RESET right after, even though every watchdog has been disabled in make menuconfig.

igrr, 2017-04-06:
ULP program will be started by timer every 20ms (as configured in your main program)... by the time the chip exits deep sleep and starts up your main application again, ULP program runs 14 times.

If you want the ULP program to run once, you need to add a `WRITE_RTC_FIELD(RTC_CNTL_STATE0_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN, 0)` line before `halt`ing the program.

Regarding TG0WDT_SYS_RESET, could you please paste the part of the log related to resets?

lmagalhaes1337, 2017-04-06:
Adding that line between the wake and halt instructions in the wake_up routine makes the SoC never wake up at all.

Regarding the putty log, here's a printscreen.

[image]

I'm just showing the first iteration, since the other iterations are all the same. It does a deepsleep reset and then the tg0wdt sys reset. I can only see the value of the test variable on this first iteration.

igrr, 2017-04-06:
Right, because the ULP program first runs *before* the chip enters deep sleep. It triggers wakeup, but the chip is still woken up, so nothing happens. Then the chip finally enters deep sleep, but the ULP program has already stopped.

I think at this point you may just remove the deep sleep, since the program does not really need it. This will help you focus on the ULP code.

Regarding the WDT reset: this is regression in the current master branch, discussed here:
https://esp32.com/viewtopic.php?f=14&t=1568&p=7292
The workaround for now is to roll back to a commit mentioned in that topic.

Another workaround for WDT reset not mentioned in that topic is to add these two lines somewhere at the start of app_main:
```
    REG_SET_FIELD(RTC_CNTL_REG, RTC_CNTL_DBIAS_WAK, 4);
    REG_SET_FIELD(RTC_CNTL_REG, RTC_CNTL_DBIAS_SLP, 4);
```
(along with `#include "soc/rtc_cntl_reg.h"`) at the top of the file.

lmagalhaes1337, 2017-04-06:
But if you're not waiting for the ULP to wake up the chip, how do you know that the ULP code has already run ? Does it need some delay to make sure that the variable has been updated by the ULP ? I'd imagine something bad would happen if both of the cores tried to access the variable at the same time.

I'm gonna try that workaround.

igrr, 2017-04-06:
Well, for the non-production use (= debugging ULP program), i would just add a vTaskDelay(100 / portTICK_PERIOD_MS) after ulp_run. That would be enough for the ULP program to finish.

If you want to make it nicer, you can check the RTC_CNTL_ULP_CP_INT_RAW bit in RTC_CNTL_INT_RAW_REG, it is set to 1 once the ULP processor uses `wake` instruction.

Even nicer, use `esp_intr_alloc` to register RTC interrupt, enable the ULP interrupt using RTC_CNTL_ULP_CP_INT_ENA bit in RTC_CNTL_INT_ENA_REG. But that's probably not really needed.

igrr, 2017-04-11:
@lmagalhaes1337 did the above resolve the issue?

lmagalhaes1337, 2017-04-11:
Hi,

Unfortunately I haven't had the time to check, but I believe it does solve the problem. Most of the problems have been solved at this point. I'll get back to you on this if there's any other problem, but I don't believe there'll be another.

Thank you very much for your help!

igrr, 2017-04-11:
Ok, please feel free to reopen this or open a new issue if you encounter problems :)

closed 2017-04-11 by igrr

lmagalhaes1337, 2017-04-20:
Hi there,

I'm going to pick this up again and check those solutions. I was also searching around the rtc_cntl_reg.h file for any register to check if the main core is already asleep, but couldn't find any. I'm sure there's any way to do this, could you point me to the right direction? Also, is there any documentation where the registers present in this header are described?

Another thing is, is the ULP capable of detecting interrupts? You were talking about the RTC interrupt, but what about external interrupts connected to interrupt capable pins of the ESP32 ? Is there any way even to access a register which contains the value present in these pins?

Thanks. If needed I can open a new ticket with these questions.

Luis

igrr, 2017-04-20:
> any register to check if the main core is already asleep

I don't think we have such a register. Will double check though.

> is there any documentation where the registers present in this header are described

Some registers will be documented in the upcoming "low power sensors" and "ULP coprocessor" chapters of the TRM. Registers related to the low-level clock and power state machines aren't on the plan to be documented yet, but we will consider this.

> is the ULP capable of detecting interrupts

No, ULP doesn't have such a capability. You can either run the ULP often enough and poll the pins, or use EXT0 or EXT1 wakeup triggers and wake up the SOC when external pins change state. Polling RTC IOs is demonstrated in examples/system/ulp/, where ULP is counting pulses on one of the input pins.
Pay attention to the fact that you need to configure the IO as input and enable "hold" function prior to going into sleep mode. This is shown in `init_ulp_program` function.
Reading the state of lower 15 RTC_IOs into R0 is done using `READ_RTC_FIELD(RTC_GPIO_IN_REG, RTC_GPIO_IN_NEXT)` macro in pulse_cnt.S.

lmagalhaes1337, 2017-04-20:
My idea was to have the ULP constantly polling a register to see if the main core was asleep. When it entered sleep, the ULP would manage the interrupts and other circuitry and deciding when the main core would wake up again.

Also, you mentioned the IO configuration. I tried to define a pin on the main app and just toggle it on the ULP, but was unable to do so. I had this code in the main app:

```
gpio_num_t gpio_num = GPIO_NUM_0;
rtc_gpio_init(gpio_num);
rtc_gpio_set_direction(gpio_num, RTC_GPIO_MODE_OUTPUT_ONLY);
rtc_gpio_pulldown_dis(gpio_num);
rtc_gpio_pullup_dis(gpio_num);
```
But it didn't work. Instead, it needs this configuration on the ULP:

```
WRITE_RTC_REG(RTC_IO_TOUCH_PAD1_REG,RTC_IO_TOUCH_PAD1_HOLD_S,1,0)
WRITE_RTC_REG(RTC_IO_TOUCH_PAD1_REG,RTC_IO_TOUCH_PAD1_MUX_SEL_S,1,1)
WRITE_RTC_REG(RTC_IO_TOUCH_PAD1_REG,RTC_IO_TOUCH_PAD1_SLP_OE_S,1,1)
WRITE_RTC_REG(RTC_IO_TOUCH_PAD1_REG,RTC_IO_TOUCH_PAD1_TO_GPIO_S,1,1)
WRITE_RTC_REG(RTC_GPIO_ENABLE_W1TS_REG,RTC_GPIO_ENABLE_W1TS_S+11,1,1)
```

Am I doing something wrong? Or *what* am I doing wrong?

igrr, 2017-04-20:
`rtc_gpio_set_direction` configures output enable bit only for normal mode. As you see, if you want the pad to be enabled in sleep mode, you need to set SLP_OE instead of the normal OE bit. I think if you do equivalent initialization in the main app, then it should work.

lmagalhaes1337, 2017-04-20:
Ok.

Two comments:

1) The WDT workaround that you showed does not work, at least for me. It still has the TG0WDT reset.

2) I've been trying to develop code to read a GPIO, namely the RTCIO5 (GPIO35). I've copied the code from the example, which reads the GPIO0 pin but I can't seem to get it working. I have this piece of code before the initialization of the ULP code.

```
ulp_io_number = 5;
gpio_num_t gpio_num = GPIO_NUM_35;
rtc_gpio_set_direction(gpio_num, RTC_GPIO_MODE_INPUT_ONLY);
rtc_gpio_pulldown_dis(gpio_num);
rtc_gpio_pullup_dis(gpio_num);
rtc_gpio_hold_en(gpio_num);
```

And then I have this in the assembly code:

```
check:
        READ_RTC_FIELD(RTC_GPIO_IN_REG, RTC_GPIO_IN_NEXT)
        move r3, io_number
        ld r3, r3, 0
        rsh r0, r0, r3
        and r0, r0, 1
        jump check, eq
        jump set_gpio_one
        halt
```

The idea is: when a HIGH was detected on that GPIO, it would jump to a code where it blinks a gpio which is connected to a LED (set_gpio_one). But it never jumps to that code. I have also added the following line and configs:

```
//WRITE_RTC_REG(RTC_IO_ADC_PAD_REG,RTC_IO_ADC1_HOLD_S,1,0)
        //WRITE_RTC_REG(RTC_IO_ADC_PAD_REG,RTC_IO_ADC1_MUX_SEL_S,1,1)
        WRITE_RTC_REG(RTC_IO_ADC_PAD_REG,RTC_IO_ADC1_SLP_IE_S,1,1)
        //WRITE_RTC_REG(RTC_IO_ADC_PAD_REG,RTC_IO_ADC1_FUN_IE_S,1,1)
        //WRITE_RTC_REG(RTC_IO_ADC_PAD_REG,RTC_IO_ADC1_FUN_SEL_S,1,0)
```
But nothing worked so far. These pins aren't so straightforward as the touch sensor pins.

igrr, 2017-04-20:
Which code produces a WDT reset, the one in the gist you have posted earlier? https://gist.github.com/lmagalhaes1337/c37c49eee3b1910985816fb1dd23854b
Please post the output of `git describe` done in your IDF folder.

Regarding GPIO reading, could you please post the full code as well, so I can reproduce the issue on my side?

lmagalhaes1337, 2017-04-20:
Here's the code that's giving me trouble:

https://gist.github.com/lmagalhaes1337/c4b4d6ace440b33073918d415511c73e

The git describe output:

esp32@ubuntu:~/esp/esp-idf$ git describe
v2.0-rc1-387-g47b8f78

The workaround I was trying was adding these two lines:

```
REG_SET_FIELD(RTC_CNTL_REG, RTC_CNTL_DBIAS_WAK, 4);
REG_SET_FIELD(RTC_CNTL_REG, RTC_CNTL_DBIAS_SLP, 4);
```

igrr, 2017-04-21:
I've just tried your code — it doesn't exhibit TG0WDT reset here, just prints the following and then does nothing:

```c++
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:2664
load:0x40078000,len:7900
ho 0 tail 12 room 4
load:0x40080000,len:252
0x40080000: _iram_start at ??:?

entry 0x40080034
0x40080034: _iram_start at ??:?

Not ULP wakeup, initializing ULP
Entering deep sleep
```

Anything else I need to know to reproduce the TG0WDT issue?

lmagalhaes1337, 2017-04-21:
I've changed the assembly code, it needed to jump to the wake_up instead of the check label. I've changed the gist to reflect that.

igrr, 2017-04-21:
With regards to GPIO reading issue, if you add `rtc_gpio_init(gpio_num);` to your host program, then it will work. I see that once i disconnect GPIO35 from ground, LED on GPIO0 starts blinking.

Also I have hacked up your example a bit to wake up the chip instead of blinking. I still don't see TG0WDT reset happening, even though I have checked out commit 47b8f78 of the IDF. What kind of development board are you using? Does TG0WDT also happen in the deep_sleep example? Could you also add sdkconfig file to the gist?

lmagalhaes1337, 2017-04-21:
I'm using custom hardware. After your response, I went looking for some bugs/defects and it seems that this issue with the TG0WDT is related to the power supply. With 2.3V it shows that problem, but increasing the voltage level to 3.3V makes the problem go away.

The rtc_gpio_init(gpio_num) i had already gotten there today, it was definitly missing. But, in the ULP example that instruction isn't present. Also, did you check if there's a way for the ULP to check if the SoC is sleeping ? For example, RTC_CNTL_CPU_STALL_EN or other bits related to stalling the pro or app cpu? They can be used to stall, but will they have info if the cpu is already stalled? But is that the same as being asleep? Or even the RTC_CNTL_SLEEP_EN ?

lmagalhaes1337, 2017-04-21:
I was able to figure out a way to determine if the SoC is asleep:

```
check_sleep:
        READ_RTC_REG(RTC_CNTL_STORE5_REG,30,1)
        and r0, r0, 1
        jump set_gpio_one, eq
        jump check_sleep
        halt
```

This piece of code is enough to determine whether the SoC is sleeping (pin 30 = '0') or not (pin 30 = '1'). Awesome!

igrr, 2017-04-24:
You can check for `RTC_CNTL_DIAG0_REG & BIT(19)` — when this bit is set, RTC is ready to receive wakeup command. I'll add this to the documentation.

Checking RTC_CNTL_STORE5_REG may not be very reliable because this is essentially checking the value of APB frequency... In the future, it is possible that we will run at APB frequency = 40MHz while not in sleep mode.

lmagalhaes1337, 2017-04-24:
Oh ok. But then that bit indicates that the SoC is sleeping? I'm sorry, I didn't quite follow.

igrr, 2017-04-24:
This bit indicates that the SoC is sleeping *and* RTC is ready to receive wakeup command. In other words, when `RTC_CNTL_DIAG0_REG & BIT(19)` bit is set, `wake` instruction can be used to do the wakeup.

lmagalhaes1337, 2017-04-24:
Ok, cool! I'm gonna change my code to check the sleep that way.

lmagalhaes1337, 2017-04-24:
Is it possible to define a GPIO to wake up the ULP and run the code again ?
I tried this:

```
/* GPIO4 - Interrupt definitions */
        WRITE_RTC_REG(RTC_GPIO_PIN4_REG,RTC_GPIO_PIN4_WAKEUP_ENABLE_S,1,1)
        WRITE_RTC_REG(RTC_GPIO_PIN4_REG,RTC_GPIO_PIN4_INT_TYPE_S,1,1)

        /* GPIO5 - Interrupt definitions */
        WRITE_RTC_REG(RTC_GPIO_PIN5_REG,RTC_GPIO_PIN5_WAKEUP_ENABLE_S,1,1)
        WRITE_RTC_REG(RTC_GPIO_PIN5_REG,RTC_GPIO_PIN5_INT_TYPE_S,1,1)

        WRITE_RTC_REG(RTC_IO_EXT_WAKEUP0_REG,RTC_IO_EXT_WAKEUP0_SEL_S,4,0x0004)
        WRITE_RTC_REG(RTC_CNTL_EXT_WAKEUP_CONF_REG,RTC_CNTL_EXT_WAKEUP0_LV_S,1,1)
        WRITE_RTC_REG(RTC_CNTL_STATE0_REG,RTC_CNTL_SLP_WAKEUP_S,1,1)
```

I wanted both of them to be able to wakeup the ULP but I guess that's not happening.

igrr, 2017-04-24:
Sorry, these registers don't have anything to do with waking up the ULP. These are for waking up the system from deep sleep, using an external signal. Take a look at `ext0_wakeup_prepare` and `ext1_wakeup_prepare` functions in components/esp32/deep_sleep.c to see how these registers are configured for these wakeup modes.

There are only two ways in the hardware to start the ULP program: repeated start by timer (which is used by `ulp_run` function) and one-time by software. You could maybe use ext1 wakeup mode to wake up the system, then trigger ULP program by restarting the timer, and then immediately going back to sleep again. Alternatively, just keep polling the GPIO using timer wakeup mode of the ULP.

lmagalhaes1337, 2017-04-28:
Hi Ivan,

Thank you for your responses! They've been really helpful. But now, I stumbled upon a new problem, again with the flash read error on the deep sleep wake up leading to the TW0WDT reset. Here's a code with the problem https://gist.github.com/lmagalhaes1337/3311b0ea8c0eed6497a5e2a672c5f787. This code works flawlessly with this git describe: v2.0-rc1-387-g47b8f78 but fails with this one v2.0-rc1-666-g0fe765a. How can I roll back ? And also, what might be the problem with the newest code?

lmagalhaes1337, 2017-04-28:
Changing the ESP32_DEEP_SLEEP_WAKEUP_DELAY in the make menuconfig menu to 1000 fixed this problem, apparently.