espressif/esp-idf #485: API for getting value of output pin

kostyan5, 2017-04-03:
There doesn't seem to be a function to read the value of a pin set as output.  

gpio_get_level() only returns the status of the "input" parameter.  If you want to toggle an output pin by doing something like this, the value of gpio_get_level(pin) never changes.
    uint32_t new_state = 1 - gpio_get_level(pin);
    gpio_set_level(pin, new_state);

In order to do a toggle, I had to use
  uint32_t new_state = 1 - (GPIO.out >> pin & 0x1);
  gpio_set_level(pin, new_state);

kewalmshah, 2017-05-11:
Hi @kostyan5 in order to get level of gpio set as output, you can set gpio pin mode to GPIO_MODE_INPUT_OUTPUT and then call gpio_get_level(pin)

FayeY, 2017-06-14:
Hi @kostyan5 , did the above suggestion help you resolve the issue?

kostyan5, 2017-06-14:
@FayeY Haven't had time to check it out.  I'll update in the next couple of weeks.

FayeY, 2017-07-05:
Hi @kostyan5 , I'll close this issue. If your problem is still unsolved, please feel free to reopen it.

closed 2017-07-05 by FayeY