espressif/esp-idf #486: rmt_driver_uninstall() unconditionally de-registers ISR for all other channels

devsaurus, 2017-04-03:
The following use case involving several RMT channels apparently does not work with the current RMT driver implementation.

Channel 0 is used persistently. I.e. after initial configuration and driver installation, the channel is operated as required and is supposed to be usable until the chip is reset.

Channel 1 is used temporarily. I.e. it cycles through the sequence of configuration, installation and uninstallation from time to time.

I observe that channel 0 becomes unresponsive once channel 1 was temporarily active for the first time. Channel 1 is not affected and it will operate as expected each time it's activated even when channel 0 is already broken.

Looking at `components/driver/rmt.c` I suspect that the first call to `rmt_driver_uninstall()` will de-register the ISR even though another channel is still installed and relies on the interrupts being serviced. This is contrary to my understanding that channels can be installed & uninstalled independently from each other since the driver API requires a channel number for both of these functions (especially for `rmt_driver_uninstall()`).
Please clarify the intended API use model.

igrr, 2017-04-06:
Thanks for pointing this out. The uninstall function should check if any of `p_rmt_obj[channel]` for other channels is still non-zero before calling `rmt_isr_deregister` and resetting `s_rmt_driver_installed`.

/cc @costaud

costaud, 2017-04-06:
I'll take this.

LDmitryL, 2017-04-24:
@igrr If there's a solution to overcome the problem of incompatibility with RMT module DHT21/22/AM2301

FayeY, 2017-06-19:
Hi, is this problem solved now, with the latest IDF?

igrr, 2017-06-19:
The fix was merged into IDF, but we are having some issues deploying to Github, so some time is needed for the fix to appear on Github.

igrr, 2017-06-20:
Fixed in https://github.com/espressif/esp-idf/commit/c77b699

closed 2017-06-20 by igrr