espressif/esp-idf #487: Calling esp_bt_controller_init() causes reset

MrSurly, 2017-04-03:
I'm working on implementing BT support for the ESP32 port of MicroPython.  For early testing, I created a simple test() function and called it from with in the MicroPython interpreter.

```c
STATIC mp_obj_t network_bluetooth_test(mp_obj_t self_in) {
    NETWORK_BLUETOOTH_DEBUG_PRINTF("Entering network_bluetooth_test\n");

    { // make_new code
        esp_bt_controller_init(); 
#if 0
        esp_err_t ret = esp_bt_controller_enable(ESP_BT_MODE_BTDM);

        switch(ret) {
            case ESP_OK:
                NETWORK_BLUETOOTH_DEBUG_PRINTF("BT initialization ok\n");
                break;
            default:
                NETWORK_BLUETOOTH_DEBUG_PRINTF("BT initialization failed, code: %d\n", ret);
        }
#endif
    }

    return mp_const_none;
}
```
As you can see, it only calls `esp_bt_controller_init()`.  Originalyl the `test()` function was being called after initializing WiFi; I moved it so that Wifi was not used, and it still results in a reset.

```
>>> import machine as 
>> b = m.Bluetooth()
Entering network_bluetooth_make_new
>>> b.test()
Entering network_bluetooth_test
BTDM CONTROLLER VERSION: 010101
btip start
copy .data from 4000d890 to 3ffae6e0, len 00001830
set .bss 0x0 from 3ffb8000 to 3ffbff70, len 00007f70
BTDM ROM VERSION 0101
BD_ADDR: 24:0A:C4:05:7E:E6
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:2372
load:0x40078000,len:6708
load:0x40080000,len:252
entry 0x40080034
I (1526) cpu_start: Pro cpu up.
I (1527) cpu_start: Single core mode
I (1528) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1541) heap_alloc_caps: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1561) heap_alloc_caps: At 3FFD1668 len 0000E998 (58 KiB): DRAM
I (1582) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1604) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1625) heap_alloc_caps: At 4009A898 len 00005768 (21 KiB): IRAM
I (1646) cpu_start: Pro cpu start user code
I (1707) cpu_start: Starting scheduler on PRO CPU.
OSError: [Errno 2] ENOENT
OSError: [Errno 2] ENOENT
MicroPython v1.8.6-1015-g6f5b5d5-dirty on 2017-04-03; ESP32 module with ESP32
Type "help()" for more information.
>>>
```

I'm using commit hash 47b8f78cb0e15fa43647788a808dac353167a485 for the ESP-IDF, which is HEAD as of this writing.

dpgeorge, 2017-04-03:
It might mave something to do with the sdkconfig settings (or lack thereof).  I'd suggest you compile a simple BT example from the IDF and then inspect the generated sdkconfig.h file, comparing it to uPy's one.

MrSurly, 2017-04-04:
I _did_ have to make some `sdkconfig.h` changes.  I'll try your suggestion.

MrSurly, 2017-04-04:
I did make some (more) changes to `sdkconfi.h`, including setting `#define CONFIG_BT_RESERVE_DRAM 0x5000` (was `0x0`).  The recommended setting of `0x10000` causes several reboots and a lockup upon boot; never even get to the μPy prompt.

It turns out `esp_bt_controller_init()` never actually returns.

Increasing the WDT (`#define CONFIG_INT_WDT_TIMEOUT_MS 1000`) causes the reset to be delayed.

MrSurly, 2017-04-04:
Okay, so tracking this a bit further.

Seems that the call to `taskYIELD_IF_USING_PREEMPTION_MUX` in [tasks.c line 1169](https://github.com/espressif/esp-idf/blob/master/components/freertos/tasks.c#L1169) is never returning, which led me to issue #374, which appears to be about using `taskYIELD_IF_USING_PREEMPTION_MUX` inside of critical regions.  It appears there might be changes to this code soon, per the comments.

I don't know enough about FreeRTOS to comment further, I'm afraid.

dpgeorge, 2017-04-05:
> The recommended setting of 0x10000 causes several reboots and a lockup upon boot; never even get to the μPy prompt.

It looks like you need 64k for the BT stack, and that requires to reduce the uPy heap to around 90k to get everything to fit (the reboots you see are due to FreeRTOS not having enough available heap).  I'd set the uPy heap to 64k just to be on the safe side while developing.

@projectgus is 64k a hard lower limit for the BT stack or can we get away with allocating less for it?

Spritetm, 2017-04-05:
Hey guys,

FYI, BT_RESERVE_DRAM is *not* a tweakable knob. (E.g. If you use 'make menuconfig', you will not find it anywhere in the menu. It gets set as a result of other settings). The only valid values for this are 0 if BT is disabled and 0x10000 if it is enabled. It does not size a pool of memory or something, it actually reserves a range of memory the BT firmware in ROM uses.

Please, if you need to modify the sdkconfig, use 'make menuconfig'. That is the documented and supported way of changing stuff. There's a good reason sdkconfig.h has a big 'DO NOT EDIT' on the top of it.

dpgeorge, 2017-04-05:
> The only valid values for this are 0 if BT is disabled and 0x10000 if it is enabled

Ok, thanks for the info.

> Please, if you need to modify the sdkconfig, use 'make menuconfig'.

We don't use the IDF build system, we just compile each source file with our own Makefile, and hence need to generate the sdkconfig.h by hand.

@MrSurly I suggest you close this issue.

Spritetm, 2017-04-05:
I still highly suggest you find a way to run 'make menuconfig' to edit the config files. Worst case, start a dummy project, copy config files there. Best case, add a subdir in your project where you can trick esp-idf into doing a make menuconfig, then linking/including the files from there. If you keep insisting on editing config options like this, I predict you're going to run into things like this a lot more.

MrSurly, 2017-04-05:
> I still highly suggest you find a way to run 'make menuconfig' to edit the config files. Worst case, start a dummy project, copy config files there. Best case, add a subdir in your project where you [...]

Thanks.  I'll see if I can make one of these suggestions work.

closed 2017-04-05 by MrSurly