espressif/esp-idf #488: spi not sendig null

dannybr, 2017-04-04:
sending only nulls does not work!
I'm developing on a esp-wroom-32.

spi_transaction_t t;
t.tx_data[0] = 0
t.length = 8;
t.flags = SPI_TRANS_USE_TXDATA;
spi_device_transmit(spi, &t) returns without error but
the logic analizer shows nothing,

if length is > 1*8, and one of the bytes is NULL it does work!
so is this firmware or hardware issue?

jaanh, 2017-04-22:
I have the same issue. Below is another example where no data is actually sent (e.g no CS pin toggling or CLK pulses).
```
t[1].tx_data[0] = 0x00;
t[1].tx_data[1] = 0x00;
t[1].tx_data[2] = 0x00;
t[1].length = 3 * 8;
```

I have found a workaround though:
t[1].tx_data[0] = 0x00;
t[1].tx_data[1] = 0x00;
t[1].tx_data[2] = 0x00;
t[1].tx_data[3] = 0x01;
t[1].length = 3 * 8; // Set length to 3 to only send the first 3 bytes

closed 2017-04-29 by igrr

dannybr, 2017-04-29:
> Closed #488 <https://github.com/espressif/esp-idf/issues/488> via bf0c323 <https://github.com/espressif/esp-idf/commit/bf0c32364aa086742551f53d8a6db0fab505cc61>.

thanks!