espressif/esp-idf #490: GDB crashes with segmentation fault while debugging

Sermus, 2017-04-04:
I have ULINK2 which i connected to ESP32 and pretty much successfully debugged.
However there is a situation when xtensa-esp32-elf-gdb crashes with segmentation fault.

The version of GDB i use:
`GNU gdb (crosstool-NG crosstool-ng-1.22.0-61-gab8375a) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf".
`

The coredump is attached.
[core.zip](https://github.com/espressif/esp-idf/files/892718/core.zip)

igrr, 2017-04-11:
gdb binaries in the precompiled toolchain archive are stripped, so I'm not sure we can learn a lot from the core file. Are you able to produce this crash reliably using some application, some sequence of commands in gdb, or both? If so, these steps would probably would help to identify the issue more than the core file.

Sermus, 2017-04-11:
1) There is no sequence. I just start gdb, then let it continue the execution flow by "c" command. After several seconds it crashes and generates coredump.
2) There is no sample code i can repro this on. This is the situation we faced debugging our production code for ESP32.

If you indicate what should be sufficient to get enough information i could recompile esp32 gdb (with debug info?) and try to repro the issue. Alternatively you could send the compiled binary to me and let me repro with it.

konkrog, 2017-10-04:
Is this issue resolved? Does the ULINK2 adapter work well together with ESP32 - or is this issue related to the use of ULINK2?

Sermus, 2017-10-04:
Nop, it's not resolved. As i can see, the issue is not related to the jtag hardware used. I face this with WROVER v3 and FT2232 from time to time.
The GDB version is:
GNU gdb (crosstool-NG crosstool-ng-1.22.0-61-gab8375a) 7.10 
(precompiled binaries taken from https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-73-ge28a011-5.2.0.tar.gz).

OCD version is:
Open On-Chip Debugger 0.10.0-dev-ga859564 (2017-07-24-16:16)
(precompiled binaries are taken from https://dl.espressif.com/dl/openocd-esp32-linux64-a859564.tar.gz)

If you let me know what i can do to provide information needed for analysis, it'll be my pleasure to do. Sometimes the issue is quite annoying.

konkrog, 2017-10-04:
Sorry I might have given the impression that I could help to fix the issue... but I'm not... I'm in the process of buying a JTAG adapter for use on ESP32, and ULINK2 looks like a good candidate, but then I found this issue... and was wondering if it might not be a good choice... 

It now sounds like ULINK2 is as good a choice as others.... so I might proceed getting one. Once I get it I might be experiencing the issue you are seeing...

konkrog, 2017-10-04:
Did you try compiling from source to include debug info instead of the precompiled versions? Doing that would give a core that could be analyzed...

Sermus, 2017-10-04:
Ok, got it. The only thing i can say about ULINK is that it's much slower than FT2232. Subjectively the latter is 10 times faster.