espressif/esp-idf #493: python cannot be found on March 30 release of MSYS

amatani, 2017-04-04:
'make' fails with the 20170330 release of "esp32_win32_environment_and_toolchain-20170330.zip"
`$ make
/bin/sh: python: command not found
make[1]: *** [/c/projects/esp32/esp-idf/components/esptool_py/Makefile.projbuild:49: /c/projects/esp32/myproj/esp-idf-template/build/bootloader/bootloader.bin] Error 127
make: *** [/c/projects/esp32/esp-idf/components/bootloader/Makefile.projbuild:31: /c/projects/esp32/myproj/esp-idf-template/build/bootloader/bootloader.bin] Error 2
`
Checked for python.exe:
`$ which python
which: no python in (/usr/local/bin:/usr/bin:/bin:/opt/bin:/c/Windows/System32:/c/Windows:/c/Windows/System32/Wbem:/c/Windows/System32/WindowsPowerShell/v1.0/:/opt/xtensa-esp32-elf/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl)
`
and path
`$ echo $PATH
/usr/local/bin:/usr/bin:/bin:/opt/bin:/c/Windows/System32:/c/Windows:/c/Windows/System32/Wbem:/c/Windows/System32/WindowsPowerShell/v1.0/:/opt/xtensa-esp32-elf/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl
`
Compared with "esp32_win32_environment_and_toolchain-20170111.zip" release and build completes.
Found the following differences:
`$  which python
/mingw32/bin/python
`
and
`$ echo $PATH
/mingw32/bin:/usr/local/bin:/usr/bin:/bin:/c/windows/system32:/c/windows:/c/windows/system32/wbem:/opt/xtensa-esp32-elf/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl
`
In both cases, using esp-idf v2 release from github

projectgus, 2017-04-04:
Hi @amatani ,

I'm not able to reproduce this. If I take a clean VM and unzip the `esp32_win32_msys2_environment_and_toolchain-20170330.zip` file (size 502251094 bytes, md5sum afe3f5ed11bfa6bcc4088547e1041d16) to `C:\MSYS32` and then follow these steps:
* Use Windows "Run" (Window-R) "C:\msys32\mingw32.exe" to open a terminal
* Type `which python` at the prompt.

... and python.exe is correctly found (its Windows path is `C:\msys32\mingw32\bin\python.exe`).

Is it possible that the unzipping process was interrupted, so not all files were unzipped to the filesystem? Or are you running a different msys32 directory's prompt?

amatani, 2017-04-04:
Hello Angus,
Thanks for looking into this.
On my Windows 10 PC, I did a fresh download of zip archive from:
https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20170330.zip
The checksum matches:
md5sum = afe3f5ed11bfa6bcc4088547e1041d16

I run C:/msys32/msys2_shell.cmd
[image]

Seems to me that "/mingw32/bin" is not in the path
The ~20170111.zip version does have this in the path.

Also, I see this:
[image]

amatani, 2017-04-04:
Sorry I closed by mistake, so reopened

amatani, 2017-04-04:
Could this be the problem due to differences in ~/msys32/etc/profile.d/esp32_toolchain.sh ?

[image]

amatani, 2017-04-05:
FYI: I added the path along with IDF_PATH and build completes

[image]

Also checked 'make flash' and it works fine.

[image]

projectgus, 2017-04-05:
What you've done will work. The other difference is running MSYS vs MINGW32:

> I run C:/msys32/msys2_shell.cmd

You need to run `C:/msys32/mingw32.exe` and then you'll get /bin/mingw32 on your path automatically.

You can tell the difference in the screenshots, my screenshot has purple MINGW32 and your screenshots have purple MSYS. Different shell types have different paths (there are annoying technical reasons for this, we need a MINGW32 "native" Python not an MSYS Python in order to use COM ports natively.)

The old instructions said to run the other command line, which was a mistake. This has been fixed in the newer instructions which should say to run `mingw32.exe`.

amatani, 2017-04-05:
OK, I see now.
[image]

I missed it at I had skipped to STEP 4 to configure IDF_PATH
I suggest that the document be updated for STEP 4 also.
[image]

Thanks. I have reverted to the new way and I like the fact that 'make monitor' works on my Windows 10 machine !!! (no more opening PuTTY, and remembering to close when firmware update is needed)

You may close this issue when appropriate (i.e. if you decide to update STEP 4 in doc or otherwise)

projectgus, 2017-04-05:
Snap, sorry about that! Will fix ASAP.

closed 2017-04-06 by igrr

fake-name, 2017-04-15:
Out of curiosity, if things like the `msys2_shell.cmd` can cause issues like this, why is it even still present?

Wouldn't it be better to remove the other ways of executing the msys shell, to reduce confusion? Or are they executed somehow?

Out of curiosity, I moved `mingw64.exe`, `msys2.exe`, and `msys2_shell.cmd` out of the directory, and things *seem* to build OK.

3DgeekStore, 2017-06-16:
Same error happened to me , but I was using mingw64 after changed to mingw32 no more error

sravanth005, 2019-03-01:
admin-unizen@adminunizen-Lenovo-ideapad-520-15IKB:~/esp-who/examples/single_chip/camera_web_server$ make flash
Toolchain path: /home/admin-unizen/esp32/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-61-gab8375a
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Compiler version: 5.2.0
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-61-gab8375a
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
The following Python requirements are not satisfied:
future>=0.15.2
cryptography>=2.1.4
Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.
Alternatively, you can run "/usr/bin/python -m pip install --user -r /home/admin-unizen/esp/esp-idf/requirements.txt" for resolving the issue.
/home/admin-unizen/esp/esp-idf/make/project.mk:500: recipe for target 'check_python_dependencies' failed
**make: *** [check_python_dependencies] Error 1**

how to correct this error ,as im a fresher do anyone knows how to correct this error send me step by step .if i need to download anything means send a direct link for that

thank you

commit 26bf1e84 (2017-04-06): Merge branch 'bugfix/doc_windows_shell' into 'master'

docs: Fix Windows shell path in Windows Setup Guide

Closes #493 https://github.com/espressif/esp-idf/issues/493

See merge request !634
commit 60e81e5e (2017-04-06): docs: Fix Windows shell path in Step 4

Closes #493 https://github.com/espressif/esp-idf/issues/493